dsmrreader/dsmr-reader #1182: Veranderen van TOPIC en HA

poudenes, 2020-11-02:
Hoi Hoi,

Is het mogelijk om de TOPIC Path aan te passen en dat deze ook gewoon nog door HA discovery wordt gezien?

Bijvoorbeeld:
```
[mapping]
# DATA = TOPIC PATH
electricity1 = dsmr/day-consumption/electricity1
electricity2 = dsmr/day-consumption/electricity2
```

naar:
```
[mapping]
# DATA = TOPIC PATH
electricity1 = DSMR_Reader/day-consumption/electricity1
electricity2 = DSMR_Reader/day-consumption/electricity2
```

poudenes, 2020-11-02:
Was te snel met posten. Lees nu de integration op HA en die geven aan:

sensor_dsmr_<mqtt_topic>

Ik ga er dan vanuit dat mijn vraag eigenlijk al beantwoord is. Nee is niet mogelijk.

dennissiemensma, 2020-11-02:
Bedankt voor je vraag. Je kunt zeker het topic pad/naam wijzigen in DSMR-reader. Alleen kan het dan inderdaad zijn dat andere systemen een ander pad verwachten (als dat niet in te stellen is).

closed 2020-11-02 by dennissiemensma

poudenes, 2020-11-02:
Als ik besluit het alsnog te doen laat ik hier wel weten wat ik gedaan heb. 
Ik denk dat ik dan handmatig alles moet invoeren en MQTT sensors aanmaken met de juiste paden erin.