dsmrreader/dsmr-reader #1183: DSMR start wel op maar webui niet meer na laatste update

canedje, 2020-11-02:
Troubleshooting: https://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html#log-files

(Heb je DSMR-reader v4.4 of hoger? Zet dan hieronder ook de output van "dsmr_debuginfo") 

---
Vandaag DSMR geupdate.
Hierna start de docker wel op, maar de webui opend niet meer.
Geen Log aanwezig om uit te lezen. ook geen dsmr_debug
Ik draai DSMR onder Unraid in een docker. Tot voor de update werkte het normaal.
hoe kan ik terug naar de vorige versie?

dennissiemensma, 2020-11-02:
Bedankt voor je melding. Wellicht kun je het beste even informeren bij Xirixiz om te debuggen waar het in zit: https://github.com/xirixiz/dsmr-reader-docker

Het is vast iets kleins. 

En anders zou het zomaar iets zoals #1129 kunnen zijn, wat een naadloze upgrade in de weg kan staan. Maar dan moet je wel eerst ergens die melding daadwerkelijk in je eigen logs zien (niet zomaar dingen in je DB aanpassen dus!).

rogierlommers, 2020-11-02:
Probleem is inmiddels hier opgelost: https://github.com/xirixiz/dsmr-reader-docker/issues/148
Deze mag dicht dus.

dennissiemensma, 2020-11-02:
Top, dank voor de terugkoppeling!

closed 2020-11-02 by dennissiemensma