dsmrreader/dsmr-reader #1187: Manual fresh install V4

lubbertkramer, 2020-11-04:
Looking into the documentation i want to do a fresh install of V4 because of the active development instead of installing V3. I see the settings are now stored in the .env file that i use for several dockers. Where can i find more information or a manual on installing V4 and which settings/enviroments are needed in the .ENV file (also the correct naming for example  the secretkey)?

---

dennissiemensma, 2020-11-04:
See https://dsmr-reader.readthedocs.io/en/v4/installation/explained.html and https://dsmr-reader.readthedocs.io/en/v4/env_settings.html and https://github.com/dsmrreader/dsmr-reader/blob/v4/.env.template

dennissiemensma, 2020-11-04:
Also, there is a docker version by xirixiz: https://github.com/xirixiz/dsmr-reader-docker

lubbertkramer, 2020-11-05:
Thanks, have it up and running in Docker with the variables mentioned in the env file above :)

closed 2020-11-05 by lubbertkramer