dsmrreader/dsmr-reader #1188: Retentie configured maar krijg nog steeds waarschuwing dat database te groot is

Diondk, 2020-11-05:
Troubleshooting: https://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html#log-files

(Heb je DSMR-reader v4.4 of hoger? Zet dan hieronder ook de output van "dsmr_debuginfo") 

---

Ik heb de data retention policy staan op Clean up most source data after one week (recommend.)
Echter blijft de grote van de database het zelfde (dit is nu al enkele weken enabled) en lijkt hij niet op te ruimen. 
De database is nu 2483 MB groot en wordt niet kleiner. 

Zie ik hier iets over het hoofd of loopt dit process niet lekker bij mij?

```

DSMR-READER
    App / Python / Database                                                        v4.6 / v3.9.0 / postgresql
    Backend sleep / Datalogger sleep / Retention cleanup                                   1.0s / 5.0s / 168h
    Telegram latest version read / Parser settings                                                 "50" / "4"

DATA
    Telegrams total (est.)                                                                             629540
    Consumption records electricity / gas (est.)                                                 16209 / 7950

UNRESOLVED ISSUES
    Database growing large: 2438 MB, consider data cleanup (if not already enabled)                                     now

POSTGRESQL SIZE OF LARGEST TABLES (> 500 MB)
    public.dsmr_datalogger_dsmrreading                                                                2354 MB

```

Calimerorulez, 2020-11-05:
Volgens mij geeft Postgres niet meteen de geschoonde ruimte vrij. Dit kun je wel (op eigen risico) forceren door een `vaccumdb` op de server te runnen, weet ik uit ervaring :)

dennissiemensma, 2020-11-05:
Je kunt inderdaad eenmalig een vacuumdb doen. Zie ook https://github.com/dsmrreader/dsmr-reader/issues/1165#issuecomment-720061201

dennissiemensma, 2020-11-06:
En mocht je geen Docker gebruiken, dan:

```
sudo su - postgres
vacuumdb -f -v -d dsmrreader
```

closed 2020-11-06 by dennissiemensma

voipmeister, 2021-02-25:
> En mocht je geen Docker gebruiken, dan:
> 
> ```
> sudo su - postgres
> vacuumdb -f -v -d dsmrreader
> ```

Ter bevestiging, dit heeft bij mij het probleem verholpen.

Is het een idee om in de docs hier wat over toe te voegen? Ik zocht de oplossing hier: https://dsmr-reader.readthedocs.io/en/v4/faq.html#how-do-i-resolve-the-warning-regarding-database-size en was in de veronderstelling dat de retentie eerst gefikst mocht zijn voordat ik de `vacuumdb` kon draaien.

dennissiemensma, 2021-02-25:
@voipmeister klopt! Je kunt de vacuum uiteraard vaker draaien, maar het heeft pas nut als er daadwerkelijk (veel) gegevens zijn verwijderd en dat hangt per geval weer af van de hardware en hoeveelheid gegevens. Soms is het met een uur opgeschoond, soms pas na enkele dagen. Vandaar de wat generieke uitleg daar.