dsmrreader/dsmr-reader #1189: versie nummer niet meer in hoofdscherm

newtech2015, 2020-11-05:
hallo,

zojuist dsmr geupdate van 4.4.3 naar 4.6.0, werkt op zich goed, echter valt me op dat niet meer links boven naast "dsmr reader" de geinstalleerde versie wordt weergegeven.

hoort dit nu zo, of is er iets mis gegaan?

gr ron

dennissiemensma, 2020-11-05:
Scherp gezien! Dat klopt. Dit is verplaatst naar de About-pagina, bijna onderin het menu.

closed 2020-11-06 by dennissiemensma

Wouter1702, 2020-12-11:
Beste Dennis,

Ik vond het wel handig dat het DSMR versienummer linksboven werd weergegeven. Ik maak gebruik van de docker versie van DSMR icm watchtowen. Door het verplaatsen van het versie nummer naar de about pagina kan ik nu helaas niet meer in 1 oogopslag zien welke versie ik gebruikt.
Wellicht is het mogelijk om het configureerbaar te maken de versie op 2 plaatsen weer te geven ;) (in de about pagina en linksboven naast DSMR Reader)

Met vriendelijke groet,
Wouter

dennissiemensma, 2020-12-11:
Hallo Wouter bedankt voor je verzoek. Ik begrijp goed dat het handig was dat je meteen de versie kon zien. Het is echter verplaatst omdat het uiteindelijk niet zoveel uitmaakt welke versie je draait. De updates die er zijn, zijn doorgaans kleine fixes of toevoegingen en veranderen de basis van DSMR-reader niet heel veel. Daarnaast is de interface er wat eenvoudiger door geworden.

Omdat iedereen zijn eigen voorkeur heeft voor hoe iets er uit ziet (je bent niet de eerste) en het onmogelijk is om iedereen tevreden te houden, laat ik het voorlopig voor wat het is. Sowieso zie je (als het goed is), binnen een week na een nieuwe release, een melding daarvan in DSMR-reader zelf, dus je zult niet snel achterlopen.