dsmrreader/dsmr-reader #1192: Import old data from p1mon

EYEVO, 2020-11-07:
Voorheen gebruikte ik p1mon om mijn meter uit te lezen hier had ik al meer als 1 jaar aan data in staan en dit wil ik in DSMR reader hebben. P1mon maakt gebruik van sqlite en schrijft dagelijks een dump naar in mijn geval dropbox. Om deze info in DSMR reader te krijgen heb ik (met behulp van @ARYENN) een python script in elkaar gefietst om dit op dag niveau te importeren middels de API. Daarna de postgres commands gedraaid die in de api docs staan. En zie nu history in de database.
Ik zie het nog niet via de webportal onder archive maar houd jullie hier op de hoogte als dit nog update.

Hierbij het script en een stukje example van de dump.

```
import requests
import time
import json

url = "https://verbruik.replacemij.com/api/v2/datalogger/dsmrreading"
api_key = "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"

with open("historie.txt", "r") as file:
    content = file.readlines()

    for i in content:
        if "e_history_dag" in i:

            dirty_values = i.split("(")[2].replace(")","").replace(";","").strip()
            clean_values = dirty_values.replace("'", "").replace(" ","T").replace("00:00:00","00:00:00Z").split(",")

            payload = {
                "timestamp": clean_values[0],
                "electricity_delivered_1": str(clean_values[1]),
                "electricity_returned_1": "0",
                "electricity_delivered_2": str(clean_values[2]),
                "electricity_returned_2": "0",
                "electricity_currently_delivered": "0",
                "electricity_currently_returned": "0",
                "extra_device_timestamp": clean_values[0],
                "extra_device_delivered": str(clean_values[7]),
            }

            response = requests.post(url, data=payload, headers={'X-AUTHKEY': api_key})

            if response.status_code != 201:
                print('Error: {}'.format(response.text))
            else:
                print('Created: {}'.format(json.loads(response.text)))

            time.sleep(0.5)
```
[historie - example.txt](https://github.com/dsmrreader/dsmr-reader/files/5504894/historie.-.example.txt)

dennissiemensma, 2020-11-07:
Bedankt voor je toelichting. Aan het voorbeeld te zien zijn het dagtotalen. Je kunt daarom ook nog overwegen om het direct om te zetten naar queries die records aanmaken in `dsmr_stats_daystatistics`. 

Dat is de belangrijkste tabel voor het Archief. Het importeren van dagtotalen als losse metingen via de API kan ook eventueel ook werken, maar eerdergenoemde is wellicht wat nauwkeuriger. Laat maar weten als je daar een voorbeeldje voor nodig hebt.

EYEVO, 2020-11-08:
Wat raar is hij lijkt alle gegevens goed gezet te hebben voor gas t/m maart 2020 daarna niet meer. Ik denk dat ik het dan inderdaad maar ga schrijven naar de DB ipv de API.

dennissiemensma, 2020-11-08:
Ik zal sowieso kijken of ik in een volgende versie het aanmaken van dagtotalen ook kan ondersteunen via de API. Dat maakt het sowieso wat makkelijker om data te importeren, zonder dat per meting te hoeven doen.

Tot die tijd kun je het als queries in eerdergenoemde tabel doen. PostgreSQL heeft overigens ook nog ondersteuning voor CSV-formaten, mocht dat makkelijker zijn, maar in de praktijk komt (oa ik zelf) het vaak neer op gewoon queries.

EYEVO, 2020-11-08:
Het is gelukt! Alles staat erin. hieronder het script voor als iemand in de toekomst het zelfde wil doen. Het zal niet de schoonheidsprijs maar het werkt :)

```
import psycopg2
import requests
import time
import json

user = "dsmrreader"
password = "dsmrreader"
host = "10.0.1.50"
port = "5432"
database = "dsmrreader"

try:
    connection = psycopg2.connect(user=user, password=password, host=host, port=port, database=database)
    cursor = connection.cursor()
except (Exception, psycopg2.Error) as error :
    print ("Error while connecting to PostgreSQL", error)

cleaned_day_data = []

count = 0

with open("historie.txt", "r") as file:
    content = file.readlines()

    for i in content:
        if "e_history_dag" in i:
            dirty_values = i.split("(")[2].replace(")","").replace(";","").strip()
            clean_values = dirty_values.replace("'", "").replace(" ","").replace("00:00:00","").split(",")

            cleaned_day_data.append(clean_values)

    for i in cleaned_day_data:

        if count == 0:
            cursor.execute("INSERT INTO dsmr_stats_daystatistics (day, electricity1, electricity2, electricity1_returned, electricity2_returned, electricity1_cost, electricity2_cost, gas, total_cost, fixed_cost) VALUES ('{}', {}, {}, 0, 0, 1, 1, {}, 0, 0.47)".format(i[0], "0", "0", "0"))
        else:
            electricity1_dif = float(i[1]) - float(cleaned_day_data[count-1][1])
            electricity2_dif = float(i[2]) - float(cleaned_day_data[count-1][2])
            gas_dif = float(i[7]) - float(cleaned_day_data[count-1][7])

            cursor.execute("INSERT INTO dsmr_stats_daystatistics (day, electricity1, electricity2, electricity1_returned, electricity2_returned, electricity1_cost, electricity2_cost, gas, total_cost, fixed_cost) VALUES ('{}', {}, {}, 0, 0, 1, 1, {}, 0, 0.47)".format(i[0], electricity1_dif, electricity2_dif, gas_dif))
        count += 1

connection.commit()
connection.close()
```

@dennissiemensma Bedankt voor je snelle feedback!

closed 2020-11-08 by EYEVO

dennissiemensma, 2020-11-08:
Fijn om te horen! Dank voor je terugkoppeling.