dsmrreader/dsmr-reader #1196: v6.x: Move top-level code to src folder

dennissiemensma, 2020-11-08:
Move code to a new `src` folder, as the current project layout is a bit sloppy. This is a huge BC-break, but it should be manageble.

dennissiemensma, 2021-07-06:
Combine with Docker-only support in future major release.

dennissiemensma, 2024-06-03:
Eerste opzet in https://github.com/dsmrreader/dsmr-reader/tree/work-in-progress/v6

Later maar verder kijken. Hoop kan nog weg, oude deploy scripts etc. En de NL-vertalingen van de Docs (niet de app) ga ik ook droppen. Teveel werk om het bij te houden.

dennissiemensma, 2024-06-04:
Alsnog alles even naar een `src` map hernoemd, zodat het wat beter het verschil in containers reflecteert, waar `/app` redelijk standaard is

dennissiemensma, 2024-06-04:
Hey @xirixiz heb jij nog tips of wensen qua projectstructuur die het wat makkelijker voor ons maken?

Zie de werk-branch qua dir structuur (vooral top-level verplaatsing/splitsing `src` / `docs`) voor een beeld:
- https://github.com/dsmrreader/dsmr-reader/tree/work-in-progress/v6

Ik ga tzt sws nog alle "tools" scripts langs en alles wat in "provisioning" staat. Etc.

----

Het uiteindelijke doel is om straks alleen nog jouw container-setup te gebruiken. Scheelt uiteindelijk hoop gedoe en upgrades, helemaal voor de wat minder technische gebruikers binnen whatever OS ze gebruiken.

Vandaar de mogelijkheid dat ik de aanstaande v6.x release aangrijp om de hele structuur niet meer compatible te hoeven houden met hoe ik het ooit (in 2015) opgezet had. Alles via de container-setup.

In jouw geval zou je dan alleen wat paden moeten aanpassen, maar zover/op dat punt ben ik nog lange na niet hoor.

xirixiz, 2024-06-05:
Hey @dennissiemensma, voor mij is het vrij eenvoudig hoor. Ik denk dat wanneer je de structuur heb gewijzigd, dat ik dan gewoon even wat paden moet aanpassen, en dat het dan al in orde is. Ik denk niet dat de wijzigingen grtote impact zullen hebben. Ik kan wel even een development release maken dat de master/main branch gebruikt om te testen.

dennissiemensma, 2024-06-05:
@xirixiz prima, hoe dan ook bedankt!

dennissiemensma, 2024-06-10:
Ci nu ook aangepast zodat die dezelfde dev-container gebruikt als ik lokaal doe. Was wat gedoe met paden ook en dit is veel makkelijker te onderhouden/reproduceren.

Plus dat het bouwen van de hele docker dev images slechts een halve minuut duurt. Dat was eerder in de CI in de runner zelf veel trager. Helemaal omdat die eerst een DB moet opzetten. Nu gebeurt dat in de (andere) container en is die wat uitgekleed.

En met het droppen van alle andere Python versies straks, duurt het hele ding nog maar anderhalve minuut.

dennissiemensma, 2024-06-10:
Ik sluit deze voor nu, rest van het rework gaat wel in de andere v6.x tickets mee

closed 2024-06-10 by dennissiemensma

dennissiemensma, 2024-06-11:
Voor later:

```
poetry install --without dev --no-root
poetry run /app/manage.py migrate  --noinput
poetry run /app/manage.py flush --no-input
poetry run /app/manage.py collectstatic --noinput
poetry run /app/manage.py dsmr_frontend_clear_cache

```