dsmrreader/dsmr-reader #1197: Geen informatie voor 'current month' op 'dashboard'-pagina

henkjanwithaar, 2020-11-10:
```

DSMR-READER
    App / Python / Database                                                        v4.7 / v3.7.3 / postgresql
    Backend sleep / Datalogger sleep / Retention cleanup                                  1.0s / 0.5s / 4032h
    Telegram latest version read / Parser settings                                               "None" / "3"

DATA
    Telegrams total (est.)                                                                             347632
    Consumption records electricity / gas (est.)                                                94885 / 23810

```

---
Sinds enige tijd (niet zeker sinds welke release) wordt er in het dashboard geen informatie meer getoond met betrekking tot de huidige maand. Op de plaats waar de info in het verleden stond staat nu "The totals for the current month are currently not (yet) available. Check again tomorrow.".

dennissiemensma, 2020-11-10:
Bedankt voor je melding. Worden er wel dagtotalen gegenereerd in het Archief? 

Zo nee, dan kun je kijken of er fouten staan in de log van `dsmr_backend` (https://dsmr-reader.readthedocs.io/en/latest/faq.html#log-files)

henkjanwithaar, 2020-11-10:
Het archief ziet er als vanouds uit en dagtotalen lijken gewoon gegenereerd te worden:

[image]

Het logfile bevat wel warnings; die zijn echter PVOutput gerelateerd.

dennissiemensma, 2020-11-10:
Staan er voor november ook dagen in het Archief of kapt die exact af op eind oktober?

henkjanwithaar, 2020-11-10:
Het laatste: geen dagen voor november in het archief.

dennissiemensma, 2020-11-10:
Dan denk ik dat je het beste debug-logging kan inschakelen: https://dsmr-reader.readthedocs.io/en/latest/faq.html#logging-levels
En dan kijken of je in de log van het backendproces meer ziet.

Ik neem aan dat het dashboard en live grafieken verder wel recente metingen laten zien?

henkjanwithaar, 2020-11-10:
Het dashboard is voor het overige actueel, net als de live grafieken. Ik heb debug logging aangezet en er komt een melding

`DEBUG    SP: Rescheduled "Generate consumption data" to 2020-11-10 21:58:51.436051+01:00 (ETA 0:00:14.992771)`

dennissiemensma, 2020-11-10:
Die debug info is heel generiek en weergeeft geen fouten, alleen wat er gebeurt. 

Ik was vergeten te vermelden dat je moet zoeken naar meldingen die beginnen met:

```
Stats: 
```

Zoals:
```
Stats: No data available
Stats: Waiting for day to pass:
Stats: Found unprocessed readings for:
Stats: Missing consumption data for:
Stats: Waiting for first gas reading on the next day
```

Dat zijn allemaal 'afslagen' bij een conditie dat die iets niet kan doen voor het genereren van de dagstatistieken. Het kan 5 minuten tot een uur duren totdat die het weer probeert.

Om dat te omzeilen kun je op de pagina `/admin/dsmr_backend/scheduledprocess/` zoeken naar "Generate day and hour statistics", die aanklikken en dan de 'plandatum/tijd' in het verleden (of 'nu) zetten.

Bij de "Rescheduled" meldingen in de log schuift die een taak even vooruit. Dat varieert per taak en situatie van een paar seconden tot soms zelfs een week. Door hem handmatig te herplannen probeert die het de eerstvolgende backend run opnieuw en zou je iets in de log moeten zien.

henkjanwithaar, 2020-11-11:
Deze Stats: melding blijft iedere vijf minuten terugkomen:

`[2020-11-11 09:37:50,285] DEBUG    Stats: Waiting for first gas reading on the next day...`

dennissiemensma, 2020-11-11:
Dank voor de update, dat verklaart het inderdaad! Ik zal nog een uitzondering toevoegen voor jouw situatie.

Draai je op docker of heb je een handmatige install gedaan?

henkjanwithaar, 2020-11-11:
> Draai je op docker of heb je een handmatige install gedaan?

Een handmatige install op raspberry pi.

dennissiemensma, 2020-11-11:
Je kunt dan kijken of dit het voor nu voor je oplost:
```
sudo su - dsmr
git fetch
git checkout -b development origin/development
./deploy.sh
```

Daar zit een voorlopige fix in.

henkjanwithaar, 2020-11-12:
Gefixed! Dank Dennis.

[image]

closed 2020-11-12 by henkjanwithaar

dennissiemensma, 2020-11-12:
Oke top! Let op dat nu je wel de testversie gebruikt, wat voor nu prima is. Maar bij de volgende release wil je weer switchen naar de hoofd-versie, waar deze fix tzt dan ook in zit.

Alvast het command voor dan:
```
sudo su - dsmr
git checkout v4
./deploy.sh
```

dennissiemensma, 2020-11-15:
Uitgebracht in [v4.8](https://github.com/dsmrreader/dsmr-reader/releases/tag/v4.8.0). Je kunt dit nu uitvoeren om terug te wisselen naar de hoofdbranch: https://github.com/dsmrreader/dsmr-reader/issues/1197#issuecomment-726290999