dsmrreader/dsmr-reader #1198: Grafieken en data voor gas zijn terug ondanks 0-waarden

henkjanwithaar, 2020-11-10:
```

DSMR-READER
    App / Python / Database                                                        v4.7 / v3.7.3 / postgresql
    Backend sleep / Datalogger sleep / Retention cleanup                                  1.0s / 0.5s / 4032h
    Telegram latest version read / Parser settings                                               "None" / "3"

DATA
    Telegrams total (est.)                                                                             347632
    Consumption records electricity / gas (est.)                                                94885 / 23810

```
---
Sinds release 4.x (dus alweer enige tijd) zijn er onder Live Graphs, Trend, Compare en Statistics weer grafieken voor gas te zien ondanks het feit dat er slechts 0 waarden worden gelezen. Deze bug was na melding #434 in 2018 in een opvolgende release verdwenen. En sinds 4.x weer terug.

dennissiemensma, 2020-11-10:
Bedankt voor je melding. De optie om gas (of teruglevering) te negeren, is verplaatst van https://github.com/dsmrreader/dsmr-reader/issues/434#issuecomment-403007002 naar een optie in de admin. Je kunt een vinkje hiervoor aanzetten in de backend config pagina:
```
/admin/dsmr_backend/backendsettings/1/change/
```

henkjanwithaar, 2020-11-10:
Geregeld. Dank voor je hulp Dennis.

closed 2020-11-10 by henkjanwithaar