dsmrreader/dsmr-reader #1200: Ondersteuning voor graaddagen in vergelijkingen

rvdh, 2020-11-13:
Ik gebruik de optie om data naar mindergas.nl te sturen, en daar kan je verbruik icm graaddagen bekijken. Heel erg waardevol, aangezien ik druk bezig ben met isolerende maatregelen in mijn huis.

Echter kan je daar alleen het gemiddelde verbruik per graaddag zien over het afgelopen jaar, dus ik weet eigenlijk nog niet precies of mijn maatregelen al een beetje helpen.

Dus zou ik graag deze functionaliteit in dsmr-reader zien, sowieso omdat ik dan ook mindergas.nl niet meer nodig heb (staat volgens mij aardig stil qua ontwikkeling). dsmr-reader heeft een veel mooiere interface en dan zit ook alles mooi op één plek.

dennissiemensma, 2020-11-13:
Bedankt voor je verzoek. In een ver verleden (#91) heb ik hier ook naar gekeken en ik kwam tot de conclusie dat het behoorlijk ingewikkeld was om het juist te krijgen, gezien de informatie bij MinderGas zelf: 

https://www.mindergas.nl/degree_days_calculation/explanation

Ik denk dat dat sinds tijd niet veranderd is. Met name de complexiteit en vele factoren maken het al bijna een studie op zichzelf om het nauwkeurig te krijgen. Daarbij zorgt dit ook voor een extra afhankelijkheid voor het extern uitlezen, bijhouden en altijd up-to-date houden van die graaddagen-gegevens.

Wellicht ben je voor nu het meeste geholpen door de graaddagen van dit jaar en vorig jaar te downloaden van: https://www.kwa.nl/diensten/graaddagen-en-koeldagen
Vervolgens kun je voor een maand (of periode), sinds je isolerende maatregelen, de graaddagen voor diezelfde periode van 2020 en 2019 optellen en vervolgens het gasverbruik voor die periodes delen door de bijbehorende graaddagen (zie "Rekenvoorbeeld" in bovenstaande link). 
Je krijg wellicht dan al een goed beeld van een besparing tot nu toe. En later, als je echt een jaar verder bent, geven de graaddagen bij Mindergas zelf misschien dan wel een vollediger beeld.

closed 2020-11-16 by rvdh

dennissiemensma, 2020-11-16:
@rvdh heb je uiteindelijk handmatig de besparing/vergelijking kunnen maken die je zelf zocht of heb je daar nog verdere assistentie bij nodig?

rvdh, 2020-11-17:
@dennissiemensma Nee, dat is me teveel moeite als ik het al zo lees haha. Heel misschien dat ik nog eens een fork maak om zelf nog eens een poging te doen :) Bedankt!

BjornB2, 2020-11-28:
Ik wilde net hetzelfde request plaatsen, dus een +1 voor integratie gewogen graaddagen. Dat is nu de enige reden dat ik naar mindergas exporteer 😊