dsmrreader/dsmr-reader #1201: The following untracked working tree files would be overwritten by merge (ci.txt)

voipmeister, 2020-11-13:
Ik wilde vandaag dsmrreader updaten van 4.3.0 naar latest. Dat lukte niet en bij nadere inspectie zag ik het volgende

```
--- Merging/updating checkout.
Updating f4f605aa..3ba3f326
error: The following untracked working tree files would be overwritten by merge:
	dsmrreader/provisioning/requirements/ci.txt
Please move or remove them before you merge.
Aborting
```

Dit bestand bevatte het volgende:
```
pylama==7.7.1
```

Ik heb met `mv dsmrreader/provisioning/requirements/ci.txt /tmp/` het bestand even 'uit de weg' verplaatst, hierna ging de upgraden wel goed.

Dit issue is ter registratie, om eventueel anderen te helpen.

dennissiemensma, 2020-11-13:
Bedankt voor het melden! Weet je toevallig of je lokaal aanpassingen gedaan had? Anders zou je een volgende keer `git reset HEAD --hard` kunnen proberen.

closed 2020-11-13 by dennissiemensma

voipmeister, 2020-11-17:
Voor zover ik weet heb ik geen aanpassingen lokaal gedaan, je `git reset` suggestie is ook een goede. Voor nu gelukkig al op gelost :-)