dsmrreader/dsmr-reader #1204: Support voor uWSGI mule

Wieter, 2020-11-15:
Goedemiddag,

Om supervisord er helemaal uit te kunnen gooien, heb ik op mijn FreeBSD installatie een kleine, maar zeer essentiële, wijziging doorgevoerd. 

Het backend-proces wordt normaliter door ./manage.py dsmr_backend uitgevoerd (danwel door supervisor, of in tmux of andere shell). Maar om van de uWSGI [mule](https://uwsgi-docs.readthedocs.io/en/latest/Mules.html) gebruik te kunnen maken (en gunicorn ook volgens mij, geen ervaring mee), **moet het backend-proces een [callable](https://djangocentral.com/callable-explained/) zijn**. 
Om niet teveel ingrijpende wijzigingen door te voeren, heb ik `./dsmr_backend/management/commands/dsmr_backend.py` gelaten zoals deze is. Helaas functioneert deze van zichzelf niet als callable want mule `dsmr_backend.management.commands.dsmr_backend:Command` geeft `django.core.exceptions.AppRegistryNotReady: Apps aren't loaded yet.`)

Uiteindelijk is simpelweg `manage.py` aangepast door een backend_mule() functie toe tevoegen:

```diff
diff --git a/manage.py b/manage.py
index 615fa743..941cf411 100755
--- a/manage.py
+++ b/manage.py
@@ -2,9 +2,17 @@
 import os
 import sys
 
-if __name__ == "__main__":
+def backend_mule():
     os.environ.setdefault("DJANGO_SETTINGS_MODULE", "dsmrreader.settings")
 
     from django.core.management import execute_from_command_line
 
+    execute_from_command_line([sys.argv[0],"dsmr_backend"])
+
+
+if __name__ == "__main__":
+    os.environ.setdefault("DJANGO_SETTINGS_MODULE", "dsmrreader.settings")
+
+    from django.core.management import execute_from_command_line
+    print(sys.argv)
     execute_from_command_line(sys.argv)
```

Nu is de inhoud van `/usr/local/etc/uwsgi/vassals/dsmr-reader_uwsgi.ini`:
```ini
# dsmr-reader_uwsgi.ini file
[uwsgi]
uid=dsmr
gid=dsmr

# Django-related settings
# the base directory (full path)
chdir           = /usr/local/www/dsmr/dsmr-reader
# Django's wsgi file
module          = dsmrreader.wsgi:application
# the virtualenv (full path)
home            = /usr/local/www/dsmr/.venv
env = DJANGO_SETTINGS_MODULE=dsmrreader.settings
mule = manage:backend_mule
# process-related settings
# master
master          = true
# maximum number of worker processes
processes       = 5

# The number of worker processes to create.
workers             = 5
  
# Create multiple threads per worker.  This is more memory-efficient if your
# application is thread-safe.
enable-threads      = true
threads             = 5

logto = /var/log/uwsgi/dsmr-reader.log
# the socket (use the full path to be safe
socket          = /var/run/uwsgi_dsmr-reader0.sock
# ... with appropriate permissions - may be needed
#chown-socket = dsmr:dsmr
#chmod-socket    = 664
# clear environment on exit
vacuum          = true
```

Thats it! Met een simpele wijziging in manage.py is dit nu al mogelijk. 

**Mooier zou zijn: een dedicated backend command/callable welke vergelijkbaar als dsmrreader.wsgi werkt,** dus zowel vanuit `./dsmr_backend/management/commands/` (dus manage.py) alsook (u)wsgi applicaties/gunicorn/.. compatible is.

dennissiemensma, 2020-11-15:
Bedankt voor je suggestie! Allereerst is het goed om te weten dat ik **geen** ervaring heb met mules en dat de kennis van wsgi beperkt is tot het inrichten van gunicorn.
Verder is het belangrijk dat `dsmr_backend/management/commands/dsmr_backend.py` (e.a.) **nooit directe callables zijn**. Django verwacht eenmaal deze (file) structuur van management commands, die via `manage.py` bereikbaar zijn. Dat er een losse file per command is, wil niet zeggen dat ze ook direct uitvoerbaar zijn. Dat is meer hoe Django het indeelt.

Ik ben zelf wat terughoudend in het aanpassen van Django-eigen code (lees: `manage.py`) en ik denk dat je hele specifieke wensen hebt, wat waarschijnlijk niet allemaal verwerkt kan worden in een generiek project zoals dit. 
Dat zeggende hebben denk ik wel dat ik je deels kan helpen, want je bent uiteraard vrij om dit project op een andere manier te hosten/draaien dan gedocumenteerd is. Voor mij ligt de grens vooral bij het moeten aanpassen van core/framework functionaliteit, ook om het goed onderhoudbaar te houden in de toekomst.

Mijn suggesties is: Als je het nu al werkend hebt via een alias/aanpassing in `manage.py`, dan staat je denk ik niets in de weg om simpelweg aparte files aan te maken die hetzelfde doen. Je kunt immers in de Mule elke file + callback instellen die je maar wilt.
Ik kan je dan helpen door een bepaalde file toe te voegen aan de gitignore, zodat je eigen variant nooit botst met releases van dit project. Bijvoorbeeld:
```
manage_mule.py
```

dennissiemensma, 2020-11-15:
Ik heb in de release van net `manage_mule.py` voor je toegevoegd in de ignore, zodat je wat vrijer bent in het maken van een eigen variant op de manage.py, zonder dat een eventuele toekomstige release (of Git) in de weg zit.

closed 2020-11-15 by dennissiemensma