dsmrreader/dsmr-reader #1205: Override voor backup tool location

Wieter, 2020-11-15:
Onlangs kwam ik er achter dat pg_dump niet uitgevoerd werd door het backend proces. 
```
[2020-11-15 15:44:49,414] ERROR    (FileNotFoundError) dsmr_backup.services.backup.run errored: [Errno 2] No such file or directory: 'pg_dump': 'pg_dump'
```
na het aanpassen van https://github.com/dsmrreader/dsmr-reader/blob/3ba3f326540bd3ecccfd7c8d4f77c7064cc05678/dsmrreader/config/defaults.py#L26 naar `DSMRREADER_BACKUP_PG_DUMP = '/usr/local/bin/pg_dump'` werkte het prima.

Kan deze parameter bij een volgende release ook als override toegevoegd worden als envvar?

dennissiemensma, 2020-11-15:
Bedankt voor je melding! Ik denk dat je in deze situatie het beste de setting kan overschrijven in `dsmrreader/settings.py`. Dat bestand staat niet onder versiebeheer en je bent sowieso vrij om daar alle elke instelling naar keuze te overschrijven. 

Je kunt dezelfde code-regel uit je comment gewoon toevoegen in eerdergenoemde file onder de imports. Dan zou die leidend moeten zijn. Vergeet uiteraard niet je processen te herladen.

Als alternatief zou je nog kunnen onderzoeken waarom de dsmr-user het `pg_dump` command niet kent als lokale alias. Zoals je misschien kan voorstellen ben ik wat terughoudend om bij elk wiswasje een envvar toe te voegen, ook omdat die gedocumenteerd en onderhouden moet worden. Helemaal wanneer er alternatieven zijn.

Wieter, 2020-11-15:
Ah ja natuurlijk, helemaal gemist wat stom van mij. Ik zet hem gelijk in de settings.py..

Maar om symptoombestrijding te voorkomen, ga ik toch even op onderzoek uit waarom pg_dump niet gevonden wordt vanuit het wsgi proces, en koppel dat hier wel even terug.

Wieter, 2020-11-15:
[Hier](https://stackoverflow.com/questions/5658622/python-subprocess-popen-environment-path) staat het issue exact beschreven. Er lijkt geen path informatie over het base-system beschikbaar te zijn vanuit de virtual-env (en terecht). Ook `/usr/bin/env` geeft niks bruikbaars terug. Waarschijnlijk een security-feature. Dus beste oplossing: alle OS-commands die met Popen uitgevoerd moeten worden met full-path hardcoden in de `settings.py`. Ik denk dat supervisor dit (al dan niet per ongeluk) wel gewoon toestond... 
Ik zal hier rekening mee houden bij het schrijven van een freebsd-specific manual t.z.t. :)

closed 2020-11-15 by Wieter