dsmrreader/dsmr-reader #1207: DSMR Reader (docker) geen data, maar backend draait wel?

juriansluiman, 2020-11-15:
```
DSMR-READER
    App / Python / Database                                                        v4.7 / v3.9.0 / postgresql
    Backend sleep / Datalogger sleep / Retention cleanup                                   1.0s / 5.0s / 672h
    Telegram latest version read / Parser settings                                               "None" / "4"

DATA
    Telegrams total (est.)                                                                                  -
    Consumption records electricity / gas (est.)                                                        - / -

UNRESOLVED ISSUES
    Waiting for the first reading ever                                                                    now
```
Mijn DSMR reader applicatie krijgt geen data binnen en ik heb eigenlijk helemaal geen idee waar de root cause van het probleem zit. De web frontend draait goed, maar geeft dus de melding

> No data found yet. Please make sure either the dsmr_datalogger process running or you supply data using the API. The dsmr_backend process should run as well. 

**Opstelling**
DSMR reader als [container](https://github.com/xirixiz/dsmr-reader-docker) gebruikt. Mijn meter lees ik uit via een P1 kabel, via USB gekoppeld aan een RPi4. Via een udev rule zorg ik dat het device een mapping krijgt naar `/dev/dsmr`. De hele pi wordt geinstalleerd via ansible. Naast de DSMR reader draait een Postgresql in een eigen container, gekoppeld via docker-compose. Mijn meter ondersteunt trouwens DSMR v4.2.

**Instellingen**
De ansible yaml lijkt een beetje op de docker-compose. Met name relevant denk ik dat ik heb geprobeerd de DSMR reader loglevel op debug te  zetten en de datalogger logging op true.

Ik gebruik DSMR reader versie 4.7 en postgresql 12.5.

```yaml
- name: Use docker compose definition to start dsmr-reader
  docker_compose:
    project_name: dsmr
    pull: yes
    definition:
      version: '3'
      services:
        dsmr-db:
          image: "{{ dsmr_db_tag }}"
          container_name: "{{ dsmr_db_name }}"
          restart: unless-stopped
          networks:
            - "{{ dsmr_docker_network }}"
          environment:
            - "PG_TZ={{ dsmr_tz }}"
            - "TZ={{ dsmr_tz }}"
            - "POSTGRES_USER={{ dsmr_db_user }}"
            - "POSTGRES_PASSWORD={{ dsmr_db_password }}"
          volumes:
            - "/etc/localtime:/etc/localtime:ro"
            - "{{ dsmr_install_dir}}/db:/var/lib/postgresql/data"

        dsmr-app:
          image: "{{ dsmr_docker_tag }}"
          container_name: "{{ dsmr_docker_container }}"
          restart: unless-stopped
          networks:
            - "{{ dsmr_docker_network }}"
            - "{{ traefik_docker_network }}"
          environment:
            - "VIRTUAL_HOST=localhost"
            - "DJANGO_TIME_ZONE={{ dsmr_tz }}"
            - "DJANGO_DATABASE_HOST={{ dsmr_db_name }}"
            - "DJANGO_DATABASE_NAME={{ dsmr_db_user }}" # Postgresql default user=db name
            - "DJANGO_DATABASE_USER={{ dsmr_db_user }}"
            - "DJANGO_DATABASE_PASSWORD={{ dsmr_db_password }}"
            - "DATALOGGER_INPUT_METHOD=serial"
            - "DATALOGGER_DEBUG_LOGGING=true"
            - "DSMRREADER_LOGLEVEL=DEBUG"
            - "DSMRREADER_ADMIN_USER={{ dsmr_admin_user }}"
            - "DSMRREADER_ADMIN_PASSWORD={{ dsmr_admin_password }}"
          volumes:
            - "/etc/localtime:/etc/localtime:ro"
            - "/dev/dsmr:/dev/ttyUSB0:ro"
          labels:
            - "traefik.docker.network={{ traefik_docker_network }}"

            - "traefik.http.routers.dsmr.entrypoints=websecure"
            - "traefik.http.routers.dsmr.rule=Host(`{{ dsmr_public_domain }}`)"
            - "traefik.http.routers.dsmr.service=dsmr"
            - "traefik.http.routers.dsmr.tls=true"
            - "traefik.http.routers.dsmr.tls.certresolver=le"

            - "traefik.http.routers.dsmr_http.entrypoints=web"
            - "traefik.http.routers.dsmr_http.rule=Host(`{{ dsmr_public_domain }}`)"
            - "traefik.http.routers.dsmr_http.middlewares=redirect-to-https"

            - "traefik.http.services.dsmr.loadbalancer.server.port=80"
          depends_on:
            - "{{ dsmr_db_name }}"

      networks:
        dsmr:
          external:
            name: "{{ dsmr_docker_network }}"
        web:
          external:
            name: "{{ traefik_docker_network }}"
```
**Logs**
Ik heb al best wat proberen te zoeken, maar ik zie eigenlijk niks geks. Ik heb het idee dat in de container vanuit supervisord er drie processen draaien: backend, datalogger en webinterface. Die datalogger wordt telkens aangeslingerd door supervisord maar sluit ook (exitcode 0) weer na een aantal seconden.

Dit zie je terug in de logs van de docker container en (denk ik) in de log van de backend. De log van de webinterface zie ik niks geks (nginx draait ook, ik krijg gewoon de site te zien). De log van de datalogger blijft wel leeg...

```cmd
$ docker logs dsmr
2020-11-15 19:38:24,104 INFO spawned: 'dsmr_datalogger' with pid 414
2020-11-15 19:38:25,103 INFO success: dsmr_datalogger entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
2020-11-15 19:38:31,638 INFO exited: dsmr_datalogger (exit status 0; expected)
2020-11-15 19:38:32,192 INFO spawned: 'dsmr_datalogger' with pid 415
2020-11-15 19:38:33,234 INFO success: dsmr_datalogger entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
2020-11-15 19:38:39,754 INFO exited: dsmr_datalogger (exit status 0; expected)

$ docker exec -ti dsmr bash
bash-5.0# ls -al /dsmr/logs/
total 16
drwxr-xr-x    2 root     root          4096 Nov  9 21:44 .
drwxr-xr-x    1 root     root          4096 Nov  9 21:45 ..
-rw-r--r--    1 root     root             4 Nov  9 21:44 .dummy
bash-5.0# ls -al /var/log/supervisor/
total 4056
drwxr-xr-x    2 root     root          4096 Nov 15 16:37 .
drwxr-xr-x    1 root     root          4096 Nov 15 16:37 ..
-rw-r--r--    1 root     root       4134146 Nov 15 19:40 dsmr_backend.log
-rw-r--r--    1 root     root             0 Nov 15 16:37 dsmr_datalogger.log
bash-5.0# tail -n 10 /var/log/supervisor/dsmr_backend.log 
[2020-11-15 19:41:40,525] DEBUG    SP: Rescheduled "Generate consumption data" to 2020-11-15 20:41:41.519502+01:00 (ETA 0:00:00.994342)
[2020-11-15 19:41:40,560] DEBUG    dsmr_backend.management.commands.dsmr_backend: Sleeping 1.0s
[2020-11-15 19:41:41,566] DEBUG    SP: 1 backend service(s) ready to run
[2020-11-15 19:41:41,567] DEBUG    SP: Running "Generate consumption data" (dsmr_consumption.services.run)
[2020-11-15 19:41:41,578] DEBUG    SP: Rescheduled "Generate consumption data" to 2020-11-15 20:41:42.575698+01:00 (ETA 0:00:00.997015)
[2020-11-15 19:41:41,594] DEBUG    dsmr_backend.management.commands.dsmr_backend: Sleeping 1.0s
[2020-11-15 19:41:42,599] DEBUG    SP: 1 backend service(s) ready to run
[2020-11-15 19:41:42,599] DEBUG    SP: Running "Generate consumption data" (dsmr_consumption.services.run)
[2020-11-15 19:41:42,611] DEBUG    SP: Rescheduled "Generate consumption data" to 2020-11-15 20:41:43.608007+01:00 (ETA 0:00:00.997031)
[2020-11-15 19:41:42,627] DEBUG    dsmr_backend.management.commands.dsmr_backend: Sleeping 1.0s
bash-5.0# tail -n 10 /var/log/supervisor/dsmr_datalogger.log 
bash-5.0# tail -n 10 /var/log/supervisor/dsmr_webinterface.log 
[2020-11-15 18:47:03 +0000] [20] [INFO] Starting gunicorn 20.0.4
[2020-11-15 18:47:03 +0000] [20] [INFO] Listening at: unix:/var/tmp/gunicorn--dsmr_webinterface.socket (20)
[2020-11-15 18:47:03 +0000] [20] [INFO] Using worker: sync
[2020-11-15 18:47:03 +0000] [26] [INFO] Booting worker with pid: 26
[2020-11-15 19:29:41 +0000] [26] [INFO] Autorestarting worker after current request.
[2020-11-15 19:29:41 +0000] [26] [INFO] Worker exiting (pid: 26)
[2020-11-15 19:29:42 +0000] [348] [INFO] Booting worker with pid: 348
[2020-11-15 19:40:09 +0000] [348] [INFO] Autorestarting worker after current request.
[2020-11-15 19:40:09 +0000] [348] [INFO] Worker exiting (pid: 348)
[2020-11-15 19:40:09 +0000] [428] [INFO] Booting worker with pid: 428
bash-5.0# 
```
Omdat ik verder de applicatie (nog) niet goed ken weet ik eigenlijk niet waar ik verder moet zoeken. De website melding zegt dat ik moet checken dat de backend en datalogger wel draaien, maar in de container laat top zien dat beide processen gewoon hun werk doen. Waar kan ik verder zoeken?

dennissiemensma, 2020-11-15:
Bedankt voor je melding. Je zou nog kunnen zoeken op een wildcard van de logs, want Supervisor er zet soms een suffix achter en dan lijkt de hoofdlog leeg:
```
tail -f /var/log/supervisor/dsmr_datalogger*
```

Ter illustratie bij mijzelf, zie vooral de naam van de logfile:
```
sudo tail -f /var/log/supervisor/dsmr_datalogger*
==> /var/log/supervisor/dsmr_datalogger.log <==
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging

==> /var/log/supervisor/dsmr_datalogger-stderr---supervisor-JIQLFI.log <==
1-0:21.7.0(00.131*kW)
1-0:41.7.0(00.146*kW)
1-0:61.7.0(00.000*kW)
1-0:22.7.0(00.000*kW)
1-0:42.7.0(00.000*kW)
1-0:62.7.0(00.000*kW)
0-1:24.1.0(003)
0-1:96.1.0(xxxx)
0-1:24.2.1(201115200000W)(03539.821*m3)
!A687
[2020-11-15 20:59:27,691] DEBUG    dsmr_datalogger.management.commands.dsmr_datalogger: Sleeping 2.0s
[2020-11-15 20:59:29,699] INFO     [2020-11-15 20:59:29.699370] Opening connection "/dev/ttyUSB0" using options: {'baudrate': 115200, 'bytesize': 8, 'parity': 'N', 'stopbits': 1, 'xonxoff': 1, 'rtscts': 0}
[2020-11-15 20:59:30,040] DEBUG    [2020-11-15 20:59:30.040744] Read 0 Byte(s)
[2020-11-15 20:59:30,375] DEBUG    [2020-11-15 20:59:30.375129] Read 0 Byte(s)
[2020-11-15 20:59:30,709] DEBUG    [2020-11-15 20:59:30.709480] Read 0 Byte(s)
[2020-11-15 20:59:31,043] DEBUG    [2020-11-15 20:59:31.043871] Read 0 Byte(s)
[2020-11-15 20:59:31,378] DEBUG    [2020-11-15 20:59:31.378215] Read 0 Byte(s)
[2020-11-15 20:59:31,712] DEBUG    [2020-11-15 20:59:31.712205] Read 66 Byte(s)
[2020-11-15 20:59:32,046] DEBUG    [2020-11-15 20:59:32.046239] Read 552 Byte(s)
[2020-11-15 20:59:32,381] DEBUG    [2020-11-15 20:59:32.381044] Read 251 Byte(s)
[2020-11-15 20:59:32,391] DEBUG    Received telegram:
/XMX5LGBBFFB231117727

1-3:0.2.8(42)
0-0:1.0.0(201115205918W)
0-0:96.1.1(xxx)
1-0:1.8.1(004654.879*kWh)
1-0:2.8.1(000000.000*kWh)
1-0:1.8.2(004754.069*kWh)
1-0:2.8.2(000000.000*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(00.283*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00007)
0-0:96.7.9(00002)
1-0:99.97.0(2)(0-0:96.7.19)(170615111202S)(0000010014*s)(170120102108W)(0000001572*s)
1-0:32.32.0(00002)
1-0:52.32.0(00002)
1-0:72.32.0(00000)
1-0:32.36.0(00000)
1-0:52.36.0(00000)
1-0:72.36.0(00000)
0-0:96.13.1()
0-0:96.13.0()
1-0:31.7.0(001*A)
1-0:51.7.0(001*A)
1-0:71.7.0(000*A)
1-0:21.7.0(00.131*kW)
1-0:41.7.0(00.151*kW)
1-0:61.7.0(00.000*kW)
1-0:22.7.0(00.000*kW)
1-0:42.7.0(00.000*kW)
1-0:62.7.0(00.000*kW)
0-1:24.1.0(003)
0-1:96.1.0(xxxx)
0-1:24.2.1(201115200000W)(03539.821*m3)
!248A
```

Wellicht zie je meer als je zelf ook een tail doet op de wildcard?

juriansluiman, 2020-11-15:
```cmd
bash-5.0# ls -l /var/log/
total 8
drwxr-xr-x    1 nginx    nginx         4096 Nov  9 21:49 nginx
drwxr-xr-x    2 root     root          4096 Nov 15 19:48 supervisor
bash-5.0# ls -l /var/log/supervisor/
total 476
-rw-r--r--    1 root     root        477415 Nov 15 20:08 dsmr_backend.log
-rw-r--r--    1 root     root             0 Nov 15 19:48 dsmr_datalogger.log
-rw-r--r--    1 root     root           300 Nov 15 19:48 dsmr_webinterface.log
bash-5.0# ls -l /var/log/nginx/
total 0
lrwxrwxrwx    1 root     root            11 Nov  9 21:49 access.log -> /dev/stdout
lrwxrwxrwx    1 root     root            11 Nov  9 21:49 error.log -> /dev/stderr
bash-5.0# 
```
:clown_face: 

Ik weet ook niet zo goed of ik het hier moet zoeken of eerder bij @xirixiz in https://github.com/xirixiz/dsmr-reader-docker

dennissiemensma, 2020-11-15:
Dat vind ik zelf ook lastig te zeggen. Het is een wat grijs gebied. Het kan zijn dat DSMR-reader tegen iets aanloopt wat niet zichtbaar is zonder die logs. Het kan ook zijn dat die datalogger uberhaupt niets doet in Docker. Het zal ergens tussenin zitten.

Ik denk dat het sowieso goed is om je issue ook aan te maken bij Xirixiz, maar wellicht kun je ook nog even zoeken in eerder aangemaakt/gesloten issues daar. Grote kans dat iemand tegen hetzelfde aan is gelopen.

----

Vanuit DSMR-reader gezien zijn veel voorkomende oorzaken:
- Verbinding werkt uberhaupt niet: [test het eerst even met `cu`](https://dsmr-reader.readthedocs.io/en/latest/installation.html#optional-your-first-reading)
- Seriele verbinding wordt gedeeld met een andere applicatie en er kan maar 1 tegelijkertijd lezen
- Verkeerde seriele instellingen. (bij een NL-meter met DSMR-v4 kun je echter weinig fout instellen in DSMR-reader)
- Vage virtualisatie issues met de seriele verbinding, maar dat ging vooral om specifieke uitzonderingen met draaien in VIrtualBox op een MacOS oid

juriansluiman, 2020-11-15:
@dennissiemensma met cu op de rpi krijg ik wel de telegrammen binnen. Binnen de alpine docker container krijg ik cu niet aan de praat, maar `tail -f /dev/ttyUSB0` zegt al permission denied. Ik ga nog even verder puzzelen maar ik verwacht dat ik hier het issue kan sluiten en bij Xirixiz zal openen. Morgen weer even verder mee...

dennissiemensma, 2020-11-15:
Bedankt voor je toelichting. Je zou kunnen kijken of de gebruiker wel toegevoegd is aan de `dialout` groep. Bij niet-Docker installaties is dit de instructie:
```
sudo usermod -a -G dialout dsmr
```

Ik zie bij Xirixiz [dezelfde instructie](https://github.com/xirixiz/dsmr-reader-docker#dsmr-reader---docker). Grote kans dat dat het is:

> You should first add the user you run Docker with on your host file system to the dialout group:
> 
> ```
> sudo usermod -aG dialout $(whoami)
> ```

juriansluiman, 2020-11-15:
Ik had de device mapping als een volume mapping gemaakt :sob: 

Dit:
```yaml
volumes:
  - "/etc/localtime:/etc/localtime:ro"
  - "/dev/dsmr:/dev/ttyUSB0"
```

Naar dit 
```yaml
volumes:
  - "/etc/localtime:/etc/localtime:ro"
devices:
  - "/dev/dsmr:/dev/ttyUSB0"
```

Kennelijk werken device mappings toch iets anders... dank voor de mentale ondersteuning ;)

closed 2020-11-15 by juriansluiman