dsmrreader/dsmr-reader #1208: Live Grafieken worden niet meer gevuld

hjkoster400d, 2020-11-16:
Installatie heb ik destijds met de hand gedaan.

Sinds de upgrade naar 4.7 (verwacht ik) wordt er geen data meer in de de live grafieken getoond. Heb hier destijds niet veel aandacht aan besteed, maar na de update naar 4.8 bleven de grafieken nog steeds leeeg. Hieronder de dump:

DSMR-READER
    App / Python / Database                                                        v4.8 / v3.7.3 / postgresql
    Backend sleep / Datalogger sleep / Retention cleanup                                   1.0s / 0.5s / 168h
    Telegram latest version read / Parser settings                                                 "50" / "4"

DATA
    Telegrams total (est.)                                                                            2878060
    Consumption records electricity / gas (est.)                                                74569 / 42368

UNRESOLVED ISSUES
    Er zijn te veel uitgaande MQTT-berichten die wachten op verzending                                      nu
    Database groeit gestaag: 13 GB, overweeg dataopschoning (indien nog niet ingeschakeld)                     een seconde geleden
    Proces loopt achter op schema: Automatic update checker                           2 dagen, 18 uur geleden
    Proces loopt achter op schema: Retention data rotation                             4 dagen, 5 uur geleden
    Proces loopt achter op schema: Backup per email                                   6 dagen, 18 uur geleden
    Proces loopt achter op schema: Create daily backup                                         1 week geleden
    Proces loopt achter op schema: Generate day and hour statistics                            1 week geleden
    Proces loopt achter op schema: Generate consumption data                                   1 week geleden
    Proces loopt achter op schema: Weather update                                              1 week geleden
    Te veel onverwerkte telegrammen: 672024                                                    1 week geleden
    Dagstatistieken lopen achter                                                        1 week, 1 dag geleden

POSTGRESQL SIZE OF LARGEST TABLES (> 500 MB)
    public.dsmr_datalogger_dsmrreading                                                                  12 GB 

En het plaatje:
[image]

Enig idee waar dit door zou kunnen komen?

dennissiemensma, 2020-11-16:
Bedankt voor je melding! Als het goed is staan op de About/Over-pagina behoorlijk wat waarschuwingen met meer informatie.

De korte versie is dat je de database heel hard gegroeid is en momenteel nog steeds groeit, ondanks dat je dataopschoning hebt ingesteld. Afhankelijk van je hardware kan dat een probleem zijn, helemaal als dit op een Pi draait.

- Allereerst is het belangrijk om de datatoevoer wat te beperken. Wil je de datalogger sleep op ten minste 10 seconden (of hoger) zetten?
- Wat voor hardware gebruik je? Een RaspberryPi met SD-kaart of draai je dit op een ander systeem?
- Zorg ervoor dat je alvast een 'backup' maakt van je dag- en uurtotalen tot nu toe. Dat kan via het 'Export' item op de Configuratie-pagina, of direct op de URL `/export`.

hjkoster400d, 2020-11-17:
Bedankt voor je snelle reactie.
- Heb de datalogger sleep op 10 seconden gezet.
- Draait in een VM op een Intel NUC 6e generatie met SSD
- Backups heb ik gemaakt

1. Ik zie inderdaad dat er veel processes achterlopen, wat kan ik daaraan doen?
2. Database retentie heb ik wel aangezet, is er een manier om de database eenmalig op te schonen wellicht?
3. De metingen worden met MQTT doorgestuurd naar mijn Home assistant, maar ik merk dat die vaak stopt met versturen. Zal ik daar even een nieuw issue voor aanmaken?

Wederom bedankt voor je hulp

dennissiemensma, 2020-11-17:
Als deze hardware wat sneller is dan een Pi, kun je nog de normale weg proberen. In principe trapt het `dsmr_backend` proces o.a. `Retention data rotation` af, die oude data opschoont. Wellicht dat dat te langzaam is geworden inmiddels.

Allereerst wil je denk ik voor de zekerheid ook nog een extra backup maken van je data. Dit is een compacte backup met alleen alle dag- en uurtotalen (naast de CSV's die je al had). Daar zou iig alles in moeten zitten tot een week terug (aan je meldingen te zien).

Het maken van de compacte backup kan met:
```
sudo su - dsmr
./manage.py dsmr_backup_create --compact
```

Als het goed is zie je het pad in de output, bij mij:
```
[2020-11-17 17:12:49,217] INFO      - Creating new partial backup: /home/dennis/PycharmProjects/dsmr-reader/backups/manually/dsmrreader-postgresql-partial-backup-2020-11-17.sql
[2020-11-17 17:12:49,351] DEBUG     - Backup exit code: 0
[2020-11-17 17:12:49,743] INFO      - Created and compressed statistics backup: /home/dennis/PycharmProjects/dsmr-reader/backups/manually/dsmrreader-postgresql-partial-backup-2020-11-17.sql.gz
```

Die file kun je dan veiligstellen. 

----

Als je dat gedaan hebt, dan kun je proberen om alle oude, verwerkte metingen op te schonen (vermoedelijk 99% van je data). Ik weet niet of dat lukt en hoe lang het duurt, maar het is de snelste manier denk ik:

```
sudo su - postgres
psql dsmrreader

DELETE FROM dsmr_datalogger_dsmrreading WHERE processed = true;
```

Looptijd is onbekend en hangt af van je hardware.

dennissiemensma, 2020-11-17:
Het kan ook nog helpen om het dsmr_backend proces (niet de datalogger) tijdelijk uit te zetten. En na het opschonen van alle data een vacuumdb te doen.

hjkoster400d, 2020-11-17:
vacuumdb is voor mij onbekend, op welke tabellen zou je dat moeten doen? en wat wordt dan de syntax?

dennissiemensma, 2020-11-17:
Vacuumdb heeft overigens alleen zin nadat alle metingen opgeschoond zijn. Meer info: https://github.com/dsmrreader/dsmr-reader/issues/1149#issuecomment-710636181

hjkoster400d, 2020-11-17:
Heb de backend stop gezet, de delete actie gedaan en een vacuum uitgevoerd. Daarna de backend weer gestart en heb een stuk minder meldingen. Live grafieken doen het nog niet trouwens, misschien duurt dat even?

Deze staan er nu nog wel:
DSMR-READER
    App / Python / Database                                                        v4.8 / v3.7.3 / postgresql
    Backend sleep / Datalogger sleep / Retention cleanup                                  1.0s / 10.0s / 168h
    Telegram latest version read / Parser settings                                                 "50" / "4"

DATA
    Telegrams total (est.)                                                                            2878060
    Consumption records electricity / gas (est.)                                                74569 / 42368

UNRESOLVED ISSUES
    Er zijn te veel uitgaande MQTT-berichten die wachten op verzending                                      nu
    Database groeit gestaag: 13 GB, overweeg dataopschoning (indien nog niet ingeschakeld)                                      nu
    Proces loopt achter op schema: Retention data rotation                                      9 uur geleden
    Te veel onverwerkte telegrammen: 740820                                             1 week, 1 dag geleden
    Dagstatistieken lopen achter                                                      1 week, 2 dagen geleden

POSTGRESQL SIZE OF LARGEST TABLES (> 500 MB)
    public.dsmr_datalogger_dsmrreading                                                                  12 GB

hjkoster400d, 2020-11-17:
Onverwerkte telegrammen lopen goed terug, heb het vermoeden dat daar het issue ook zit. Dus wacht het nog even af. database is nog wel steeds 13Gb blijkbaar.....

dennissiemensma, 2020-11-17:
Oke dat verwijderen is dan lekker vlot gegaan! Je zou nog een count kunnen doen op het totaal metingen in diezelfde tabel. Of een vacuumdb draaien met -z als extra argument. Ik dacht dat dat analyze was, zodat die ook forceert interne telletjes bij te werken. Ik zie namelijk dat de teller bij DATA in je eerste en laatste bericht gelijk zijn. Het zijn schattingen ivm de snelheid, maar ik had wel een verschil verwacht.

hjkoster400d, 2020-11-18:
Live grafieken worden weer gevuld.
Database is geslonken tot 6Gb, geeft daar nog steeds een melding over. Kan ik daar nog iets aan doen?

[image]

dennissiemensma, 2020-11-18:
Ik denk dat je het een klein weekje moet geven en daarna nog een laatste keer `vacuumdb` doen. 

Je hebt namelijk gisteren de datalogger sleep hoger gezet, waardoor je sindsdien geen 60 telegrammen maar 6 telegrammen per minuut binnenkrijgt. Samen met de retentie van een week, verwacht ik dat de volle dagen (tot gisteren, met 60 telegrammen per minuut) pas over 6 dagen helemaal geroteerd zijn.
Dat zeggende hebben lijkt 6GB me wat veel voor slechts een week aan data, zelfs met elke seconde. Dus los van bovenstaande kun je met de queries uit het volgende issue kijken wat de voortgang is van de 'natuurlijke' dataopschoning. Je hebt welliswaar de telegrammen opgeschoond, maar er zijn nog andere tabellen die ook een deel van die data hebben. Ik zou dat trouwens lekker automatisch laten opschonen, die achterstand haalt die vanzelf in. Zie hieronder dus voor hoe je de voortgang kan zien.

Queries (en toelichting op bovenstaand verhaal): https://github.com/dsmrreader/dsmr-reader/issues/1071#issuecomment-670690844

dennissiemensma, 2020-11-18:
Ter referentie, als alle opschoning bij is krijg je ongeveer zoiets. Waarbij alleen de recente periode die niet opgeschoond is (bij mij ook een week) significant meer data heeft dan de rest:

```
        month          | total 
------------------------+-------
 2020-11-01 00:00:00+00 | 63611
 2020-10-01 00:00:00+01 |  1490
 2020-09-01 00:00:00+01 |  1440
 2020-08-01 00:00:00+01 |  1488
 2020-07-01 00:00:00+01 |  1488
 2020-06-01 00:00:00+01 |  1440
 2020-05-01 00:00:00+01 |  1488
 2020-04-01 00:00:00+01 |  1440
 2020-03-01 00:00:00+00 |  1486
 2020-02-01 00:00:00+00 |  1392
 2020-01-01 00:00:00+00 |  1488
 2019-12-01 00:00:00+00 |  1488
 2019-11-01 00:00:00+00 |  1440
 2019-10-01 00:00:00+01 |  1490
 2019-09-01 00:00:00+01 |  1076
 2019-08-01 00:00:00+01 |  1488
 2019-07-01 00:00:00+01 |  1488
 2019-06-01 00:00:00+01 |  1440
 2019-05-01 00:00:00+01 |  1488
 2019-04-01 00:00:00+01 |  1440
 2019-03-01 00:00:00+00 |  1486
 2019-02-01 00:00:00+00 |  1344
 2019-01-01 00:00:00+00 |  1488
 2018-12-01 00:00:00+00 |  1488
 2018-11-01 00:00:00+00 |  1440
 2018-10-01 00:00:00+01 |  1490
 2018-09-01 00:00:00+01 |  1440
 2018-08-01 00:00:00+01 |  1488
 2018-07-01 00:00:00+01 |  1488
 2018-06-01 00:00:00+01 |  1374
 2018-05-01 00:00:00+01 |  1462
 2018-04-01 00:00:00+01 |   950
 2018-03-01 00:00:00+00 |  1364
 2018-02-01 00:00:00+00 |  1344
 2018-01-01 00:00:00+00 |  1488
 2017-12-01 00:00:00+00 |  1488
 2017-11-01 00:00:00+00 |  1440
 2017-10-01 00:00:00+01 |  1401
 2017-09-01 00:00:00+01 |  1406
 2017-08-01 00:00:00+01 |  1436
 2017-07-01 00:00:00+01 |  1414
 2017-06-01 00:00:00+01 |  1294
 2017-05-01 00:00:00+01 |  1488
 2017-04-01 00:00:00+01 |  1440
 2017-03-01 00:00:00+00 |  1486
 2017-02-01 00:00:00+00 |  1344
 2017-01-01 00:00:00+00 |  1488
 2016-12-01 00:00:00+00 |  1488
 2016-11-01 00:00:00+00 |  1440
 2016-10-01 00:00:00+01 |  1490
 2016-09-01 00:00:00+01 |  1440
 2016-08-01 00:00:00+01 |  1488
 2016-07-01 00:00:00+01 |  1488
 2016-06-01 00:00:00+01 |  1440
 2016-05-01 00:00:00+01 |  1488
 2016-04-01 00:00:00+01 |  1440
 2016-03-01 00:00:00+00 |  1484
 2016-02-01 00:00:00+00 |  1392
 2016-01-01 00:00:00+00 |  1488
 2015-12-01 00:00:00+00 |   966
```

dennissiemensma, 2020-11-23:
@hjkoster400d heb je een update voor bovenstaande? Zie je inmiddels het aantal opgeslagen metingen teruglopen en blijven de grafieken nu goed? 

En hoever lijkt de dataopschoning? Als die voor je gevoel helemaal bij is zou je nog een laatste `vacuumdb` kunnen doen. De grootte zou dan significant minder moeten zijn.

hjkoster400d, 2020-11-24:
Database is zeker een stuk kleiner geworden, heb zojuist nog een vacuumdb uitgevoerd en de database is nu 6031MB.

DSMR-READER
    App / Python / Database                                                        v4.8 / v3.7.3 / postgresql
    Backend sleep / Datalogger sleep / Retention cleanup                                  1.0s / 10.0s / 168h
    Telegram latest version read / Parser settings                                                 "50" / "4"

DATA
    Telegrams total (est.)                                                                              58149
    Consumption records electricity / gas (est.)                                                46835 / 38071

UNRESOLVED ISSUES
    Database groeit gestaag: 6031 MB, overweeg dataopschoning (indien nog niet ingeschakeld)                                      nu

POSTGRESQL SIZE OF LARGEST TABLES (> 500 MB)
    public.dsmr_datalogger_dsmrreading                                                                5609 MB

De database wordt wel telkens heel iets groter, hoe groot zou de database normaliter moeten zijn? Zijn er misschien andere tabellen die nog aandacht behoeven?

dennissiemensma, 2020-11-24:
Bedankt voor de update. Als de metingen goed opgeschoond zijn, begrijp ik niet zo goed waar de grootte van de database nog in zit. 

Probeer deze variant ook even of dat nog wat uit maakt:

```
sudo su - postgres
vacuumdb -f -v -z -d dsmrreader
```

----

En wat geven al deze queries voor output?
```
sudo su - postgres
psql
```

### 1
```
SELECT d.datname as Name,  pg_catalog.pg_get_userbyid(d.datdba) as Owner,
    CASE WHEN pg_catalog.has_database_privilege(d.datname, 'CONNECT')
        THEN pg_catalog.pg_size_pretty(pg_catalog.pg_database_size(d.datname))
        ELSE 'No Access'
    END as Size
FROM pg_catalog.pg_database d
    order by
    CASE WHEN pg_catalog.has_database_privilege(d.datname, 'CONNECT')
        THEN pg_catalog.pg_database_size(d.datname)
        ELSE NULL
    END desc -- nulls first
    LIMIT 20;
```

----

Even weer CTRL + D en dan:

```
psql dsmrreader
```

### 2
```
SELECT nspname || '.' || relname AS "relation",
    pg_size_pretty(pg_total_relation_size(C.oid)) AS "total_size"
  FROM pg_class C
  LEFT JOIN pg_namespace N ON (N.oid = C.relnamespace)
  WHERE nspname NOT IN ('pg_catalog', 'information_schema')
    AND C.relkind <> 'i'
    AND nspname !~ '^pg_toast'
  ORDER BY pg_total_relation_size(C.oid) DESC
  LIMIT 20;
```

### 3
```
select count(id) from dsmr_datalogger_dsmrreading;
```
 

### 4
```
select count(id) from dsmr_consumption_electricityconsumption;
```

### 5
```
select count(id) from dsmr_consumption_gasconsumption;
```

### 6
```
select date_trunc('month', timestamp) as month, count(id) as total from dsmr_datalogger_dsmrreading group by month order by month desc;
```

hjkoster400d, 2020-11-25:
Dank weer voor je hulp, hieronder de resultaten:

**1.**
`
    name    |   owner    |  size
------------+------------+---------
 dsmrreader | dsmrreader | 43 MB
 postgres   | postgres   | 7693 kB
 template1  | postgres   | 7693 kB
 template0  | postgres   | 7693 kB
`

**2**
`                     relation                     | total_size
--------------------------------------------------+------------
 public.dsmr_consumption_electricityconsumption   | 16 MB
 public.dsmr_datalogger_dsmrreading               | 11 MB
 public.dsmr_consumption_gasconsumption           | 3976 kB
 public.dsmr_stats_hourstatistics                 | 2072 kB
 public.dsmr_weather_temperaturereading           | 1736 kB
 public.dsmr_stats_daystatistics                  | 208 kB
 public.dsmr_frontend_notification                | 152 kB
 public.django_admin_log                          | 88 kB
 public.dsmr_backend_scheduledprocess             | 88 kB
 public.django_session                            | 72 kB
 public.auth_user                                 | 64 kB
 public.dsmr_consumption_energysupplierprice      | 56 kB
 public.auth_permission                           | 56 kB
 public.django_migrations                         | 48 kB
 public.dsmr_frontend_sortedgraph                 | 40 kB
 public.django_content_type                       | 40 kB
 public.dsmr_influxdb_influxdbintegrationsettings | 32 kB
 public.dsmr_mqtt_jsontelegrammqttsettings        | 32 kB
 public.auth_group_permissions                    | 32 kB
 public.dsmr_mqtt_message                         | 32 kB
`

**3**
`
 count
-------
 55501
(1 row)
`

**4**
`
 count
-------
 46440
(1 row)
`

**5**
`
 count
-------
 38360
(1 row)
`

**6**
`
         month          | total
------------------------+-------
 2020-11-01 00:00:00-04 | 55513
(1 row)
`

hjkoster400d, 2020-11-25:
Alles lijkt weer in order na de laatste vacuumdb. 

[image]

dennissiemensma, 2020-11-25:
Fijn om te horen! 43MB klinkt stukken aannemelijker met de aantallen die je nu in je DB hebt.

closed 2020-11-25 by dennissiemensma