dsmrreader/dsmr-reader #1210: Dropbox enhanced permissions

bobveringa, 2020-11-20:
Hi,
Ik kreeg een geautomatiseerde melding van dropbox dat de DMSR Reader app misschien aangepast moest worden om de nieuwe refresh tokens te ondersteunen. Omdat long-lived access tokens vanaf 30-09-2021 niet meer ondersteund worden. 
Ik kon geen bestaande issues vinden waarin dit werd vermeld. 

Worden refresh tokens als ondersteund in DMSR Reader en zo ja vanaf welke versie?

dennissiemensma, 2020-11-21:
Bedankt voor je melding. Ik heb gisteren hetzelfde mailtje ontvangen, dus ik ga daar naar kijken.

Het is goed om te weten dat de huidige manier waarop DSMR-reader Dropbox gebruikt niet de bedoeling is, maar ik daar destijds geen workaround voor had. 
Dit heeft er vooral mee te maken dat DSMR-reader geen vaste host of locatie heeft. Ik zie dat er inmiddels op dat gebied ook ontwikkelingen zijn geweest, dus misschien dat ik nu een echt integratie kan maken. Zodat bij een volgende release je gewoon via de webinterface van DSMR-reader direct naar Dropbox wordt geleid en de vraag krijgt om permissie te geven tot toegang, ipv de huidige manier die 'misbruikt' dat iedereen zijn eigen app registreert en oneindig een developer-token gebruikt.

dennissiemensma, 2020-11-21:
Qua techniek: https://www.dropbox.com/lp/developers/reference/oauth-guide?oref=e

Interessant voor mij is om te kijken naar PKCE, een techniek die volgens mij precies bedoeld is om bovenstaand probleem op te lossen. Het is nog even onder voorbehoud of dat inderdaad de oplossing is.

dennissiemensma, 2020-11-21:
Ik zie dat het ook mogelijk is om een integratie te doen zonder redirect terug. Dat is nog een mooie fallback als het teveel gedoe is om de redirect werkend te krijgen. 
De gebruiker moet dan wel de code zelf invoeren in DSMR-reader, maar dat is vergeleken met hoe het nu moet, absoluut prima.

[image]

[image]

Ik vind zelf de uitleg op het eerste scherm ook nog wat slecht uitgewerkt. Het is een superlange zin, met aan het eind het belangrijkste: **DSMR-reader heeft alleen toegang tot een eigen map binnen ieders dropbox en kan alleen daar bestanden lezen en aanmaken**.

Ik kan me voorstellen dat er wat gebruikers zijn die afhaken door die onhandige formulering van Dropbox. Wellicht dat ik ze vraag of ze willen overwegen het puntsgewijs voor gebruikers te tonen. Volgens mij doet/deed Facebook dat ook altijd en is veel overzichtelijker. En anders kan ik wel dergelijke screenshots opnemen in de handleiding oid.

dennissiemensma, 2020-11-21:
Verder is het jammer dat er geen losse permissie is om bijvoorbeeld **niet de profielinfo** te hoeven inzien, wat dus wel gevraagd wordt op het eerste screenshot hierboven. Want ik kan dat als Developer ook niet uitzetten of beperken:

_(de tweede checkbox staat geforceerd aan)_
[image]

Het enige wat DSMR-reader nodig heeft is hoeveel beschikbare schijfruimte er nog over is. Straks heeft die ook toegang tot persoonsgegevens blijkbaar en daar zit ik eigenlijk niet op te wachten. 
Dit is tevens een puntje wat ik bij Dropbox kan leggen, al verwacht ik niet dat ze daar op korte termijn iets aan kunnen of gaan doen.

dennissiemensma, 2020-11-21:
Ik zie dat een redirect sowieso niet gaat werken, want daarvoor wordt (terecht) HTTPS vereist en dat is in de meeste installaties van DSMR-reader geen optie vermoed ik. Ook omdat SSL certificaten met hosts of IP's van interne netwerken geen optie is helaas. 

```The URI that the Dropbox server will redirect the user to after the
            user finishes authorizing your app.  This URI must be HTTPS-based and pre-registered
            with the Dropbox servers, though localhost URIs are allowed without pre-registration and
            can be either HTTP or HTTPS.
```

Wellicht dat ik hier 4 a 5 jaar geleden ook tegenaan liep en dat ik daarom koos voor de huidige oplossing. Dus dan wordt het een kwestie van kijken of ik het invoeren van de Access Code werkend krijg.

Calimerorulez, 2022-01-12:
Ik heb geprobeerd mijn nieuwe DSMR-reader installatie aan Dropbox te knopen, maar de backup mislukt nu inderdaad structureel, omdat er geen long lived tokens meer gebruikt worden.

Op een andere, oudere installatie, op een andere locatie, werkt het nog wel, dat is een long lived token, die verlopen blijkbaar niet...

dennissiemensma, 2022-01-13:
Bedankt voor je melding. Ik zal hier proberen tzt naar te kijken. Op dit moment heb ik helaas nog geen directe oplossing voor je, al was ik destijds een heel eind gekomen kan ik me herinneren.

dennissiemensma, 2022-01-23:
Het is nog wel de vraag hoelang ik deze manier kan ondersteunen. In de OAuth Guide van Dropbox staat nu ook al expliciet:

> **Do not instruct your users to register their own Dropbox application to use your app** - you just need to register your app once. Your end-users will connect to that app via the OAuth flow.

Alleen is dat voor DSMR-reader geen optie, omdat iedereen een andere URL heeft want de "server" is waar DSMR-reader draait, op een niet-publiekelijke locatie.

dennissiemensma, 2022-01-24:
Wellicht dat ik probeer om gewoon een vaste Dropbox app te maken voor DSMR-reader. Het lijkt namelijk afdoende om alleen een public key te delen i.c.m. PKCE. 
Alleen zal ik er dan wel een apart Dropbox account voor willen maken. Want de public app key gaat dan in deze repo komen. Ik wil eigenlijk niet mn persoonlijke account eraan gelinked hebben dan.

dennissiemensma, 2022-01-28:
Opzet gemerged naar toekomstige v5 branch. Moet het nog wel uitgebreid testen.

dennissiemensma, 2022-01-30:
Ik draai het inmiddels en het lijkt te werken. Sowieso moet ik v5 even draaien, maar voor nu lijkt het goed.

closed 2022-01-30 by dennissiemensma

dennissiemensma, 2022-02-07:
Ik moet trouwens niet vergeten om over een tijdje te kijken hoeveel gebruikers er gelinked zijn aan de DSMR-reade dropbox app. Er zit een max op van 500 voor development mode. 

> You may apply for production status at anytime, but your application will not be reviewed until your app has at least 50 Dropbox users (legacy Business API apps are not subject to this requirement). While your app has development status, it can link up to 500 Dropbox users or 5 Dropbox teams, so in most cases, it's safe to release your app to the world before it's been granted production status.

dennissiemensma, 2022-02-07:
Daarna moet ik het laten reviewen, maar daar moet ik niet op wachten. Zodra ik 50 gebruikers heb zal ik het indienen daar. Heb al wat branding en kort stukje beschrijving klaargezet in Dropbox app console.

dennissiemensma, 2022-02-08:
Inmiddels 8 gebruikers gekoppeld. Ik kan gelukkig niet zien wie en ook niet bij die bestanden, omdat het allemaal opgelsagen wordt in de lokale DSMR-reader installatie en een kopie in ieders eigen Dropbox-account.

[image]

Nog even ter referentie welke App Key gekoppeld moet worden. Zie de screenshot hierboven, dat moet matchen met deze:
https://github.com/dsmrreader/dsmr-reader/blob/v5/dsmrreader/config/defaults.py#L53-L54

dennissiemensma, 2022-02-10:
Sterker nog, ik denk dat ik dan niet meer mag aanbieden dat iemand zn eigen app maakt. Ik heb dat nog in de docs/admin staan.

dennissiemensma, 2022-02-21:
Ik zit nu op 44 gebruikers, dus loopt mooi door. Inmiddels heb ik ook voor v5.1 straks de flow aangepast zodat men niet meer een eigen app kan instellen. 
Ik weet namelijk zeker dat Dropbox daar anders (terecht) opmerkingen over gaat maken, dus bij deze is mijn "fout" uit het verleden straks hersteld.

dennissiemensma, 2022-02-27:
> Your Dropbox API app, Official DSMR-Reader, has linked 50 Dropbox users total, which means it's now eligible to be reviewed for production status.

> Production status entitles your app to link more than 500 users, which is currently the maximum number of users it can link. You can apply for production from your app's info page found in the [App Console](https://www.dropbox.com/).

> If you've already submitted a production application, no further action is necessary -- your submission will be reviewed within the next several days. **Otherwise, you have two weeks to apply for and receive production approval before your app's ability to link additional Dropbox users will be automatically frozen.**

dennissiemensma, 2022-02-27:
Dus dat betekent dat ik binnen twee weken iets klaar moet hebben staan voor ze.

dennissiemensma, 2022-02-28:
Ik heb een publieke installatie van DSMR-reader klaargezet. Kwam nog wel achter een klein bugje met het instellen van de initiele app key, dus wellicht moet ik dat nog wat versimpelen.

dennissiemensma, 2022-02-28:
Workaround gemaakt. App key is nu helemaal hardcoded en staat ook niet in DB. Is tevens makkelijker.

dennissiemensma, 2022-03-01:
Ik zal kijken of ik morgen of overmorgen de review kan aanvragen.

dennissiemensma, 2022-03-03:
Oke 54 gebruikers. Tijd om prod review in te dienen. Demo-omgeving draait nu ook al paar dagen.

dennissiemensma, 2022-03-03:
Ter referentie, mocht ik het nog een keer opnieuw ter review zetten:

```
Official "DSMR-reader" app for backing up your (daily) DSMR-reader database backups to your Dropbox account. 

"DSMR-reader" is a DSMR-protocol reader, telegram data storage and energy consumption visualizer. It is open-source on Github (https://github.com/dsmrreader/dsmr-reader) and free for non-commercial use. It is usually installed on a RaspberryPi near your smart meter at home.

DMSR means "Dutch Smart Meter Requirements" protocol and is installed on every smart meter in The Netherlands in household, tracking electricity and gas usage.

You can connect this app to your own Dropbox account by following the instructions in DSMR-reader. After connecting it to your account, it will upload your local (daily) database backups to a dedicated App folder in your Dropbox storage. This app will not (and cannot) access any other files in your Dropbox storage. Please ensure to verify this when you authorize/connect this app to your account (Dropbox will list any permissions this app requires, including the dedicated app folder).
Also note that the files are only stored on (1) wherever you host DSMR-reader and (2) a copy will be sent to your own storage on the Dropbox servers. DSMR-reader does not store your files anywhere else.
```

```
- Sync local (backup) folder to user's dropbox

You can review the code/Dropbox SDK usage yourself here: 
https://github.com/dsmrreader/dsmr-reader/blob/v5/dsmr_dropbox/services.py
```

```
You could install the open-source project yourself, but I presume you'd rather focus on the Dropbox integration (which is only a very small part of the project)

So I've installed the project on a public server for you, along with some dummy/fake data. You should be able to login as admin, go directly to the Dropbox integration and fopllow the stepts stated there.

Afterwards the app reviewed SHOULD be connected to your testing account and there SHOULD be files (database backups) uploaded shortly, within a few minutes.

How to:
- Open in browser: /admin/dsmr_backup/dropboxsettings/1/change/
- Login with the username and password provided in "Test account credentials" below
- You should see a page named "Change Dropbox configuration"
- Click the auth link within the 'yellow' panel (located at Step 1), it should open a new browser tab.
- Follow your own website's instructions and copy the Access Code at the end of the flow
- Go back to the DSMR-reader browser tab (where you clicked the link) and enter the code in the "Access Code by Dropbox" field
- Click "Save and continue edit" in the bottom right corner
- The page should be refreshed and the "Dropbox refresh token:" filed should have a green check mark (when the access code entered was OK)
```

dennissiemensma, 2022-03-03:
Ter referentie, mocht ik het nog een keer opnieuw ter review zetten:

## What does your app do?

```
Official "DSMR-reader" app for backing up your (daily) DSMR-reader database backups to your Dropbox account. 

"DSMR-reader" is a DSMR-protocol reader, telegram data storage and energy consumption visualizer. It is open-source on Github (https://github.com/dsmrreader/dsmr-reader) and free for non-commercial use. It is usually installed on a RaspberryPi near your smart meter at home.

DMSR means "Dutch Smart Meter Requirements" protocol and is installed on every smart meter in The Netherlands in household, tracking electricity and gas usage.

You can connect this app to your own Dropbox account by following the instructions in DSMR-reader. After connecting it to your account, it will upload your local (daily) database backups to a dedicated App folder in your Dropbox storage. This app will not (and cannot) access any other files in your Dropbox storage. Please ensure to verify this when you authorize/connect this app to your account (Dropbox will list any permissions this app requires, including the dedicated app folder).
Also note that the files are only stored on (1) wherever you host DSMR-reader and (2) a copy will be sent to your own storage on the Dropbox servers. DSMR-reader does not store your files anywhere else.
```

## How does your app use the Dropbox API?
```
- Sync local (backup) folder to user's dropbox

You can review the code/Dropbox SDK usage yourself here: 
https://github.com/dsmrreader/dsmr-reader/blob/v5/dsmr_dropbox/services.py
```

## How can we try your app's Dropbox integration?
```
You could install the open-source project yourself, but I presume you'd rather focus on the Dropbox integration (which is only a very small part of the project)

So I've installed the project on a public server for you, along with some dummy/fake data. You should be able to login as admin, go directly to the Dropbox integration and fopllow the stepts stated there.

Afterwards the app reviewed SHOULD be connected to your testing account and there SHOULD be files (database backups) uploaded shortly, within a few minutes.

How to:
- Open in browser: /admin/dsmr_backup/dropboxsettings/1/change/
- Login with the username and password provided in "Test account credentials" below
- You should see a page named "Change Dropbox configuration"
- Click the auth link within the 'yellow' panel (located at Step 1), it should open a new browser tab.
- Follow your own website's instructions and copy the Access Code at the end of the flow
- Go back to the DSMR-reader browser tab (where you clicked the link) and enter the code in the "Access Code by Dropbox" field
- Click "Save and continue edit" in the bottom right corner
- The page should be refreshed and the "Dropbox refresh token:" filed should have a green check mark (when the access code entered was OK)
```

dennissiemensma, 2022-03-03:
> Status: In review

dennissiemensma, 2022-03-04:
Het is gelukt! Ze hebben supersnel gereviewd.

> Congratulations, your app's been upgraded to production status! You may now use your app key to make your app available to Dropbox users. Please keep in mind your app must follow the [Dropbox branding guidelines](https://www.dropbox.com/l/AAAsw8lWjNiCWp7m8a6ER_nuQBtOHDcjPV4).

dennissiemensma, 2022-05-04:
Het leek er voor lange tijd op dat de gebruikersstatistieken niet meer werkten, maar nu ik opnieuw check, heeft Dropbox ze bijgewerkt. 
Inmiddels 100+ gebruikers. Dat had ik niet verwacht, gezien het vrij recent is toegevoegd en niet iedereen een voorkeur heeft voor Dropbox.

Dit is dan ook direct alle en de enige informatie die ik kan zien:
[image]

dennissiemensma, 2022-05-04:
Ik vraag me trouwens ook af wat die API-calls-piek tussen eind maart en half april heeft veroorzaakt. Hoe dan ook mooi dat het uiteindelijk weer "vlak" is.

dennissiemensma, 2022-07-26:
[image]

dennissiemensma, 2023-01-31:
Tijd niet meer gekeken, update:

[image]