dsmrreader/dsmr-reader #1212: Geen recente metingen ontvangen

PascalBu, 2020-11-22:
Platform: RPI3B direct install

Krijg plotseling de melding "  Geen recente metingen ontvangen " zonder dat er iets veranderd is in de setup.

Dingen proberen uit te zoeken:
- update naar laatste versie (van 4.6 > 4.8)
- reboot
- USB kabel opnieuw verbonden
- lsusb laat de USB kabel zien wel gewoon zien
- cu geeft geen output (op DSMR versie 2/3, dus weet niet goed of dat klopt), maar eerder dus wel gewoon data binnen gekregen
- extra data logger log via aanpassen .env geeft niet veel informatie extra (of ik weet niet goed waar ik moet kijken?)

- Output van Debug-info:
DSMR-READER
    App / Python / Database                                                        v4.8 / v3.7.3 / postgresql
    Backend sleep / Datalogger sleep / Retention cleanup                                   1.0s / 5.0s / 672h
    Telegram latest version read / Parser settings                                               "None" / "3"

DATA
    Telegrams total (est.)                                                                             234179
    Consumption records electricity / gas (est.)                                                    41498 / -

UNRESOLVED ISSUES
    Geen recente metingen ontvangen                                                            16 uur geleden

Wat kan ik nog doen om te troubleshooten?

dennissiemensma, 2020-11-22:
Bedankt voor je melding. Het is zaak dat je het eerst werkend krijgt via cu. Daarvoor zul je wel de datalogger van dsmrreader moeten stoppen om te voorkomen dat ze elkaar in de weg zitten. En ik neem aan dat je de juiste parameters voor je meterversie hebt geprobeerd (V2 IPV V4+)? "cu -l /dev/ttyUSB0 -s 9600 --parity=none"

PascalBu, 2020-11-22:
Datalogger had ik nog niet gestopt, maar helaas ook na 'sudo supervisorctl stop dsmr-datalogger' en daarna jouw cu command geen output verder dan 'connected'.

Ligt het dan aan de USB kabel? Het gekke is dus dat ik niks veranderd heb. Wel is er een aantal dagen geleden stroomuitval geweest, maar daarna liep alles nog gewoon...

dennissiemensma, 2020-11-22:
Het is lastig te zeggen. Helemaal als je niets hebt veranderd. En ik heb zelf (voor zover ik kan herinneren) nooit gehad of gelezen dat een kabel de geest gaf, maar het is zeker wel een mogelijkheid. Als het in cu niet werkt, werkt het in dsmrreader sowieso ook niet, dus is het zaak om het eerst daar werkend te krijgen. Je zou nog wat tips kunnen vragen op Tweakers in een topic zoals deze, voordat je een andere kabel maakt/bestelt: https://gathering.tweakers.net/forum/list_messages/1578510/last. Want ook zonder dsmrreader zou je gewoon de poort moeten kunnen uitlezen.

jschilperoord, 2020-11-22:
Ik heb hier sinds kort ook last van. @PascalBu Welk OS draai je op je PI? Ik gebruik totaal andere hardware nml een PCEngines APU2. Ik heb het idee dat ik het probleem sinds de upgrade naar ubuntu 20.04 heb. Al draai ik DSMR in een Docker container zal de driver natuurlijk voor de USB kabel misschien nieuwer zijn ivm een nieuwere kernel. Overigens help het bij mij om gewoon de kabel even te ontkoppelen en opnieuw in de USB poort te steken. Dat riekt naar een kabelprobleem?

PascalBu, 2020-11-23:
Ik draai gewoon Raspbian, heb geen automatische updates aanstaan. Heb inmiddels een nieuwe kabel besteld, eens kijken wat dat doet.

dennissiemensma, 2020-11-23:
Dank voor je update. Laat maar weten of dat het uiteindelijk oplost!

dennissiemensma, 2020-12-06:
@PascalBu heeft een andere kabel het probleem voor je opgelost?

PascalBu, 2020-12-06:
@dennissiemensma Het blijkt dat netbeheerder Enexis een storing heeft, waarbij de standen niet uitgelezen kunnen worden. Kwam erachter omdat de energieleverancier ook geen standen meer kon aanleveren. Dus verwacht niet veel van nieuw kabeltje, eerder van actie die Enexis moet gaan ondernemen.

Wat mij betreft kan hier een slotje op. Als het toch niet werkt nadat de storing is opgelost, meld ik me wel weer.

closed 2020-12-06 by PascalBu

dennissiemensma, 2020-12-06:
Prima, dank voor je snelle update!