dsmrreader/dsmr-reader #1213: What are the default Login credentials and how to change these ?

midget1234, 2020-11-23:
Wat zijn de default Login credentials vanuit de installatie ? Hoe zijn deze te wijzigen ? Ik zie geen (duidelijke) info in de FAQ en geen uitleg in de (gesloten) topics op Github.

[image]

niekniek89, 2020-11-23:
> Wat zijn de default Login credentials vanuit de installatie ? Hoe zijn deze te wijzigen ? Ik zie geen (duidelijke) info in de FAQ en geen uitleg in de (gesloten) topics op Github.
> 
> [image]

Goedemorgen,

De default login credentials zijn:

Gebruikersnaam: admin
Wachtwoord: admin

Zodra je bent ingelogd, kies je voor "sla over en ga direct naar het beheer".
Dan kan je rechtsboven kiezen voor "wachtwoord wijzigen".

dennissiemensma, 2020-11-23:
@midget1234 bedankt voor je vraag. Er zijn **geen standaard credentials**, die moet je (bewust) instellen. In de FAQ staat beschreven wat je moet doen: https://dsmr-reader.readthedocs.io/nl/v4/faq.html#how-do-i-set-admin-credentials

Ik zal dat in een toekomstige release nog wat uitbreiden.

Als workaround kun je ook handmatig je gebruiker aanmaken:
```
sudo su - dsmr
./manage.py createsuperuser --user admin --email root@localhost
```

Het emailadres wordt niet gebruikt (DSMR-reader heeft geen standaard mailfunctionaliteit), dus je kunt daar een dummy voor invullen.

midget1234, 2020-11-23:
Merci! Topic kan wat mij betreft geloten worden. Wellich idd handig om in de FAQ iets duidelijker te beschrijven. Het statement "Stel DSMR_USER en DSMR_PASSWORD in via de Env-instellingen" was voor mij in ieder geval niet voldoende.  Bedankt voor de snelle reacties!

dennissiemensma, 2020-11-23:
Bedankt voor je terugkoppeling. Ik heb het net aangepast voor de volgende release in fa66cb97006f7ead45f7fcccf92b6089c85b388e.

closed 2020-11-23 by dennissiemensma

Tibor123214, 2023-04-08:
Hi, ik heb de instructies gevolgd maar krijg een vreemde foutmelding.

(.venv) dsmr@rpi:~/dsmr-reader $ ./manage.py dsmr_superuser
Traceback (most recent call last):
  File "/home/dsmr/dsmr-reader/./manage.py", line 10, in <module>
    execute_from_command_line(sys.argv)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.9/site-packages/django/core/management/__init__.py", line 419, in execute_from_command_line
    utility.execute()
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.9/site-packages/django/core/management/__init__.py", line 413, in execute
    self.fetch_command(subcommand).run_from_argv(self.argv)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.9/site-packages/django/core/management/base.py", line 354, in run_from_argv
    self.execute(*args, **cmd_options)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.9/site-packages/django/core/management/base.py", line 398, in execute
    output = self.handle(*args, **options)
  File "/home/dsmr/dsmr-reader/dsmr_backend/management/commands/dsmr_superuser.py", line 17, in handle
    default=config("DSMRREADER_ADMIN_PASSWORD"),
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.9/site-packages/decouple.py", line 245, in __call__
    return self.config(*args, **kwargs)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.9/site-packages/decouple.py", line 107, in __call__
    return self.get(*args, **kwargs)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.9/site-packages/decouple.py", line 92, in get
    raise UndefinedValueError('{} not found. Declare it as envvar or define a default value.'.format(option))
decouple.UndefinedValueError: DSMRREADER_ADMIN_PASSWORD not found. Declare it as envvar or define a default value.

Iemand een idee wellicht?