dsmrreader/dsmr-reader #1214: DSMR webinterface werkt niet

marco2152, 2020-11-23:
De DSMR webinterface werkt niet. Ik heb de volgende acties uitgevoerd (zonder resultaat tot nu toe):
1. DSMR update uitgevoerd
2. Zachte DSMR reload uitgevoerd (./reload.sh)
3. Harde DSMR reload uitgevoerd (sudo supervisorctl restart all)

Foutmelding na laatste commando: 
* Reloading process: dsmr_webinterface (Gunicorn)kill: (589): Operation not permitted

dennissiemensma, 2020-11-23:
Bedankt voor je melding. Wat is de output van:

```
sudo supervisorctl status
```

dennissiemensma, 2020-11-23:
Houd er trouwens rekening mee dat je `sudo` command **niet als `dsmr`** gebruiker kan uitvoeren. Zowel je derde commando als die van mij net.

Je zult dan even een gebruiker hoger (meestal `pi` of `root`) moeten zijn. Dat kan meestal met `logout` of CTRL+D, er vanuitgaande dat je eerst "van boven naar beneden" bent ingelogd met `sudo su - dsmr`.

marco2152, 2020-11-24:
Hi Dennis - ja, beide commando's uitgevoerd als pi user. Hier de output:
[image: image.png]
Groeten,
Marco

dennissiemensma, 2020-11-24:
Bedankt voor je reactie. Afbeeldingen uploaden naar Github per e-mail werkt niet altijd, dus je kunt het beste even inloggen en opnieuw proberen.

marco2152, 2020-11-24:
[image]

dennissiemensma, 2020-11-24:
Ik denk dat je iets te snel bent geweest. Er staat nu letterlijk: `Uploading image.png…`

Je kunt voordat je op 'Comment' klikt even de preview van je bericht doen hier of die goed is geupload. Soms duurt het een paar tellen en dan zie je in je bericht zoiets verschijnen:

```
![Selection_011](https://user-images.githubusercontent.com/16581744/100152659-507f5080-2ea3-11eb-8159-42c557ca105e.png)
```

marco2152, 2020-11-24:
Blijft moeilijk dit, poging 5
[image]

dennissiemensma, 2020-11-24:
Ah helder. Ik denk dat je het beste even in de logfile kunt kijken wat de fout is. Met:
```
sudo tail /var/log/supervisor/dsmr_webinterface.log -n 30
```

Wat is de output?

dennissiemensma, 2020-11-24:
Het zou namelijk zomaar kunnen dat je zoiets ziet: `Error: Already running on PID 1234 (or pid file '/var/tmp/gunicorn--dsmr_webinterface.pid' is stale)`

marco2152, 2020-11-24:
Ja, klopt helemaal. In mijn geval is PID 589.

Groeten,
Marco

dennissiemensma, 2020-11-24:
In dat geval zul je de pidfile moeten weggooien en dan opnieuw het webinterface proces proberen te herstarten:

```
sudo rm /var/tmp/gunicorn--dsmr_webinterface.pid
sudo supervisorctl restart dsmr_webinterface
```

marco2152, 2020-11-24:
En opgelost! Bedankt Dennis, heel fijn dat je zo laat nog me kon helpen.

Groeten,
Marco

dennissiemensma, 2020-11-24:
Fijn om te horen, dank voor je snelle reactie.

closed 2020-11-24 by dennissiemensma