dsmrreader/dsmr-reader #1217: Welke Postgres queries voor dashboard, richting Grafana

SaturnusDJ, 2020-11-25:
Ik draai DSMR Reader 3.2. (Niet meer geüpgradet vanwege de dashboard verandering.)

Data --> MQTT --> Home Assistant --> InfluxDB --> Grafana.
Maar data gaat ook naar Postgres natuurlijk.
Duplicatie, dus ik probeer nu direct Grafana Postgres te laten lezen.

Wel gelukt:
1. Grafiek voor gas- en elektriciteitsverbruik laatste 24 uur: interval 1 minuut.
2. Grafiek voor gas- en elektriciteitsverbruik per dag. Echter, de stand 'tot nu toe' van de huidige dag staat hier niet in, maar wel via InfluxDB.

Niet gelukt:
3. Het het dagoverzicht wat v3.2 op de startpagina laat zien. Dus: elektriciteit en gas dat vandaag verbruikt is, plus de kosten.
4. Live stroomverbruik (wordt vermoedelijk niet naar de database gestuurd?)

Wie kan helpen met de Postgres query of de Grafana GUI?
Heb al in de v3 broncode zitten kijken maar zonder succes.

dennissiemensma, 2020-11-26:
Bedankt voor je vraag. Het is wat lastig om hier de directe queries neer te zetten, omdat ik een ORM gebruik en daardoor de queries zelf niet maak en daarom vind je die niet in de code. Ik kan je wellicht wel op weg helpen wat je waar kunt vinden en een dump maken van enkele queries.

----

Alle metingen komen initieel binnen in de `dsmr_datalogger_dsmrreading` tabel. Voor elk telegram staat er een record en de hoeveelheid records per minuut hangt af van je meterversie en welke instellingen je in DSMR-reader gebruikt.

Deze tabel is de meest 'live' informatie die er is. Afhankelijk van je meter en DSMR-reader versie (ik weet niet meer welke v5-meter-velden in welke release zijn toegevoegd) staan hier ook gegevens zoals:
- Watt op dit moment (alle fases)
- Watt op dit moment (per fase)
- Volt op dit moment (per fasdsmr_datalogger_meterstatisticse)
- Ampere op dit moment (per fase)
- Ampere op dit moment (per fase)

Laatste meting:
```
SELECT * FROM dsmr_datalogger_dsmrreading ORDER BY timestamp DESC LIMIT 1;
```

----

- Metingen staan dus in `dsmr_datalogger_dsmrreading`.
- Meterstatistieken (uit telegrammen) staan in `dsmr_datalogger_meterstatistics` in een record dat telkens bijgewerkt wordt.
- Gegroepeerde metingen staan in `dsmr_consumption_electricityconsumption` en `dsmr_consumption_gasconsumption`. Deze wordt gebruikt voor de gesplitste elektra/gas grafieken.
- Dagtotalen staan (na afloop van een dag) in `dsmr_stats_daystatistics`. 
- De uurtotalen in `dsmr_stats_hourstatistics`.

----

Voor het berekenen van de dagtotalen wordt nu de eerste en laatste meting van de dag gepakt. Het verschil is het verbruik van de dag en daar worden dan vervolgens de geldende tarieven overheen gerekend. Dit gebeurt in meerdere queries/stappen.

----

Overigens kun je ook nog overwegen om de metingen uit DSMR-reader direct in InfluxDB (of soortgelijke) op te slaan en daar je eigen aggregaties in te doen. En dan kun je de gegevens in DSMR-reader na een week doorlooptijd opschonen, zodat die zijn eigen aggregaties kan blijven doen (die nemen weinig ruimte in), maar de data niet meer dubbel staat. Daarmee houd je het gros van je data op 1 plek, in je InfluxDB.

dennissiemensma, 2020-11-26:
De exacte betekenis (of een omschrijving) van de meting-velden kun je hier vinden:
- https://github.com/dsmrreader/dsmr-reader/blob/v4/dsmr_datalogger/models/reading.py#L23
- Voor meterstatistieken hier: https://github.com/dsmrreader/dsmr-reader/blob/v4/dsmr_datalogger/models/statistics.py#L13

SaturnusDJ, 2020-11-29:
Bedankt voor je uitleg!! Je alternatief om InfluxDB als hoofd aan te houden en Postgres op te schonen is interessant. Ik zal hier over nadenken en uitzoeken wat het handigste uitkomt in de toekomst. Helaas zou daarmee wel een jaar aan data niet in InfluxDB staan en daarmee niet te lezen in Grafana.

dennissiemensma, 2020-11-29:
Sinds DSMR-reader 4 zit er ingebouwde ondersteuning in voor export naar InfluxDB. Ook (eenmalig) met terugwerkend kracht: #1055.

Als je desondanks niet wil upgraden, kun je overwegen om de dsmrreader-database te exporteren en importeren als tijdelijke tweede dsmrreader-database. Die koppel je dan aan een tijdelijke nieuwe DSMR-reader 4 installatie, met puur als doel om die oude metingen te exporteren met de ingebouwde tool. 
De nieuwe installatie doet dan verder niets en ontvangt ook geen nieuwe data en verwijder je weer nadat je alles hebt overgezet. De oude v3 installatie blijft ondertussen onveranderd door draaien.

Een andere manier is om te kijken of je het rechtstreeks uit de database naar InfluxDB krijgt. In principe is dat ook prima mogelijk, maar zul je een scriptje moeten maken die dat met terugwerkende kracht doet. Daar kan ik je evt bij assisteren.

SaturnusDJ, 2020-12-05:
Dat is heel mooi, echt goede ideeën, bedankt voor de input.

Naar aanleiding van zoeken naar 'Postgres vs. InfluxDB' lijkt het (https://portavita.github.io/2018-07-31-blog_influxdb_vs_postgresql) erop dat Postgres betere prestaties heeft, tenkoste van benodigde opslag. Kijk ik naar de huidige Postgres database, dan is dat ongeveer 3GB voor 2 jaar, zonder opschonen, dat valt mee. Ik ben benieuwd hoe dat na het opschonen zal zijn dus zal dat proberen te testen. Vanuit dit perspectief neig ik er naar om de voorkeur aan Postgres te geven, maar dan wordt het wel even stoeien met de queries in Grafana.

Wat het opschonen betreft zie ik dit staat in de documentatie: _The application will slowly apply retention, cleaning up a maximum of 24 hours worth of data on each backend run executed._ Is het dan wel mogelijk opschonen uit te testen in een kloon van de database en applicatie? Aangezien de kloonsetup geen nieuwe data meer uit gaat lezen?

dennissiemensma, 2020-12-05:
Bedankt voor je update. Qua datagrootte heb je het sowieso erg zelf in de hand, namelijk door de datalogger sleep. Als die bijvoorbeeld al van 1 seconde naar 10 seconden gaat, scheelt dat grofweg 9x het aantal opgeslagen metingen. Helemaal op de lange termijn is dat aantrekkelijk, afhankelijk van je hardware natuurlijk.

----

Qua opschoning [doet DSMR-reader (ongeveer) dit](https://github.com/dsmrreader/dsmr-reader/blob/v4/dsmr_datalogger/services/retention.py#L38-L48), bijvoorbeeld bij een retentie van een week:
```
SELECT DATE_TRUNC('hour', "dsmr_datalogger_dsmrreading"."timestamp" AT TIME ZONE 'UTC') AS "item_hour", COUNT("dsmr_datalogger_dsmrreading"."id")
FROM "dsmr_datalogger_dsmrreading"
WHERE ("dsmr_datalogger_dsmrreading"."processed" AND
       "dsmr_datalogger_dsmrreading"."timestamp" < NOW() - INTERVAL '7 day')
GROUP BY DATE_TRUNC('hour', "dsmr_datalogger_dsmrreading"."timestamp" AT TIME ZONE 'UTC')
HAVING COUNT("dsmr_datalogger_dsmrreading"."id") > 2
ORDER BY "item_hour" ASC
LIMIT 24
```

[image]

Hiermee wordt gezocht naar 'uren' die meer dan 2 metingen hebben. Vervolgens [wordt van die uren de eerste en laatste (de '2') bewaard, de rest wordt weggegooid](https://github.com/dsmrreader/dsmr-reader/blob/v4/dsmr_datalogger/services/retention.py#L59-L75). Omdat dit een best intensief proces is op grote datasets, doet die dit telkens voor max 24 uren per keer.

Je kunt op een kloon gewoon het backend-proces draaien, **al zul je wat moeten oppassen met dat die geen andere zaken triggert, zoals uploads naar externe systemen of dropbox of MQTT of andere zaken**. Die kun je dan het beste eerst via de webinterface van de kloon uitschakelen.

dennissiemensma, 2020-12-17:
Mocht je nog meer info of iets nodig hebben, laat hier gerust weten en/of mention me als ik er overheen lees. Ik ga er vanuit dat je met bovenstaande een heel eind op weg kan.

closed 2020-12-17 by dennissiemensma

SaturnusDJ, 2020-12-18:
Prima, dankjewel. Op dit moment bezig met het kijken naar hoe een paar andere projecten voor data ingericht gaan worden, dat heeft invloed op de keuze voor InfluxDB of Postgres, of beide.

SaturnusDJ, 2021-11-28:
Om hier op terug te komen...zowaar bijna een jaar later...

Het Grafana dashboard is omgezet van InfluxDB naar Postgres. Qua prestaties lijkt dat iets beter, maar niet veel. Maar...data retention is nog niet aangepast dus daar valt nog wat te winnen. Queries had ik toentertijd mee gevochten en gisteren en vandaag weer. 

Voor live energie stat:
`SELECT "timestamp" AS "time", electricity_currently_delivered FROM dsmr_datalogger_dsmrreading ORDER BY timestamp DESC LIMIT 1;`

Voor live gas stat:
`SELECT "read_at" AS "time", currently_delivered FROM dsmr_consumption_gasconsumption ORDER BY read_at DESC LIMIT 1;`

Voor het verbruik voor vandaag was het lastiger. Hulp gevraagd en het is gelukt. https://stackoverflow.com/questions/70146083/postgres-query-for-difference-between-latest-and-first-record-of-the-day

Voor energie verbruik vandaag stat:
```
SET TIMEZONE='Europe/Amsterdam';
select max(delivered_1) - min(delivered_1) as current_total, 
       read_at::date as time
from dsmr_consumption_electricityconsumption
where read_at >= current_date and read_at < current_date + 1
group by read_at::date;
```

Voor gas verbruik vandaag stat:
```
SET TIMEZONE='Europe/Amsterdam';
select max(delivered) - min(delivered) as current_total, 
       read_at::date as time
from dsmr_consumption_gasconsumption
where read_at >= current_date and read_at < current_date + 1
group by read_at::date;
```

De grafieken gingen gemakkelijker.

[image]

Kosten moeten ook nog gefixt of anders verwijderd. Water komt nog uit Influx, zou wel tof zijn als ik dat in Postgres weet te krijgen. Zelfs iets voor schrijven dan.

Later nog een update, hopelijk voor 2023. 😆

dennissiemensma, 2021-11-29:
@SaturnusDJ bedankt voor je update! Ik ben niet helemaal thuis in InfluxDB, maar ik zou uiteindelijk wel verwachten dat de performance andersom beter is, maar wellicht hangt het ook af van de specifieke aggregraties en of je wel of niet relaties nodig hebt in PostgreSQL. 
En daarnaast kan ik me ook voorstellen dat ook timeseries-database uiteindelijk slechter performance als de data toch teveel wordt. Linksom of rechtsom wordt de performance in bijna elk systeem beter met wat opschoning! En los daarvan kun je ook ervoor kiezen om bepaalde data te aggregeren en los op te slaan, wat DSMR-reader ook doet om bijvoorbeeld het Archief vlot te houden qua laadtijd. 

Hoe dan ook ben ik benieuwd waar je over een jaar staat. Het ziet er iig visueel mooi uit!

SaturnusDJ, 2021-11-30:
https://portavita.github.io/2018-07-31-blog_influxdb_vs_postgresql is nog steeds de enige vindbare test online.
InfluxDB: tragere queries, minder ruimte nodig. Postgres: snellere queries, meer ruimte nodig.

Op dit moment is de Postgres database 3,3GB. Vorig jaar zei ik 3GB, misschien een afronding van 2,6GB, dus de groei per jaar zonder retentie policy is zelfs minder dan verwacht.

Ik zit nog steeds op versie 3.2. Settings.rst gelezen, maar voor de 100% duidelijkheid: `Backend process sleep` staat op 90. Dit betekent dat de applicatie alle metingen voor zo'n periode in het geheugen houd en dan pas naar de database schrijft?
`Datalogger process sleep` op 30, misschien maak ik daar 15 van. Dit betekent dat de USB adapter elke 15 seconden uitgelezen wordt? Is het dan ook zo dat hoe lager dit getal, hoe preciezer de gegevens in de database?

dennissiemensma, 2021-11-30:
> Dit betekent dat de applicatie alle metingen voor zo'n periode in het geheugen houdt en dan pas naar de database schrijft?

Nee die instelling is puur om eventuele lichte hardware wat te ontlasten. Elke keer dat een backend run draait, checkt/doet die een vast aantal dingen. Het verhogen van deze sleep benadert de database minder, maar heeft geen invloed op de opslag van de metingen, alleen op de verwerking daarvan.

> Dit betekent dat de USB adapter elke 15 seconden uitgelezen wordt? 

Ja in principe wel, al kun je daar vaak nog wel een seconde bij optellen. Na het lezen en wegschrijven van de meting, wacht de datalogger 15 seconden. Dit is vooral handig bij DSMR v5 meters om die te trottlen en indirect daarmee ook de hoeveelheid metingen die opgeslagen worden. Zie https://github.com/dsmrreader/dsmr-reader/issues/1071#issuecomment-670690844

> Is het dan ook zo dat hoe lager dit getal, hoe preciezer/correcter de gegevens in de database zijn?

Ja en nee. Het hangt van wat je wilt weten. Voor het dag/uurverbruik maakt het niet uit of je 1 meting per seconden of 1 per minuut hebt. 
Echter, als je voor elke seconde het gemiddelde Voltage of Ampere wilt weten (en nodig hebt om wat voor reden), dan zijn meer metingen nauwkeuriger, al kun je er vanuit gaan dat bijvoorbeeld elke 5 of 10 seconden een meting ook al een goed beeld geeft. 
Het hangt dus echt van je behoeften af en vandaar dat het ook configureerbaar is.

----

Je kunt een keertje tijdelijk debug logging inschakelen om te zien wat er precies gebeurt. Ook voor de datalogger.

SaturnusDJ, 2021-12-24:
Bedankt weer. De verandering van 30 naar 15 levert geen extra belasting op zie ik. Ik denk dat het enige interessante (voor mij) kan zijn dat het live elektriciteitsverbruik meer up-to-date is, wat in Home Assistant wellicht voor zaken te gebruiken is.

Eerder spraken we over dataretentie. Nu bedacht ik mij...wordt de database wel sneller door opschonen? Kleiner is logisch, maar sneller...

Een kopie van de db gemaakt en die op laten schonen... Dat duurde een dag of drie. Een paar queries getest en daaruit valt niet te concluderen dat een opgeschoonde db nou echt sneller is. Ik houd hem dan wel lekker groot. Naast de DSMR backup wordt er elke nacht ook een db (.psql-bestand) backup gemaakt, dus hopelijk zit dat goed.

Daarmee zijn alle zaken opgelost. Dank nogmaals. :D

dennissiemensma, 2021-12-25:
Je bent uiteraard helemaal vrij in je eigen databeheer en de keuze daarbij. De dataopschoning is toegevoegd nav meerdere meldingen. De database is welliswaar niet sneller, maar het voorkomt dat die traag wordt. Er is geen vaste grens daarvoor, ook de hardware speelt erg mee. Maar weet dat je er op termijn last van kan hebben. Zie #1208. Er zijn wel meer meldingen, ik kan me iemand herinneren met 11 miljoen telegrammen ofzo. Dus weet er van.

dennissiemensma, 2021-12-25:
Nog een andere referentie: https://github.com/dsmrreader/dsmr-reader/issues/1264#issuecomment-762457128 (40M metingen, 8 GB). Je kunt het dus prima draaien zoals je het hebt, maar weet dan in ieder geval de symptonen. 
Als je de datalogger sleep overigens op 15 hebt staan dan krijg je sowieso al 15x zo weinig metingen als normaal en dus zal het een aardige tijd duren voordat je echt een limiet raakt. De sleep is voor de doorvoer, de retentie voor het gevolg.

SaturnusDJ, 2021-12-25:
Dank. Ik zat even in dat topic te lezen en nu begrijp ik niet meer hoe de processen werken. Daartoe probeer ik je suggestie te volgen om debug logging aan te zetten. Maar dat lijkt niet te gebeuren.

```
docker run -d --name="dsmr-reader_debug" --restart=unless-stopped -e SD_LOGLEVEL=debug -e SD_USER=root -e SD_GROUP=root -e SD_AUTOSTART_DATALOGGER=true -e SD_AUTORESTART_DATALOGGER=true -e SD_AUTOSTART_BACKEND=true -e SD_AUTORESTART_BACKEND=true -e SD_AUTOSTART_MQTT=true -e SD_AUTORESTART_MQTT=true --device /dev/ttyUSB0:/dev/ttyUSB0 xirixiz/dsmr-reader-docker:arm32v6-v3.2.0
```
Zie de SD_LOGLEVEL env. (Gevoelige info heb ik uit het commando weggelaten. De database draait extern op een server.) 

Hardcoded naar debug:
```
vi /dsmr/dsmrreader/settings.py
vi /etc/supervisor.d/supervisord.ini
supervisorctl -c /etc/supervisor.d/supervisord.ini reread
supervisorctl -c /etc/supervisor.d/supervisord.ini update
supervisorctl -c /etc/supervisor.d/supervisord.ini restart all
```
Zelfs nog container restart, maar nope.

Dit is de container log, komt overeen met /log/supervisord.log:
```
2021-12-25 11:55:28,029 DEBG 'dsmr_webinterface' stderr output:
[2021-12-25 11:55:28 +0000] [23] [INFO] Starting gunicorn 20.0.4
2021-12-25 11:55:28,036 DEBG 'dsmr_webinterface' stderr output:
[2021-12-25 11:55:28 +0000] [23] [INFO] Listening at: unix:/var/tmp/gunicorn--dsmr_webinterface.socket (23)
2021-12-25 11:55:28,038 DEBG 'dsmr_webinterface' stderr output:
[2021-12-25 11:55:28 +0000] [23] [INFO] Using worker: sync
2021-12-25 11:55:28,050 DEBG 'dsmr_webinterface' stderr output:
[2021-12-25 11:55:28 +0000] [33] [INFO] Booting worker with pid: 33
2021-12-25 11:55:29,053 INFO success: dsmr_webinterface entered RUNNING state, process has stayed up for > than 3 seconds (startsecs)
2021-12-25 11:55:32,289 DEBG 'dsmr_datalogger' stdout output:
The current logging level only logs warnings and errors, to reduce I/O. More information can be found here: https://dsmr-reader.readthedocs.io/en/v2/troubleshooting.html#logging
2021-12-25 11:55:32,397 DEBG 'dsmr_backend' stdout output:
The current logging level only logs warnings and errors, to reduce I/O. More information can be found here: https://dsmr-reader.readthedocs.io/en/v2/troubleshooting.html#logging
2021-12-25 11:55:32,399 DEBG 'dsmr_backend' stdout output:
2021-12-25 11:55:33,821 DEBG 'dsmr_mqtt' stdout output:
The current logging level only logs warnings and errors, to reduce I/O. More information can be found here: https://dsmr-reader.readthedocs.io/en/v2/troubleshooting.html#logging
```

SaturnusDJ, 2021-12-25:
Ik heb het. Het # stond nog in settings.py. :')

SaturnusDJ, 2021-12-25:
Lijkt erop dat `Backend process sleep` is wat ik dacht. Met die op 90 en 15 voor `Datalogger process sleep` worden er steeds ~6 readings per keer verstuurd. Theoretisch gezien dan, want het blijkt erg te schommelen.

Stel `Backend process sleep` wordt op 900 gezet, en er is even stroomuitval, worden de meeste waarden dan later nog naar de db geschreven? In andere woorden, worden readings die nog niet door het backend process zijn verwerkt in het geheugen (weg zonder stroom) óf op de opslag (terug zonder stroom) tijdelijk bewaard?

Het zou nice zijn als het live electriciteitsverbruik bijv. elke seconde zou updaten in de webinterface en over MQTT, maar dat er nog steeds maar elke 15 seconde een reading in de wachtrij voor de database komt. Maar misschien is deze feature er al? Ik loop natuurlijk 2 jaar achter.

dennissiemensma, 2021-12-26:
Je zult dan je backend sleep op 1 moeten zetten en je datalogger op 15. Backend sleep is alleen voor lichte systemen relevant en deels afhankelijk van welke versie van dsmrreader je draait. Een tijd geleden is de wachtrij van taken deels verplaats naar de database, waardoor die bij elke backend run vrijwel niets doet, totdat een taak aan de beurt is. De datalogger schrijft direct naar de database en houdt niets in geheugen.