dsmrreader/dsmr-reader #1218: Geen dagstatistieken meer sinds 1 okt. Wel live metingen zichtbaar. Retentie?

BjornB2, 2020-11-27:
Hey! 
Ik zie zojuist dat sinds 1 oktober kennelijk geen metingen/dagstatistieken meer worden opgeslagen. Al zoekende kwam ik er achter dat ik geen retentie ingesteld had en dat er dus niks verwijderd werd. Ik gebruik DSMR al een aantal jaren, dus ik gok dat de DB volgelopen is. Ik heb nu wel een retentie ingesteld. Worden de gegevens nu vanzelf opgeschoond of moet ik iets handmatig doen om dit te forceren?

Ik heb overigens net probleemloos kunnen updaten van 3.8 naar de laatste 4.x. Toppertje :)

BjornB2, 2020-11-27:
[image]

dennissiemensma, 2020-11-27:
Bedankt voor je melding. Zijn er nog andere waarschuwingen? Je kunt denk ik het beste even in #1208 kijken voor wat queries om te kijken hoever de retentie is. En het is ook handig om een infodump toe te voegen hier: https://dsmr-reader.readthedocs.io/en/latest/faq.html#debug-info-dump-for-github-issues

BjornB2, 2020-11-28:
Hey Dennis, 

Dank voor je bericht. Ik denk dat mijn nieuwe retentieinstelling inmiddels 'verwerkt' is. Ik zie opeens weer data, ook voor de periode tussen 1 oktober en nu. Top!
Thanks voor deze super tool 😊 

Björn

dennissiemensma, 2020-11-29:
Top, fijn om te horen!

closed 2020-11-29 by dennissiemensma