dsmrreader/dsmr-reader #1219: Too many outgoing MQTT

Deepdelver, 2020-11-27:
Ik krijg de volgende melding wanneer ik in dsmr reader v4.3.3 op status klik:

Too many outgoing MQTT messages queued for transit
Since: now

Hoe los ik dit op? Want er komen nu geen mqtt berichten binnen. Ik heb al geprobeerd te zoeken of ik de backlog kan verhogen. Maar kan hier niks over vinden.

dennissiemensma, 2020-11-27:
Bedankt voor je vraag. Dit betekent meestal dat je ergens MQTT aan hebt staan in DSMR-reader, maar dat je de broker niet goed hebt ingesteld. Of DSMR-reader kan geen berichten afleveren naar je broker, zodat het ophoopt.

Je kunt het beste je MQTT-instellingen in DSMR-reader controleren. En anders kun je het beste beginnen bij de logfile van de `dsmr_backend`: https://dsmr-reader.readthedocs.io/en/latest/faq.html#log-files

Deepdelver, 2020-11-29:
Thnx, ik dacht dat de instellingen gewoon goed stonden maar dat hij nu niet verstuurd vanwege de backlog die te groot is. Ik zal even de log erbij gaan toveren en dan kom ik er op terug.

Deepdelver, 2020-12-04:
ik kom dit tegen in de logs, kun je daar wat mee?

```
[2020-12-04 20:36:12,541] WARNING  MQTT: Ignoring dsmr/consumption/gas/read_at to max messages in queue reached
[2020-12-04 20:36:12,602] WARNING  MQTT: Ignoring dsmr/consumption/gas/delivered to max messages in queue reached
[2020-12-04 20:36:12,661] WARNING  MQTT: Ignoring dsmr/consumption/gas/currently_delivered to max messages in queue reached
[2020-12-04 20:36:12,741] DEBUG    Compact: Processed reading: 712036 @ 2020-12-04 20:35:07+01:00
[2020-12-04 20:36:13,168] DEBUG    Compact: Processed reading: 712037 @ 2020-12-04 20:35:20+01:00
[2020-12-04 20:36:13,602] DEBUG    Compact: Processed reading: 712038 @ 2020-12-04 20:35:28+01:00
[2020-12-04 20:36:14,027] DEBUG    Compact: Processed reading: 712039 @ 2020-12-04 20:35:36+01:00
[2020-12-04 20:36:14,447] DEBUG    Compact: Processed reading: 712040 @ 2020-12-04 20:35:44+01:00
[2020-12-04 20:36:14,873] DEBUG    Compact: Processed reading: 712041 @ 2020-12-04 20:35:52+01:00
[2020-12-04 20:36:15,022] DEBUG    Compact: Waiting for newer readings before grouping data...
[2020-12-04 20:36:15,195] DEBUG    SP: Rescheduled "Generate consumption data" to 2020-12-04 20:36:30.039537+01:00 (ETA 0:00:14.844844)
[2020-12-04 20:36:16,054] DEBUG    dsmr_backend.management.commands.dsmr_backend: Sleeping 1.0s
[2020-12-04 20:36:17,263] DEBUG    SP: 0 backend service(s) ready to run

```

dennissiemensma, 2020-12-04:
Bedankt voor je update. Ik denk dat je het beste ook nog even kunt kijken bij de instellingen van de MQTT broker in DSMR-reader, of dat uberhaupt (goed) ingesteld staat.

[image]

Deepdelver, 2020-12-04:
Inmiddels werkt het :D. nogmaals dank voor de hulp. Ligt het aan mij of was deze melding wel wat cryptisch? ( kan geen verbinding maken met de mqtt server, controleer de instellingen. Had duidelijker geweest voor mij.

dennissiemensma, 2020-12-04:
Fijn om te horen! 

Ik ben het eens dat de melding wat beter kan. In de meeste gevallen werkt het vaak in 1x, dus ik twijfel doorgaans hoeveel tijd ik in de foutafhandeling moet steken, vandaar deze manier.

dennissiemensma, 2020-12-04:
Vergeet trouwens niet om debuglogging weer uit te schakelen, helemaal als je op een Pi/SD-kaart draait.

closed 2020-12-04 by dennissiemensma

Deepdelver, 2020-12-05:
Zover ik weet had ik niks veranderd aan de debug ligging, maar zal het nog even controleren.

Wat betreft de foutafhandeling, snap ook dat dit veel tijd kost. Je/jullie hebben een mooi stukje software geschreven. Geen idee hoe ingewikkeld het is, maar anders wil ik best kijken of ik de tijd kan vinden om de error meldingen wat kan verduidelijken.

dennissiemensma, 2020-12-05:
> Zover ik weet had ik niks veranderd aan de debug ligging, maar zal het nog even controleren.

Ik vermoed dat je in je .env `DSMRREADER_LOGLEVEL=DEBUG` hebt staan. Die mag dan weg.

> Wat betreft de foutafhandeling, snap ook dat dit veel tijd kost. Je/jullie hebben een mooi stukje software geschreven. Geen idee hoe ingewikkeld het is, maar anders wil ik best kijken of ik de tijd kan vinden om de error meldingen wat kan verduidelijken.

Bedankt voor je aanbod. Ik zou er niet te veel tijd in stoppen, want als er veel meldingen zijn dan wordt de foutafhandeling vanzelf wel weer uitgebreid. 
De complexiteit zit in dat het asynchroon draait op de achtergrond, dus dan moeten fouten ergens tussentijds opgeslagen worden, maar ook weer niet in alle gevallen. En als een fout zichzelf oplost, of door interactie van een gebruiker, dan moeten die eerdere meldingen weer gewist worden. Vandaar dat het nu op deze ad-hoc manier er in zit.