dsmrreader/dsmr-reader #1220: nieuwe installatie alles werkt maar geen frames in website

frenk64, 2020-11-29:
Net een nieuwe installatie gedaan kan in de website komen maar zie dat alleen text zonder de frames en kleuren

dennissiemensma, 2020-11-29:
Bedankt voor je melding. Het handigste is om even een screenshot te maken en te kijken of je foutmeldingen ziet in je browser console (F12).

frenk64, 2020-11-30:
Bij deze dan ik zie fouten voor het laden van de scripts met F12
[image]

frenk64, 2020-11-30:
Op basis van de f12 info even verder gezocht

Wat ik nu zie dat de benodigde script en images in /home/dsmr......... staan
de NGINX error.log error is not found : /var/www/dsrmr.... en die directory bestaat niet

Denk dat er ergens een link moet zijn klopt dat?

frenk64, 2020-11-30:
Ik heb option 2 gebruikt voor de installatie, omdat ik nu via de f12 melding erachter kwam dat NGINX dus de "templates"" niet kon vinden verder gezocht en zag in optie 1 dat de directory /var/www/dsmrreader gemaakt wordt die dus bij optie 2 kennelijk niet gemaakt wordt. Verder staat erbij dat dan de bestanden later daar geplaatst worden.

Dus het is opgelost de directory gemaakt en de bestanden daarheen gekopieerd 
F5 in de browser en voila ziet er mooi uit, het is nu ff wachten op het kabeltje

closed 2020-11-30 by frenk64

dennissiemensma, 2020-11-30:
Bedankt voor je update. Ik weet niet hoe je de bestanden hebt gekopieerd, maar je hoeft dat niet handmatig te doen. Ook omdat je wellicht dan met toekomstige updates in de war komt.

Als het goed is heb je deze stappen uitgevoerd: https://dsmr-reader.readthedocs.io/en/latest/installation.html#webserver-nginx-part-1
```
sudo mkdir -p /var/www/dsmrreader/static
sudo chown -R dsmr:dsmr /var/www/dsmrreader/
```

Daarna: https://dsmr-reader.readthedocs.io/en/latest/installation.html#bootstrapping

```
sudo su - dsmr
./manage.py collectstatic --noinput
```

Dat zou afdoende moeten zijn,

frenk64, 2020-12-01:
Dag Dennis bedankt voor de melding, ik miste inderdaad nog de frames als ik naar het configuratie sub scherm ging.

Maar inderdaad dit is de volledige oplossing, 
Na het uitvoeren ./manage.. werken de andere schermen nu ook.

de eerste drie sudo opdrachten had ik gisteren al gedaan voor het handmatig  kopiëren, maar dat is duidelijk niet voldoende. 

btw ook CU geïnstalleerd om straks de poort even te checken, wat ik destijds al eens gedaan voor een collega