dsmrreader/dsmr-reader #1221: Kan geen gerbinding maken met de database

Paul-dH, 2020-12-01:
Hoi!

Ik probeer DSMR (docker container) te gebruiken met een losse Postgress server, maar hij wil maar geen verbinding maken met de database host.

De melding uit docker logs -f DSMRReader
```
[ INFO ] Verifying if the DSMR web credential variables have been set...
[ INFO ] Fixing /dev/ttyUSB* security...
[ INFO ] Removing existing PID files...
[ INFO ] Creating log directory...
[ INFO ] Verifying if Postgres in running...
[ FAIL ] Could not connect to database server. Exiting...
```

De output van pg_isready:
```
docker exec -it DSMRReader /bin/bash
bash-5.0# pg_isready -h $DB_HOST -p $DB_PORT
10.1.0.55:5432 - accepting connections
```

Het docker commando om de container op te brengen is:
```
docker run -d --name='DSMRReader' --net='bridge' -e TZ="Europe/Berlin" -e HOST_OS="Unraid" -e 'DB_HOST'='10.1.0.55' -e 'DSMR_USER'='admin' -e 'DSMR_EMAIL'='root@domain.com' -e 'DSMR_PASSWORD'='admin' -e 'VIRTUAL_HOST'='localhost' -e 'TZ'='Europe/Amsterdam' -e 'DB_NAME'='dsmrreader' -e 'DB_USER'='dsmrreader' -e 'DB_PASS'='dsmrreader' -e 'DSMRREADER_ADMIN_PASSWORD'='admin' -e 'DB_PORT'='5432' -p '8888:80/tcp' -v '/mnt/user/appdata/dsmr-reader/backups':'/home/dsmr/app/backups':'rw' --device='/dev/ttyUSB0' 'xirixiz/dsmr-reader-docker'
```

Iemand een idee hoe ik dit kan fixen, bedankt alvast!!

dennissiemensma, 2020-12-01:
Bedankt voor je melding. Ik denk dat je je issue het beste opnieuw kunt aanmaken in het Docker-project: https://github.com/xirixiz/dsmr-reader-docker/issues

Paul-dH, 2020-12-02:
Hoi Dennis, thanks en zal ik doen.

closed 2020-12-02 by Paul-dH