dsmrreader/dsmr-reader #1222: Melding van grootte database -- wat is normale grootte ?

braham2019, 2020-12-01:
Ik heb al geruime tijd de database retentie cleanup opstaan. Nu rest de vraag of het normaal is dat de database dan meer dan 5 GB groot is ? 

DSMR-READER
    App / Python / Database                                                        v4.8 / v3.7.3 / postgresql
    Backend sleep / Datalogger sleep / Retention cleanup                                   1.0s / 0.5s / 168h
    Telegram latest version read / Parser settings                                             "None" / "101"

DATA
    Telegrams total (est.)                                                                            1131480
    Consumption records electricity / gas (est.)                                                29371 / 27086

UNRESOLVED ISSUES
    Database growing large: 5546 MB, consider data cleanup (if not already enabled)                                     now

POSTGRESQL SIZE OF LARGEST TABLES (> 500 MB)
    public.dsmr_datalogger_dsmrreading                                                                5332 MB

dennissiemensma, 2020-12-01:
Bedankt voor je melding. 

- Ik denk dat je het beste even #1208 kunt doorlezen om wat checks te doen en te kijken hoever de opschoning loopt. Na verloop van tijd zul je eenmalig een `vacuumdb` kunnen uitvoeren.
- Verder kan ik je aanraden om direct al de datalogger sleep te verhogen van `0.5` naar `5` seconden of hoger (bijvoorbeeld `10`). Dat zorgt ervoor dat er minder snel data binnenkomt, wat de nummer 1 oorzaak is voor een groeiende database.

Overigens hangt het af van je situatie of 5GB 'groot' is. Voor een Pi kan ik aanraden het zo laag mogelijk te houden. Als je het draait op een volwaardige server of bijvoorbeeld op een NAS/losse SSD, dan is het natuurlijk een ander verhaal. 

Maar in alle gevallen is een kleine database altijd aan te raden, zowel voor performance als backup-grootte.

dennissiemensma, 2020-12-01:
Het kan trouwens zijn dat de opschoning al aardig eind is, gezien je tellers niet bijzonder hoog zijn (1 miljoen metingen, laatst was iemand met 20+ miljoen).

Naast het verhogen van de sleep, kun je direct daarna dit al proberen:

```
sudo su - postgres
vacuumdb -f -v -z -d dsmrreader
```

Mogelijk krimpt je database dan al. Postgresql houd namelijk ruimte vast bij grote opschoningen en dat wordt hergebruikt bij nieuwe metingen. Echter, bij een grote opschoning waarbij de oude datagrootte niet meer terugkomt, forceert de vacuumdb die opschoning zelf.

dennissiemensma, 2020-12-06:
@braham2019 heb je dit inmiddels kunnen oplossen n.a.v. bovenstaande informatie?

braham2019, 2020-12-07:
Ja, ik heb sleep verhoogd naar 5 seconden en gestofzuigd.  Net effect is dat de DB nu nog maar 35 MB groot is, dus dat stofzuigen was dringend nodig ;-)

closed 2020-12-07 by braham2019