dsmrreader/dsmr-reader #1223: Database start niet meer

GerbenKwakkel, 2020-12-02:
Zojuist probeerde ik DSMR te updaten, echter lijkt de boel nu niet meer in lucht te krijgen. De webinterface geeft onderstaande melding:

OperationalError: could not connect to server: Connection refused
	Is the server running on host "localhost" (::1) and accepting
	TCP/IP connections on port 5432?
could not connect to server: Connection refused
	Is the server running on host "localhost" (127.0.0.1) and accepting
	TCP/IP connections on port 5432?

Als ik in de logging kijk van postgresql, zie ik dit:

```
cat /var/log/postgresql/postgresql-11-main.log
2020-12-02 17:55:51.742 GMT [441] LOG:  lock file "postmaster.pid" contains wrong PID: 0 instead of 441
2020-12-02 17:55:51.743 GMT [441] LOG:  performing immediate shutdown because data directory lock file is invalid
2020-12-02 17:55:51.743 GMT [441] LOG:  received immediate shutdown request
2020-12-02 17:55:51.767 GMT [502] dsmrreader@dsmrreader WARNING:  terminating connection because of crash of another server process
2020-12-02 17:55:51.767 GMT [502] dsmrreader@dsmrreader DETAIL:  The postmaster has commanded this server process to roll back the current transaction and exit, because another server process exited abnormally and possibly corrupted shared memory.
2020-12-02 17:55:51.767 GMT [502] dsmrreader@dsmrreader HINT:  In a moment you should be able to reconnect to the database and repeat your command.
2020-12-02 17:55:51.771 GMT [665] WARNING:  terminating connection because of crash of another server process
2020-12-02 17:55:51.771 GMT [665] DETAIL:  The postmaster has commanded this server process to roll back the current transaction and exit, because another server process exited abnormally and possibly corrupted shared memory.
2020-12-02 17:55:51.771 GMT [665] HINT:  In a moment you should be able to reconnect to the database and repeat your command.
2020-12-02 17:55:51.772 GMT [660] dsmrreader@dsmrreader WARNING:  terminating connection because of crash of another server process
2020-12-02 17:55:51.772 GMT [660] dsmrreader@dsmrreader DETAIL:  The postmaster has commanded this server process to roll back the current transaction and exit, because another server process exited abnormally and possibly corrupted shared memory.
2020-12-02 17:55:51.772 GMT [660] dsmrreader@dsmrreader HINT:  In a moment you should be able to reconnect to the database and repeat your command.
2020-12-02 17:55:51.774 GMT [448] WARNING:  terminating connection because of crash of another server process
2020-12-02 17:55:51.774 GMT [448] DETAIL:  The postmaster has commanded this server process to roll back the current transaction and exit, because another server process exited abnormally and possibly corrupted shared memory.
2020-12-02 17:55:51.774 GMT [448] HINT:  In a moment you should be able to reconnect to the database and repeat your command.
2020-12-02 17:55:51.852 GMT [441] LOG:  database system is shut down
2020-12-02 18:04:07.765 GMT [1100] LOG:  listening on IPv6 address "::1", port 5432
2020-12-02 18:04:07.766 GMT [1100] LOG:  listening on IPv4 address "127.0.0.1", port 5432
2020-12-02 18:04:07.773 GMT [1100] LOG:  listening on Unix socket "/var/run/postgresql/.s.PGSQL.5432"
2020-12-02 18:04:08.032 GMT [1101] LOG:  database system was interrupted; last known up at 2020-11-23 20:21:21 GMT
2020-12-02 18:04:08.094 GMT [1102] dsmrreader@dsmrreader FATAL:  the database system is starting up
2020-12-02 18:04:08.101 GMT [1103] dsmrreader@dsmrreader FATAL:  the database system is starting up
2020-12-02 18:04:08.310 GMT [1101] PANIC:  replication checkpoint has wrong magic 5714534 instead of 307747550
2020-12-02 18:04:08.314 GMT [1100] LOG:  startup process (PID 1101) was terminated by signal 6: Aborted
2020-12-02 18:04:08.314 GMT [1100] LOG:  aborting startup due to startup process failure
2020-12-02 18:04:08.330 GMT [1100] LOG:  database system is shut down
pg_ctl: could not start server
Examine the log output.
```

Welke versie ik had durf ik niet met zekerheid te zeggen, de laatste keer updaten was zeker nog uit oktober. Wat zou ik kunnen proberen om dit te verhelpen?

dennissiemensma, 2020-12-02:
Bedankt voor je melding. Draai je dit op een Pi? Je zou nog een herstart kunnen proberen van de hele Pi.

Het is wat lastig te zeggen wat er precies aan de hand is. Draaide het tot aan de update wel goed? Gezien de fout:
```
database system was interrupted; last known up at 2020-11-23 20:21:21 GMT
```

Dit geeft ook aan dat er iets mis is, al zegt het mij helaas niet direct wat:
```
PANIC:  replication checkpoint has wrong magic 5714534 instead of 307747550
```

Heb je nog een recente backup?

GerbenKwakkel, 2020-12-02:
Ik heb geen recente backup, stom.

Tot aan de update draaide het goed, de Pi heb ik ook al eens volledig een herstart gegeven.

De rest van de foutmelding op de webinterface:

`Traceback:
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/handlers/base.py", line 179, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/views/generic/base.py", line 70, in view
    return self.dispatch(request, *args, **kwargs)

  File "/home/dsmr/dsmr-reader/dsmr_frontend/mixins.py", line 10, in dispatch
    if not FrontendSettings.get_solo().always_require_login:

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/solo/models.py", line 54, in get_solo
    obj, created = cls.objects.get_or_create(pk=cls.singleton_instance_id)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/manager.py", line 85, in manager_method
    return getattr(self.get_queryset(), name)(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/query.py", line 573, in get_or_create
    return self.get(**kwargs), False

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/query.py", line 425, in get
    num = len(clone)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/query.py", line 269, in __len__
    self._fetch_all()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/query.py", line 1308, in _fetch_all
    self._result_cache = list(self._iterable_class(self))

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/query.py", line 53, in __iter__
    results = compiler.execute_sql(chunked_fetch=self.chunked_fetch, chunk_size=self.chunk_size)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/sql/compiler.py", line 1154, in execute_sql
    cursor = self.connection.cursor()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/base/base.py", line 259, in cursor
    return self._cursor()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/base/base.py", line 235, in _cursor
    self.ensure_connection()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/base/base.py", line 219, in ensure_connection
    self.connect()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/utils.py", line 90, in __exit__
    raise dj_exc_value.with_traceback(traceback) from exc_value

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/base/base.py", line 219, in ensure_connection
    self.connect()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/base/base.py", line 200, in connect
    self.connection = self.get_new_connection(conn_params)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/postgresql/base.py", line 187, in get_new_connection
    connection = Database.connect(**conn_params)

  File "/usr/lib/python3/dist-packages/psycopg2/__init__.py", line 130, in connect
    conn = _connect(dsn, connection_factory=connection_factory, **kwasync)
`

dennissiemensma, 2020-12-02:
Bedankt voor je aanvulling. Het zit hoe dan ook in de database. Je kunt nog dit proberen, maar ik vermoed dat de uitkomst niet anders wordt:
```
sudo systemctl restart postgresql
```

Ik kan niet helemaal zeker zeggen of dat komt door bijvoorbeeld een corrupt SD-kaartje, maar dit zijn wel vaak de tekenen. Ook dat het wegschrijven goed lijkt te gaan voor het OS, totdat er iets herstart wordt en het naar voren komt. SD-kaartjes zijn hier berucht om, de levensduur varieert van enkele maanden tot vele jaren. Geheel willekeurig.

Wat je sowieso nog kan doen is kijken of je een kopie kan maken van `/var/lib/postgresql/`. Daar schrijft postgresql standaard de gegevens heen. Dan heb je daar een kopie van, los van of het te herstellen is.

Verder kun je kijken of het filesystem nog goed is, al zegt ook dat niet altijd alles. Maar als daar een fout naar voren komt, weet je in ieder geval dat het aan het filesystem ligt. Dit kan met `fsck` en kan warschijnlijk niet terwijl de SD-kaart in gebruik is. Hier staat wel wat informatie: https://raymii.org/s/blog/Broken_Corrupted_Raspberry_Pi_SD_Card.html. Maar het is wel vrij geavanceerd en als je er niet bedreven in bent, maakt het meer stuk, dus voel niet de verplichting om die kant op te gaan.

Heb je uberhaupt geen backup (meer)?

GerbenKwakkel, 2020-12-02:
Het vreemde is, dat wanneer ik mijn Pi herstart de webinterface en mijn data daarin beschikbaar is voor +/- 1-2 minuten. Daarna krijg ik weer de error te zien zoals hierboven.

Als ik de status opvraag van postgresql krijg ik dit:

```
sudo systemctl status postgresql
● postgresql.service - PostgreSQL RDBMS
   Loaded: loaded (/lib/systemd/system/postgresql.service; enabled; vendor preset: enabled)
   Active: active (exited) since Mon 2020-11-23 21:17:22 CET; 1 weeks 1 days ago
  Process: 513 ExecStart=/bin/true (code=exited, status=0/SUCCESS)
 Main PID: 513 (code=exited, status=0/SUCCESS)

Nov 23 21:17:22 Kwakkel-SLIMMEMETER systemd[1]: Starting PostgreSQL RDBMS...
Nov 23 21:17:22 Kwakkel-SLIMMEMETER systemd[1]: Started PostgreSQL RDBMS.

```

dennissiemensma, 2020-12-02:
Afhankelijk van hoe groot je data is, kun je dit proberen in dat kleine window:
```
sudo su - dsmr
./manage.py dsmr_backup_create --full

# En anders
./manage.py dsmr_backup_create --compact
```

Als alternatief kun je een export van je dagtotalen maken via de webinterface op:
```
/export
```

Lukt dat?

GerbenKwakkel, 2020-12-02:
Bedankt voor je reactie.

Een compact backup heb ik nu te pakken, kan ik daar een nieuwe setup bouwen op basis van die backup?
Bij het make van de een full backup krijg ik de volgende melding:

```
./manage.py dsmr_backup_create --full
pg_dump: WARNING:  terminating connection because of crash of another server process
DETAIL:  The postmaster has commanded this server process to roll back the current transaction and exit, because another server process exited abnormally and possibly corrupted shared memory.
HINT:  In a moment you should be able to reconnect to the database and repeat your command.
pg_dump: Dumping the contents of table "dsmr_datalogger_dsmrreading" failed: PQgetCopyData() failed.
pg_dump: Error message from server: SSL SYSCALL error: EOF detected
pg_dump: The command was: COPY public.dsmr_datalogger_dsmrreading (id, "timestamp", electricity_delivered_1, electricity_returned_1, electricity_delivered_2, electricity_returned_2, electricity_currently_delivered, electricity_currently_returned, extra_device_timestamp, extra_device_delivered, processed, phase_currently_delivered_l1, phase_currently_delivered_l2, phase_currently_delivered_l3, phase_currently_returned_l1, phase_currently_returned_l2, phase_currently_returned_l3, phase_voltage_l1, phase_voltage_l2, phase_voltage_l3, phase_power_current_l1, phase_power_current_l2, phase_power_current_l3) TO stdout;
Traceback (most recent call last):
  File "./manage.py", line 10, in <module>
    execute_from_command_line(sys.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/__init__.py", line 401, in execute_from_command_line
    utility.execute()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/__init__.py", line 395, in execute
    self.fetch_command(subcommand).run_from_argv(self.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/base.py", line 330, in run_from_argv
    self.execute(*args, **cmd_options)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/base.py", line 371, in execute
    output = self.handle(*args, **options)
  File "/home/dsmr/dsmr-reader/dsmr_backup/management/commands/dsmr_backup_create.py", line 37, in handle
    folder=base_folder
  File "/home/dsmr/dsmr-reader/dsmr_backup/services/backup.py", line 123, in create_full
    on_backup_failed(process_handle=backup_process)
  File "/home/dsmr/dsmr-reader/dsmr_backup/services/backup.py", line 195, in on_backup_failed
    error_message = process_handle.stderr.read()
AttributeError: 'NoneType' object has no attribute 'read'
```

GerbenKwakkel, 2020-12-02:
Bij het checken van de backup, zie ik dat ik wel backups maakte lokaal in DSMR.

Even uitzoeken hoe ik dit ga herstellen. Bedankt voor het meedenken tot zover.

dennissiemensma, 2020-12-02:
Top! Stel die backups dan vooral zsm veilig buiten je SD kaart en check het filesystem van de sd kaart, voordat je die verder gebruikt. Databases raken niet zomaar corrupt. Voor de restore: Je hoeft in principe alleen dsmrreader uit te zetten, de database te verwijderen, opnieuw aanmaken en terugzetten. Daarna kun je het weer starten en zou die verder moeten gaan op punt van restore. Zie de FAQ voor meer info en anders hoor ik graag als het niet lukt.

dennissiemensma, 2020-12-06:
@GerbenKwakkel ben je hier nog verder mee gekomen of heb je geen data kunnen herstellen?

GerbenKwakkel, 2020-12-08:
@dennissiemensma vanavond, zojuist, ben ik er eindelijk weer aan toegekomen. Echter kom ik niet echt verder:

```
pi@SLIMMEMETER:~ $ zcat /home/dsmr/dsmr-reader/backups/dsmrreader-postgresql-backup-Monday.sql.gz | sudo -u postgres psql dsmrreader
psql: could not connect to server: No such file or directory
        Is the server running locally and accepting
        connections on Unix domain socket "/var/run/postgresql/.s.PGSQL.5432"?
```

dennissiemensma, 2020-12-08:
Je zult wel eerst je database weer moeten starten. Heb je nog je SD-kaartje kunnen controleren met bijvoorbeeld `fsck` of soortgelijke tool?

Anders zou je kunnen overwegen om zowel de database als progres te verwijderen en opnieuw te installeren. Volgens mij zou dit moeten werken:
```
sudo apt-get purge postgresql
sudo rm -rf /var/lib/postgresql/
sudo apt-get install postgresql
```

**Let op, dit verwijdert dus definitief je gegevens en zijn niet meer te herstellen!** Je kunt sowieso nog kijken of je eerst nog een kopie kan maken van `/var/lib/postgresql/` zodat je die parallel een keertje kunt proberen te herstellen.

Daarna kun je weer een lege database aanmaken en nogmaals proberen te importeren:
```
sudo -u postgres createuser -DSR dsmrreader
sudo -u postgres createdb -O dsmrreader dsmrreader
sudo -u postgres psql -c "alter user dsmrreader with password 'dsmrreader';"
```

dennissiemensma, 2020-12-17:
@GerbenKwakkel heeft bovenstaande je verder op weg geholpen?

GerbenKwakkel, 2020-12-21:
@dennissiemensma tot op heden ben ik er nog niet aan toegekomen. Voor nu mag je het issue wel sluiten, ik denk dat de SD-kaart corrupt is.

closed 2020-12-22 by dennissiemensma