dsmrreader/dsmr-reader #1224: ERR_CONNECTION_REFUSED (achter een reverse proxy)

MRL-33, 2020-12-02:
Ik heb mijn systeem (Raspberry Pi 3B+) ingericht volgens de nieuwe situatie (zie hieronder) en nu kan ik niet meer op de DSMR-Reader website terecht.

Oude situatie
Ik heb in Docker een container DSMR-Reader lopen. Deze is geconfigureerd met docker-compose.
De webpagina was altijd bereikbaar met: 192.168.x.y:7777

Nieuwe situatie
Nu heb ik ook een extra container ([Nginx Proxy Manager](https://nginxproxymanager.com/)) toegevoegd wat als Reverse Proxy fungeert. Poort 80 en 443 zijn geforward naar dit systeem.
De webpagina is nu niet meer bereikbaar met 192.168.x.y:7777
Docker-compose.yaml is niet gewijzigd t.o.v. de oude situatie.

Wat moet ik wijzigen om de webpagina te kunnen bereiken op 192.168.x.y:7777 ?
Heb je nog meer informatie nodig om mijn probleem te begrijpen/op te lossen?

Bestand "docker-compose.yaml"
`
version: '3.8'

##the dashboard is reachable at: http://<HOSTNAME>:7777
##volume: folder_on_host:folder_inside_container

services:
  dsmrdb:
    image: postgres:12-alpine
    container_name: dsmrdb
    restart: always
    volumes:
      - /etc/localtime:/etc/localtime:ro
      - ./dsmrdb:/var/lib/postgresql/data
    environment:
      - TZ=$TZ
      - PG_TZ=$TZ
      - POSTGRES_USER=$POSTGRES_USER
      - POSTGRES_PASSWORD=$POSTGRES_PASSWORD
      - POSTGRES_DB=$POSTGRES_DB

  dsmr:
    image: xirixiz/dsmr-reader-docker:latest
    container_name: dsmr
    depends_on:
      - dsmrdb
    cap_add:
      - NET_ADMIN    
    links:
      - dsmrdb
    restart: always
    volumes:
      - /etc/localtime:/etc/localtime:ro
      - ./dsmr_backups:/dsmr/backups
    environment:
      - TZ=$TZ
      - VIRTUAL_HOST=localhost
      - SD_AUTOSTART_MQTT=false
      - SD_AUTORESTART_MQTT=false
    ports:
      - 7777:80
      - 7779:443
    devices:
      - /dev/ttyUSB0:/dev/ttyUSB0
`

bestand ". env"
`
##enviroment

##timezone
TZ=Europe/Amsterdam

##database
POSTGRES_USER=dsmrreader
POSTGRES_PASSWORD=dsmrreader
POSTGRES_DB=dsmrreader

##Nginx
ENABLE_HTTP_AUTH=false
HTTP_AUTH_USERNAME=
HTTP_AUTH_PASSWORD=
`

dennissiemensma, 2020-12-02:
Bedankt voor je vraag. Ik ben niet helemaal thuis in die mapping via Docker. 

Hoe de normale installatie van DSMR-reader werkt is dat Nginx [al het verkeer dat **niet** op `/static/` komt](https://github.com/dsmrreader/dsmr-reader/blob/v4/dsmrreader/provisioning/nginx/dsmr-webinterface#L2), doorstuurt naar de socket op `/var/tmp/gunicorn--dsmr_webinterface.socket`. Die socket komt uit het `dsmr_webinterface` proces van DSMR-reader.

Je zou je issue ook nog bij Xirixiz kunnen leggen en ander overwegen om de default setup te gebruiken: https://github.com/xirixiz/dsmr-reader-docker/issues

MRL-33, 2020-12-03:
Bedankt voor je antwoord. 
Oef.  Je antwoord snap ik nog niet helemaal omdat ik nog niet erg bedreven ben in het configureren in Nginx.
Ik zal Xirixiz eens benaderen voor hulp.

dennissiemensma, 2020-12-06:
@MRL-33 heb je inmiddels de oplossing hiervoor gevonden?

closed 2020-12-17 by dennissiemensma