dsmrreader/dsmr-reader #1225: Verkeerde datum in DSMR Reader (5 maanden in het verlenden

snoekieboe, 2020-12-03:
Beste,

Ik heb op basis van Xiriz's images middels docker compose DSMR reader in Docker geinstalleerd. Daarnaast maak ik gebruik van een remote data logger draaiende op een raspberry pi.

Alles werkt naar behoren maar de datum die in het dashboard wordt getoond is die van 5 maanden geleden. Vandaag toont hier 02-07-2020 ipv 03-12-2020. 

Ik heb in de docker containers.. op de docker host... en op de remote data logger de tijdzonde configuratie gecontroleerd en het commando "data" geeft ook de juiste datum terug.

Wat kan ik doen om er voor te zorgen dat het dashboard gewoon de juiste datum toont?

Dank voor jullie hulp!

dennissiemensma, 2020-12-03:
Bedankt voor je melding. De datumtijd wordt bepaald op basis van wat je slimme meter teruggeeft. Soms staat die klok niet goed, dus daar zou je als eerste moeten kijken of die afwijkt. Zie issue #686 voor meer informatie als startpunt.

dennissiemensma, 2020-12-06:
@snoekieboe ben je inmiddels in de gelegenheid geweest om te kijken of het aan de klok in de slimme meter zelf lag?

snoekieboe, 2020-12-08:
Zeker... ik heb inmiddels contact met netbeheerder Enexis maar daar eerlijk gezegd niets wijzer geworden.
Zij geven aan dat als de Meter netjes de meterstanden registreert de datum en tijd ook correct zijn en ik de problemen moet zoeken in de apparatuur waarmee ik de P1 poort uitlees (RPI met remote datalogger).

Kortom, nog steeds een verkeerde datum.

Hieronder een dump van mijn meest recente telegram bericht waar ik overigens niet de velden in zie die in issue 686 genoemd worden:

/Ene5\T210-D ESMR5.0

1-3:0.2.8(50)
0-0:1.0.0(200707200748S)
0-0:96.1.1(4530303438303030303436383034333230)
1-0:1.8.1(001144.773*kWh)
1-0:1.8.2(000000.000*kWh)
1-0:2.8.1(000112.696*kWh)
1-0:2.8.2(000000.000*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(00.145*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00025)
0-0:96.7.9(00009)
1-0:99.97.0(2)(0-0:96.7.19)(200608073837S)(0000001107*s)(200603091132S)(0000000304*s)
1-0:32.32.0(00005)
1-0:52.32.0(00005)
1-0:72.32.0(00005)
1-0:32.36.0(00000)
1-0:52.36.0(00000)
1-0:72.36.0(00000)
0-0:96.13.0()
1-0:32.7.0(232.0*V)
1-0:52.7.0(233.0*V)
1-0:72.7.0(232.0*V)
1-0:31.7.0(000*A)
1-0:51.7.0(001*A)
1-0:71.7.0(000*A)
1-0:21.7.0(00.045*kW)
1-0:41.7.0(00.094*kW)
1-0:61.7.0(00.005*kW)
1-0:22.7.0(00.000*kW)
1-0:42.7.0(00.000*kW)
1-0:62.7.0(00.000*kW)
!71E6

dennissiemensma, 2020-12-08:
Bedankt voor je update. Vreemd dat ze dat aangeven, want in je telegram staat wel degelijk dat de huidige date op "7 juli 2020" staat. 

Deze regel zoek je:
```
0-0:1.0.0(200707200748S)
```

Dit betekent: 
`.    200707200748S`
`.    20 07 07 20 07 48 S`
20`20`-`07`-`07`-`20`:`07`:`48` `Summertime`

Klopt het trouwens dat je geen (slimme) gasmeter erbij hebt gekregen? Die heeft namelijk ook nog een interne klok (dacht ik) en zou ook nog een mooi twee ijkpunt zijn

snoekieboe, 2020-12-08:
Ja klopt... nieuwbouw dus gasloos :-)

Ik ga gewoon nog eens met enexis bellen.. thnx!

snoekieboe, 2020-12-08:
Zou eigenlijk wel tof zijn als de remote datalogger de mogelijkheid zou hebben deze timestamp te overschrijven met zijn eigen lokale tijd. Zou niet nodig moeten zijn maar wel een mooie oplossing als je dit soort uitdagingen hebt en een weerbarstige netbeheerder :-)

dennissiemensma, 2020-12-08:
Uiteindelijk is daar wel iets voor te maken hoor, maar ik probeer eerst de oorzaak te vinden. Workaround kan altijd nog. 

Heb je nog een ander telegram voor me? Ik probeerde bovenstaande te parsen en dat lukte niet. Er zit een CRC-handtekening in de telegrammen, op de laatste regel (hier `!71E6`). DSMR-reader controleert die om te checken dat er geen bits zijn weggevallen in het telegram (gebeurt nog wel eens met een seriele verbinding).

Want ik neem aan dat je wel regelmatig data-updates ziet in DSMR-reader? Ondanks dat de datum niet klopt? Het kan namelijk ook aan je kabel liggen, maar het zal niet zo zijn dat telkens dezelfde bit wegvalt en dat die toevallig elke keer uitkomt op "juli". Maar ik check graag een telegram waarbij de CRC ook valideert.

Verder kun je even kijken of je ook duidelijk bij nieuwe telegrammen telkens de laatste cijfers van `0-0:1.0.0(200707200748S)` ziet oplopen (dus dat de tijd wel mee gaat bij elk nieuw telegram).

snoekieboe, 2020-12-08:
Meest recente telegram hieronder ( en de timestamp loopt gewoon op )

/Ene5\T210-D ESMR5.0

1-3:0.2.8(50)
0-0:1.0.0(200707205318S)
0-0:96.1.1(4530303438303030303436383034333230)
1-0:1.8.1(001144.883*kWh)
1-0:1.8.2(000000.000*kWh)
1-0:2.8.1(000112.696*kWh)
1-0:2.8.2(000000.000*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(00.144*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00025)
0-0:96.7.9(00009)
1-0:99.97.0(2)(0-0:96.7.19)(200608073837S)(0000001107*s)(200603091132S)(0000000304*s)
1-0:32.32.0(00005)
1-0:52.32.0(00005)
1-0:72.32.0(00005)
1-0:32.36.0(00000)
1-0:52.36.0(00000)
1-0:72.36.0(00000)
0-0:96.13.0()
1-0:32.7.0(225.0*V)
1-0:52.7.0(223.0*V)
1-0:72.7.0(224.0*V)
1-0:31.7.0(000*A)
1-0:51.7.0(001*A)
1-0:71.7.0(000*A)
1-0:21.7.0(00.046*kW)
1-0:41.7.0(00.092*kW)
1-0:61.7.0(00.005*kW)
1-0:22.7.0(00.000*kW)
1-0:42.7.0(00.000*kW)
1-0:62.7.0(00.000*kW)
!8F06

dennissiemensma, 2020-12-08:
Je tweede telegram lijkt inderdaad gewoon goed qua CRC, dus je meter stuurt daadwerkelijk deze gegevens over je kabel. En omdat je ook nog aangeeft dat het bij je vorige woning goed ging, zou ik zelf ook geen andere oorzaak meer weten.

Wellicht leest de netbeheerder het anders uit, ik weet niet helemaal hoe dat aan hun zijde werkt, maar als het gewoon via hetzelfde telegram-mechanisme gaat zou ik verwachten dat zij ook die foute datum moeten zien. Tenzij hun software op hun kantoor dat zelf automatisch recht trekt natuurlijk, daar heb ik verder geen zicht op.

snoekieboe, 2020-12-12:
Tijd voor een (laatste) update. Enexis heeft begin november problemen gehad met de onboarding van nieuwe meters. Mijn meter, opgeleverd op 03-11-2020, hoorde helaas ook bij deze batch. Als gevolg van deze problemen kon de meter niet de juiste tijd uit het netwerk ophalen met de door mij beschreven problemen als gevolg.

Enexis heeft een nieuwe synchronisatie van de meter uitgevoerd en sinds gisterenavond bevatten de telegramberichten de juiste tijd en..... ziet alles er prima uit in DSMR reader.

Dank voor de ondersteuning Dennis!

dennissiemensma, 2020-12-12:
Bedankt voor je update en fijn om te horen dat het nu eindelijk goed werkt!

closed 2020-12-12 by dennissiemensma

mlakerveld, 2020-12-17:
In november gaf Enexis aan dat mijn slimme meter werd vervangen/repareerd omdat hij niet uit te lezen was. (Is ie volgens mij al een jaar niet maar ja).
De monteur heeft een nieuwe opgehangen, die wel bereik had. In de instructies staat dat ze de eerste week veel test uitlezen en in die week ook alleen laag tarief rekenen 👍 . We zijn nu 3 weken verder en ik heb nog steeds alleen laag tarief.

En mijn DSMR Reader is helemaal kaduuk gegaan (crasht steeds een aantal uur na reboot, logisch als je oude telegrams krijgt die al maanden in de database staan?). Mijn vermoeden na het lezen van deze issue heb ik hetzelfde. De telegrams komen binnen op augustus van dit jaar -_-

Zal ook eens een belletje doen. Want ik gok dat het uitlezen nog steeds niet werkt. Laatste telegram toegevoegd, ik snap de structuur niet helemaal, maar ik zie 2x een 2020 08 datum en ook een 2020 07 datum toch?

`
/Ene5\T210-D ESMR5.0

1-3:0.2.8(50)
0-0:1.0.0(200816161835S)
0-0:96.1.1(4530303438303030303437323033393230)
1-0:1.8.1(000174.673*kWh)
1-0:1.8.2(000000.000*kWh)
1-0:2.8.1(000000.089*kWh)
1-0:2.8.2(000000.000*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(00.445*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00013)
0-0:96.7.9(00006)
1-0:99.97.0(1)(0-0:96.7.19)(200725151601S)(0000001985*s)
1-0:32.32.0(00002)
1-0:52.32.0(00002)
1-0:72.32.0(00002)
1-0:32.36.0(00000)
1-0:52.36.0(00000)
1-0:72.36.0(00000)
0-0:96.13.0()
1-0:32.7.0(230.0*V)
1-0:52.7.0(230.0*V)
1-0:72.7.0(226.0*V)
1-0:31.7.0(000*A)
1-0:51.7.0(000*A)
1-0:71.7.0(001*A)
1-0:21.7.0(00.088*kW)
1-0:41.7.0(00.000*kW)
1-0:61.7.0(00.357*kW)
1-0:22.7.0(00.000*kW)
1-0:42.7.0(00.000*kW)
1-0:62.7.0(00.000*kW)
0-1:24.1.0(003)
0-1:96.1.0(4730303732303034303032353236373230)
0-1:24.2.1(200816161500S)(00051.991*m3)
!401C`

dennissiemensma, 2020-12-17:
@mlakerveld deze moet je hebben:

```
# Slimme meter
0-0:1.0.0(200816161835S)

# Gasmeter
0-1:24.2.1(200816161500S)(00051.991*m3)
```

Deze is volgens mij iets met storingen ofzo, maar daar doet DSMR-reader niets mee:
```
1-0:99.97.0(1)(0-0:96.7.19)(200725151601S)(0000001985*s)
```

Je kunt zelf de vertaling doen met dit document: https://github.com/dsmrreader/dsmr-reader/blob/v4/docs/_static/downloads/netbeheer-nederland-dsmr-specs-v5.0.2.pdf
Vanaf pagina 20 daar staat een tabel met alle codes en wat het grofweg is.

dennissiemensma, 2020-12-17:
Wat betreft jullie issue met verkeerde tijd. Ik denk dat ik dan wel iets kan maken wat jullie in staat stelt om DSMR-reader "nu" als inkomende datumtijd te gebruiken. Die zal dan het telegram inhoudelijk negeren qua datumtijd.