dsmrreader/dsmr-reader #1226: ubuntu  upgrade 18.04 naar 20.04.1 faalt

rodeho, 2020-12-04:
Ha Dennis

ik had mijn Ubuntu VM geüpgraded van 18.04 naar 20.04, en daarna werkt DSMR reader niet meer. Met een snapshot was het zo weer hersteld, is 20.04 supported? Zo ja welke commandos kan ik draaien om te bij te dragen aan het onderzoeken van het probleem?

```
sudo supervisorctl status
dsmr_backend                     STARTING
dsmr_webinterface                FATAL     Exited too quickly (process log may have details)
```

dennissiemensma, 2020-12-04:
Bedankt voor je melding. Ik heb `20.04` tot nu toe niet geprobeerd en ontwikkel zelf nog in `18.04`. 

Ik denk dat je het beste kunt beginnen bij de logfiles om te te zien wat de inhoudelijke foutmelding is: https://dsmr-reader.readthedocs.io/en/latest/faq.html#log-files

Mijn eerste vermoeden is dat de (standaard) Python-versie geupgrade is na zo'n grote upgrade en dat de vorige versie niet meer ondersteund wordt door het OS. In die gevallen is het redelijk eenvoudig te fixen, maar dat is dus puur een aanname. Ik ben benieuwd naar de inhoudelijke foutmelding.

Mocht je hiervoor trouwens je 'echte' DSMR-reader VM voor moeten upgraden dan zou ik het voor nu even laten voor wat het is. Dan kijk ik er wel een keer apart naar in een eigen testomgeving.

rodeho, 2020-12-05:
Ja klopt het is mijn 'echte' DSMR-reader, liever hou ik die intact. Stel dat ik nog een poging zou ondernemen, welke logfiles/commandos zijn relevant voor het onderzoek?

dennissiemensma, 2020-12-06:
Ik verwacht dat je de meeste informatie kunt vinden in de logfiles in `/var/log/supervisor/`.

dennissiemensma, 2020-12-30:
@rodeho kan het zijn dat je tegen dezelfde foutmelding aanliep als iemand anders in #1241? Kun je je dat nog herinneren?

rodeho, 2021-01-02:
@dennissiemensma sorry nee dat staat mij niet meer bij. Ik moet nog een keer de uitgebreide test doen, maar homeassistant is eating all the hobby time ;-) werkt overigens goed met MQTT vanuit DSMR.

dennissiemensma, 2021-01-20:
Ik heb net in virtualbox een installatie gedaan in Ubuntu 20.04 en dat lijkt goed. Vermoedelijk is de fix voor #1241 afdoende geweest.

closed 2021-01-20 by dennissiemensma

rodeho, 2021-02-02:
Heb een upgrade gedaan, helaas nog steeds gefaald. Kan ik nog wat logs aanleveren? Heb een snapshot van de foute situatie, kan makkelijk terug.

```
The current logging level only logs warnings and errors, to reduce I/O. More information can be found here: https://dsmr-reader.readthedocs.io/en/latest/troubleshooting.html#logging
The current logging level only logs warnings and errors, to reduce I/O. More information can be found here: https://dsmr-reader.readthedocs.io/en/latest/troubleshooting.html#logging
```

```
Current thread 0x00007fb21f815740 (most recent call first):
Could not find platform independent libraries <prefix>
Could not find platform dependent libraries <exec_prefix>
Consider setting $PYTHONHOME to <prefix>[:<exec_prefix>]
Fatal Python error: Py_Initialize: Unable to get the locale encoding
ModuleNotFoundError: No module named 'encodings'

```

```
ImportError: cannot import name 'dist'
[2021-02-02 08:38:33 +0000] [26132] [INFO] Worker exiting (pid: 26132)
[2021-02-02 08:38:33 +0000] [25977] [INFO] Shutting down: Master
[2021-02-02 08:38:33 +0000] [25977] [INFO] Reason: Worker failed to boot.
[2021-02-02 08:38:38 +0000] [28563] [INFO] Starting gunicorn 20.0.4
[2021-02-02 08:38:38 +0000] [28563] [INFO] Listening at: unix:/var/tmp/gunicorn--dsmr_webinterface.socket (28563)
[2021-02-02 08:38:38 +0000] [28563] [INFO] Using worker: sync
[2021-02-02 08:38:38 +0000] [28779] [INFO] Booting worker with pid: 28779
[2021-02-02 08:38:38 +0000] [28779] [ERROR] Exception in worker process
Traceback (most recent call last):
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/gunicorn/arbiter.py", line 583, in spawn_worker
    worker.init_process()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/gunicorn/workers/base.py", line 119, in init_process
    self.load_wsgi()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/gunicorn/workers/base.py", line 144, in load_wsgi
    self.wsgi = self.app.wsgi()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/gunicorn/app/base.py", line 67, in wsgi
    self.callable = self.load()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/gunicorn/app/wsgiapp.py", line 49, in load
    return self.load_wsgiapp()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/gunicorn/app/wsgiapp.py", line 39, in load_wsgiapp
    return util.import_app(self.app_uri)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/gunicorn/util.py", line 358, in import_app
    mod = importlib.import_module(module)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 941, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 665, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 678, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/dsmr/dsmr-reader/dsmrreader/__init__.py", line 1, in <module>
    from django.utils.version import get_version
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/__init__.py", line 1, in <module>
    from django.utils.version import get_version
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/utils/version.py", line 6, in <module>
    from distutils.version import LooseVersion
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/distutils/__init__.py", line 25, in <module>
    from distutils import dist, sysconfig
ImportError: cannot import name 'dist'
[2021-02-02 08:38:38 +0000] [28779] [INFO] Worker exiting (pid: 28779)
[2021-02-02 08:38:38 +0000] [28563] [INFO] Shutting down: Master
[2021-02-02 08:38:38 +0000] [28563] [INFO] Reason: Worker failed to boot.
Could not find platform independent libraries <prefix>
Could not find platform dependent libraries <exec_prefix>
Consider setting $PYTHONHOME to <prefix>[:<exec_prefix>]
Fatal Python error: Py_Initialize: Unable to get the locale encoding
ModuleNotFoundError: No module named 'encodings'
```

rodeho, 2021-02-02:
@dennissiemensma heb de VM nu gecloned en kan hem nu gebruiken zonder dat het de productie beïnvloed. Laat maar weten wat je nog meer aan logs wilt zien?

```
@dsmr:~$ sudo supervisorctl status all
dsmr_backend                     BACKOFF   Exited too quickly (process log may have details)
dsmr_webinterface                FATAL     Exited too quickly (process log may have details)
```

```
(dsmrreader) dsmr@dsmr:~/dsmr-reader$ ./deploy.sh

 --- You are currently running version:
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/dsmr/dsmr-reader/dsmrreader/__init__.py", line 1, in <module>
    from django.utils.version import get_version
ModuleNotFoundError: No module named 'django'

     [!] FAILED to call Django (did you activate the 'dsmrreader' VirtualEnv before running?)

```

Als ik dan wat virtenv zaken tracht te re-deployen

```
o@dsmr:~$ sudo -u dsmr /home/dsmr/.virtualenvs/dsmrreader/bin/pip3 install -r /home/dsmr/dsmr-reader/dsmrreader/provisioning/requirements/base.txt
Traceback (most recent call last):
  File "/home/dsmr/.virtualenvs/dsmrreader/bin/pip3", line 5, in <module>
    from pip._internal.cli.main import main
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/pip/_internal/cli/main.py", line 10, in <module>
    from pip._internal.cli.autocompletion import autocomplete
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/pip/_internal/cli/autocompletion.py", line 9, in <module>
    from pip._internal.cli.main_parser import create_main_parser
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/pip/_internal/cli/main_parser.py", line 7, in <module>
    from pip._internal.cli import cmdoptions
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/pip/_internal/cli/cmdoptions.py", line 28, in <module>
    from pip._internal.models.target_python import TargetPython
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/pip/_internal/models/target_python.py", line 3, in <module>
    from pip._internal.pep425tags import get_supported, version_info_to_nodot
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/pip/_internal/pep425tags.py", line 7, in <module>
    from pip._vendor.packaging.tags import (
ModuleNotFoundError: No module named 'pip._vendor.packaging.tags'
@dsmr:~$ sudo -u dsmr /home/dsmr/.virtualenvs/dsmrreader/bin/python3 /home/dsmr/dsmr-reader/manage.py migrate
Traceback (most recent call last):
  File "/home/dsmr/dsmr-reader/manage.py", line 8, in <module>
    from django.core.management import execute_from_command_line
ModuleNotFoundError: No module named 'django'
@dsmr:~$ sudo -u dsmr /home/dsmr/.virtualenvs/dsmrreader/bin/python3 /home/dsmr/dsmr-reader/manage.py collectstatic --noinput
Traceback (most recent call last):
  File "/home/dsmr/dsmr-reader/manage.py", line 8, in <module>
    from django.core.management import execute_from_command_line
ModuleNotFoundError: No module named 'django'

```

rodeho, 2021-02-02:
```
@dsmr:~$ python
Python 2.7.18 (default, Aug  4 2020, 11:16:42)
[GCC 9.3.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.

```

Zou de python versie oud zijn?

rodeho, 2021-02-02:
nog even in #1241 gekeken maar dat helpt niet

```
dsmr@dsmr:~/dsmr-reader$ virtualenv ~/.virtualenvs/dsmrreader  --python python3
created virtual environment CPython3.8.5.final.0-64 in 249ms
  creator CPython3Posix(dest=/home/dsmr/.virtualenvs/dsmrreader, clear=False, global=False)
  seeder FromAppData(download=False, distro=latest, setuptools=latest, pyparsing=latest, pep517=latest, six=latest, progress=latest, lockfile=latest, msgpack=latest, packaging=latest, urllib3=latest, chardet=latest, html5lib=latest, CacheControl=latest, webencodings=latest, ipaddr=latest, certifi=latest, distlib=latest, pkg_resources=latest, contextlib2=latest, pip=latest, appdirs=latest, requests=latest, retrying=latest, wheel=latest, idna=latest, pytoml=latest, colorama=latest, via=copy, app_data_dir=/home/dsmr/.local/share/virtualenv/seed-app-data/v1.0.1.debian)
  activators BashActivator,CShellActivator,FishActivator,PowerShellActivator,PythonActivator,XonshActivator
(dsmrreader) dsmr@dsmr:~/dsmr-reader$ ./deploy.sh

 --- You are currently running version:
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/dsmr/dsmr-reader/dsmrreader/__init__.py", line 1, in <module>
    from django.utils.version import get_version
ModuleNotFoundError: No module named 'django'

     [!] FAILED to call Django (did you activate the 'dsmrreader' VirtualEnv before running?)

[!] Halted in pre-deployment: Either user ABORTED or script FAILED
(dsmrreader) dsmr@dsmr:~/dsmr-reader$ exit
logout
rodeho@dsmr:~$ sudo supervisorctl restart all
dsmr_backend: stopped
dsmr_backend: ERROR (spawn error)
dsmr_webinterface: ERROR (spawn error)
rodeho@dsmr:~$ sudo su - dsmr
```

dennissiemensma, 2021-02-02:
Bedankt voor je update! Ik denk dat je laatst comment aardig in de buurt komt. 

Je kunt daar waarschijnlijk **geen** installatie doen met een `deploy`, omdat die eerst controleert of Django geinstalleerd is. Je zult daarom een deel van de installatie moeten volgen. 

Dat heeft overeenkomsten met een major upgrade, waarbij meestal de virtualenv (met Django) verwijderd wordt en opnieuw aangemaakt met een nieuwe(re) Python-versie, vergelijkbaar met wat je hebt gedaan: https://dsmr-reader.readthedocs.io/en/v4/faq/v4_upgrade.html#switch-dsmr-reader-to-v4-x

Probeer dit eens:

```
sudo su - dsmr
pip3 install -r dsmrreader/provisioning/requirements/base.txt

./deploy.sh
```

rodeho, 2021-02-02:
Ha @dennissiemensma helaas, die had ik ook al geprobeerd; het draaien van de volledige installatie.

PIP is duidelijk in de war.

```
@dsmr:~$ sudo su - dsmr
(dsmrreader) dsmr@dsmr:~/dsmr-reader$ pip3 install -r dsmrreader/provisioning/requirements/base.txt
Traceback (most recent call last):
  File "/home/dsmr/.virtualenvs/dsmrreader/bin/pip3", line 5, in <module>
    from pip._internal.cli.main import main
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/pip/_internal/cli/main.py", line 10, in <module>
    from pip._internal.cli.autocompletion import autocomplete
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/pip/_internal/cli/autocompletion.py", line 9, in <module>
    from pip._internal.cli.main_parser import create_main_parser
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/pip/_internal/cli/main_parser.py", line 7, in <module>
    from pip._internal.cli import cmdoptions
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/pip/_internal/cli/cmdoptions.py", line 28, in <module>
    from pip._internal.models.target_python import TargetPython
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/pip/_internal/models/target_python.py", line 3, in <module>
    from pip._internal.pep425tags import get_supported, version_info_to_nodot
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/pip/_internal/pep425tags.py", line 7, in <module>
    from pip._vendor.packaging.tags import (
ModuleNotFoundError: No module named 'pip._vendor.packaging.tags'
(dsmrreader) dsmr@dsmr:~/dsmr-reader$
```

dennissiemensma, 2021-02-02:
Werkt deze ook niet?

```
sudo su - dsmr
pip3 install pip --upgrade
```

dennissiemensma, 2021-02-02:
Mijn test was overigens met een schone 20.04 installatie, dus wellicht dat een upgrade niet helemaal lekker alles meeneemt.

Als je toch een kloon hebt kun je zoiets proberen, al kan het meer stuk maken:

```
sudo apt-get remove python3-pip
sudo apt-get install python3-pip
```

Daarna de virtualenv van dsmrreader weer verwijderen en opnieuw aanmaken. Daarin dan kijken of pip dan beter werkt.

rodeho, 2021-02-03:
@dennissiemensma helaas geen van allen werkt, het enige wat tot resultaat heeft geleid is het verwijderen van de gehele .virtenv map. En dan de volledige https://dsmr-reader.readthedocs.io/en/v4/installation.html guide door lopen. Data is bewaard gebleven omdat dit in de postgress DB zit, dit ter info voor anderen die tegen het probleem aanlopen.

dennissiemensma, 2021-02-03:
Bedankt voor je aanvulling. Bedoel je dat je het nu wel werkend hebt gekregen?

rodeho, 2021-02-03:
> Bedankt voor je aanvulling. Bedoel je dat je het nu wel werkend hebt gekregen?

yes, .virtenv map in home/dsmr/ verwijderen en dan de installatie commandos uit de link weer runnen. De door jouw genoemde commandos boden geen oplossing, het lijkt op een corrupt bestand in de .virtenv map oid.

mjanssens, 2021-09-02:
Bij upgrade van debian buster naar bullseye soortgelijke problemen in de virtualenv.
Verwijderen en opnieuw aanmaken van een virtualenv is de oplossing, zoals https://github.com/dsmrreader/dsmr-reader/issues/1226#issuecomment-772314269
Relatief simpele handelingen die inderaad uit de documentatie te halen zijn.

dennissiemensma, 2021-09-03:
Bedankt voor je toevoeging en bevinding!