dsmrreader/dsmr-reader #1227: Importeren oude csv dagtotalen naar andere DSMR reader

bottle12, 2020-12-04:
Heb issues gehad met m'n sd kaartje. Heb daarom op dat moment besloten om nieuwe installatie te maken van DSMR-reader op een nieuwe pi.  De oude sd aan de kant gelegd om later nog eens te bekijken. Is uiteindelijk gelukt om data veilig te stellen en deze van versie 2 naar versie 4 te migreren. Heb nu dus 2x versie 4.x op twee rpi's. Export kunnen. maken van de oude dagtotalen  en die zou ik graag willen importeren naar de nieuwe in.
Zijn daar mogelijkheden voor? 

ik hoor het graag.

Daarnaast zou ik graag willen weten hoe ik het supertoffe werk dat je gemaakt hebt als blijk van waardering financieel kan supporten. 
 
mvg
Martijn

dennissiemensma, 2020-12-04:
Bedankt voor je vraag. Erg fijn dat je nog je oude data hebt kunnen redden. In dat geval is het vermoedelijk 10 minuutjes werk om het goed te krijgen. Je kunt in #1145 een soortgelijke flow vinden, al zul je daar zelf wat moeten knutselen met het ID-veld dat uit de dump moet.

Echter, als je in jouw geval ook nog bij de oude uurtotalen kan, kan ik je aanraden om die ook nog over te zetten. En als je oude database uberhaupt nog beschikbaar is, kun je overwegen **onderstaande commands** te gebruiken om een dump te maken die je wat makkelijker kan inlezen.

----

# (!) Wat je ook doet, zorg dat je eerst een extra backup maakt van je doel-database voordat je iets inleest!
```
# Op het systeem waar je het gaat inlezen:
sudo su - dsmr
./manage.py dsmr_backup_create --full --compact

# En bewaar deze backups buiten je Pi, kan altijd handig zijn. In de output staan als het goed is de paden.
```

----

Het inlezen van data zonder ID's kan nog wel stuiten op andere bestaande data, bijvoorbeeld als dezelfde dag of uur er al in staat. Je kunt de SQL-bestanden naar wens aanpassen en regels met bestaande records verwijderen. Als dat er tientallen of meer zijn, dan zullen we even een andere manier moeten bedenken om het over te zetten.

## Dagstatistieken
```
# Dagstatistieken uit oude database ZONDER id kolom:
echo "COPY public.dsmr_stats_daystatistics (day, total_cost, electricity1, electricity2, electricity1_returned, electricity2_returned, electricity1_cost, electricity2_cost, gas, gas_cost, average_temperature, highest_temperature, lowest_temperature, fixed_cost) FROM stdin;" > day_statistics_dump.sql
psql -d dsmrreader -c "COPY public.dsmr_stats_daystatistics (day, total_cost, electricity1, electricity2, electricity1_returned, electricity2_returned, electricity1_cost, electricity2_cost, gas, gas_cost, average_temperature, highest_temperature, lowest_temperature, fixed_cost) TO stdout" >> day_statistics_dump.sql

# Kopieer day_statistics_dump.sql naar waar je nieuwe database staat.

# Dagstatistieken naar nieuwe database:
psql -f day_statistics_dump.sql -d dsmrreader
psql -d dsmrreader
SELECT setval(pg_get_serial_sequence('"dsmr_stats_daystatistics"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_stats_daystatistics";
```

## Uurstatistieken
```
# Uurstatistieken uit oude database ZONDER id kolom:
echo "COPY public.dsmr_stats_hourstatistics (hour_start, electricity1, electricity2, electricity1_returned, electricity2_returned, gas) FROM stdin;" > hour_statistics_dump.sql
psql -d dsmrreader -c "COPY public.dsmr_stats_hourstatistics (hour_start, electricity1, electricity2, electricity1_returned, electricity2_returned, gas) TO stdout" >> hour_statistics_dump.sql

# Kopieer hour_statistics_dump.sql naar waar je nieuwe database staat.

# Uurstatistieken naar nieuwe database:
psql -f hour_statistics_dump.sql -d dsmrreader
psql -d dsmrreader
SELECT setval(pg_get_serial_sequence('"dsmr_stats_hourstatistics"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_stats_hourstatistics";
```

----

Voorbeeld:
```
postgres@pc:~$ createdb -O dsmrreader test_import

# Puur voor deze test omdat het een lege database is zonder tabelstructuur.
postgres@pc:~$ psql -f schema.sql test_import 

postgres@pc:~$ echo "COPY public.dsmr_stats_daystatistics (day, total_cost, electricity1, electricity2, electricity1_returned, electricity2_returned, electricity1_cost, electricity2_cost, gas, gas_cost, average_temperature, highest_temperature, lowest_temperature, fixed_cost) FROM stdin;" > day_statistics_dump.sql

postgres@pc:~$ psql -d dsmrreader -c "COPY public.dsmr_stats_daystatistics (day, total_cost, electricity1, electricity2, electricity1_returned, electricity2_returned, electricity1_cost, electricity2_cost, gas, gas_cost, average_temperature, highest_temperature, lowest_temperature, fixed_cost) TO stdout" >> day_statistics_dump.sql

postgres@pc:~$ psql -f day_statistics_dump.sql -d test_import 
COPY 1794

postgres@pc:~$ psql -d dsmrreader
psql (10.14 (Ubuntu 10.14-0ubuntu0.18.04.1))
Type "help" for help.

test_import=# SELECT setval(pg_get_serial_sequence('"dsmr_stats_daystatistics"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_stats_daystatistics";
 setval 
--------
   1794
(1 row)
```

bottle12, 2020-12-06:
het is allemaal gelukt super bedankt voor je uitleg.
Is het een idee om deze uitleg in de documentatie op te nemen. Als zijnde
importeer  oude statistieken naar nieuwe omgeving oid.
heb het commando wel iets aangepast btw "sudo -u postgres  psql ..." . ik
kreeg anders foutmelding dat dsmr geen role had.
Ik ben echt super happy met je applicatie :) en waardeer je skill/effort
enorm.

mvg
martijn

dennissiemensma, 2020-12-06:
Bedankt voor je terugkoppeling. Ik zal het toevoegen aan de documentatie voor de volgende release.

closed 2020-12-06 by dennissiemensma