dsmrreader/dsmr-reader #1228: DSMR-Reader werkt niet meer 502 Bad Gateway

TheCrow-zz, 2020-12-05:
Sinds een paar dagen draai ik DSMR-Reader op mijn Raspberry pi 4.
Vanmorgen draaide alles nog prima, een paar uur later zag ik dat bij de dashboard de 'Consumed WATT' op 0 stond en de 'Returned WATT' rond de 400 WATT schommelde, ik weet vrijwel zeker dat hij deze waardes door elkaar haalt. 
Ik heb Gegoogled en de volgende tips al geprobeerd:

Na een 'sudo reboot' command in de shell krijg ik een '502 Bad Gateway- nginx/1.14.2' foutmelding.

Bij command: 'sudo supervisorctl'
dsmr_backend                     RUNNING   pid 569, uptime 0:08:29
dsmr_datalogger                  RUNNING   pid 568, uptime 0:08:29
dsmr_webinterface                FATAL     Exited too quickly (process log may have details)

En bij command: 'sudo supervisorctl start dsmr_webinterface: dsmr_webinterface: ERROR (spawn error)

Onderstaand ook geprobeerd:
(dsmrreader) dsmr@raspberrypi:~/dsmr-reader $ ./manage.py runserver
Performing system checks...

System check identified no issues (0 silenced).
December 05, 2020 - 11:39:56
Django version 3.1.3, using settings 'dsmrreader.settings'
Starting development server at http://127.0.0.1:8000/
Quit the server with CONTROL-C.

Een nieuwe 'sudo reboot' poging helpt ook niet.

dennissiemensma, 2020-12-05:
Bedankt voor je melding. Ik denk dat je twee issues hebt. Voor de verkeerde waardes kun je het beste een telegram delen zodat we kunnen kijken wat er in staat. Deze kun je vinden onderaan de pagina van deze URL:
```
/admin/dsmr_datalogger/meterstatistics/1/change/
```

Voor het andere issue, vermoed ik dat het PID-bestand nog in de weg zit. Je kunt eerst dit proberen: https://github.com/dsmrreader/dsmr-reader/issues/1214#issuecomment-733248513. Als dat het niet oplost hoor ik het graag.

TheCrow-zz, 2020-12-05:
Na de succesvolle installatie van dsmr-reader, had ik een backup/image gemaakt van de microSD kaart, deze heb ik inmiddels teruggezet en nu werkt hij weer. Het zou dus zomaar kunnen dat ik dit probleem weer over een paar dagen krijg, ik ga dan je bovenstaand advies proberen. 
Bedankt.

closed 2020-12-17 by dennissiemensma