dsmrreader/dsmr-reader #1230: Medlding "Dagstatistieken lopen achter"

erikkral, 2020-12-10:
VERVANG DEZE TEKST MET TOELICHTING OP DE FOUTSITUATIE EN JE SETUP/DEBUGINFO-DUMP

Al een tijdje draai ik DSMR reader in een Docker op Ubuntu. Vanochtend zag ik dat er al twee dagen geen nieuwe gegevens meer binnenkwamen in DSMR reader. Dit probleem heb ik wel vaker en lijkt meer te liggen aan dat de connectie tussen de slimme meter en Ubuntu wegvalt. Ik heb vanochtend Ubuntu herstart en toe kwamen de nieuwe telegrammen weer binnen. 
Helaas blijf ik tegen deze foutmelding aanlopen:
[image]

Debug info dump:
```
DSMR-READER
    App / Python / Database                                                        v4.9 / v3.9.0 / postgresql
    Backend sleep / Datalogger sleep / Retention cleanup                                     1.0s / 0.5s / -h
    Telegram latest version read / Parser settings                                                 "42" / "4"

DATA
    Telegrams total (est.)                                                                            1325220
    Consumption records electricity / gas (est.)                                                219134 / 3376

UNRESOLVED ISSUES
    Dagstatistieken lopen achter                                                      2 dagen, 11 uur geleden
```

Het probleem zou veroorzaakt kunnen zijn dat ik vanochtend een update heb gedaan van de docker container zonder dat ik de database heb gestopt. 
Wat kan ik doen om deze foutmelding te verhelpen? 
Alles lijkt verder goed te zijn want de data sinds vanochtend word gewoon vertoond in de live grafieken. Een herstart van DSMR lijkt in ieder geval niet te werken...

dennissiemensma, 2020-12-10:
Bedankt voor je melding. Je kunt het beste DEBUG logging inschakelen en kijken naar de log dan `dsmr_backend`. 

In #1197 staat meer informatie over hoe en wat en in dit comment staat welke melding je zoekt in de logs: https://github.com/dsmrreader/dsmr-reader/issues/1197#issuecomment-724968648. De oorzaak is vermoedelijk anders dan in dat issue, maar het geeft je een idee waar je als eerste kan kijken.

erikkral, 2020-12-11:
Ik had wat moeite om de debug logging aan te zetten, ben nog lerende met docker.
Toen ik vanochtend naar de DSMR reader keek was de melding verdwenen:

[image]

Per saldo heb ik er niets aan gedaan.
Loopt er een nachtverwerking o.i.d. wat dit probleem automatisch oplost?

dennissiemensma, 2020-12-11:
Bedankt voor je update. De dagstatistieken worden inderdaad een paar uur na middernacht gemaakt. Omdat dit doorgaans elke nacht gebeurt, zouden die maximaal iets meer dan een dag achter moeten lopen, maar niet meerdere dagen.

Wellicht hing het backend-proces en heeft een herstart dat opgelost. Zolang metingen binnenkomen en opgeslagen worden, kan DSMR-reader altijd met terugwerkend kracht alles alsnog verwerken. Vermoedelijk is dat hier ook gebeurd.

closed 2020-12-17 by dennissiemensma