dsmrreader/dsmr-reader #1231: Dashboard laatste update altijd 46 seconden geleden

Diondk, 2020-12-10:
Hoi,

Draai DSMR via docker, met een datalogger op een raspberry pi in de meterkast.
Werkt allemaal perfect. 

Maar ik heb gisteren geupdate en het viel me op dat dashboard de laatste update altijd op 46 seconds ago blijft staan. Terwijl ik het gebruik wel constant zie verspringen. 

Het is mij niet eerder opgevallen, dus weet ook niet 100% zeker of dit met de laatste update is mee gekomen. 

[image]

```

DSMR-READER
    App / Python / Database                                                        v4.9 / v3.9.0 / postgresql
    Backend sleep / Datalogger sleep / Retention cleanup                                   1.0s / 5.0s / 168h
    Telegram latest version read / Parser settings                                                 "50" / "4"

DATA
    Telegrams total (est.)                                                                             500617
    Consumption records electricity / gas (est.)                                                 16479 / 9196

```

Groet,
Dion

dennissiemensma, 2020-12-11:
Bedankt voor je melding. Ik begrijp dat het wat verwarrend overkomt. Als je op die plek op het `(?)` klikt staat er een beknopte uitleg, maar wellicht is die nog niet duidelijk genoeg.

DSMR-reader gebruikt hier het tijdstip van de slimme meter, wat in elk telegram staat. Ter vergelijking, bij mijn eigen slimme meter krijg ik ook tekens te zien dat die op z'n minst "25 seconden geleden" ontvangen is. Ik kan me wel herinneren dat dat niet altijd zo is geweest bij mezelf, dus misschien worden die interne klokken van de meters periodiek bijwerkt of iets. (Of helemaal niet)
Er zijn overigens ook gebruikers waarbij het zelfs andersom is: De slimme meter draait dan op een tijd iets in de toekomst.

Ik heb hier dan ook niet echt een goede oplossing voor. Uiteindelijk was een van de doelen van DSMR-reader om simpelweg de meter uit te lezen en dit komt er blijkbaar terug. En ik denk dat het belangrijk genoeg is om dat te blijven doen, want er zijn zoveel manieren om een telegram in DSMR-reader te krijgen, dat gebeurt niet altijd "direct". Dan kan het verwarrend zijn om een andere tijd te weergeven.

Je kunt het overigens zelf zien door onderaan deze pagina te kijken: 
```
/admin/dsmr_datalogger/meterstatistics/1/change/
```

Daar staat als het goed is telkens je laatste telegram en de derde regel is doorgaans zoiets:
```
0-0:1.0.0(201211213459W)
```
Hier kun je lezen hoe je daar het tijdstip uithaalt: https://github.com/dsmrreader/dsmr-reader/issues/1225#issuecomment-740960428

dennissiemensma, 2020-12-11:
Simultaan kun je trouwens zoiets doen op de Pi of waar je DSMR-reader hebt geinstalleerd, ter vergelijking:
```
watch -n1 date
```

dennissiemensma, 2020-12-17:
@Diondk heeft bovenstaande je geholpen? 

Er komt sowieso op termijn een feature waarmee je DSMR-reader de tijd van "binnenkomen" kan laten gebruiken ipv wat in het telegram staat. Mogelijk is dat ook op termijn een workaround.

Diondk, 2020-12-18:
Ja zeker, dit was voldoende uitleg, zolang het maar klopt dan is het prima. Was mij niet eerder opgevallen dus vandaar ook de vraag.

Ik zal het issue sluiten! Bedankt!

closed 2020-12-18 by Diondk