dsmrreader/dsmr-reader #1232: Enkel tarief problemen

dgreebe, 2020-12-14:
De data van de remote datalogger lijkt wel te kloppen als het gaat om de GAS stand. Electra klopt ook, maar bij enkel tarrief klopt te prijs niet. De prijs blijft hangen als HOOG tarief start. 
Verbruik word wel goed gemeten maar de prijs klopt niet.

dennissiemensma, 2020-12-14:
Bedankt voor je bericht. Ik denk dat je het beste een screenshot en een telegram kan delen om te verduidelijken wat je exact bedoelt.

dennissiemensma, 2020-12-17:
@dgreebe heb je aanvullende informatie, zodat ik je beter kan helpen?

dgreebe, 2020-12-18:
mijn energie leverancier biedt enkel tarief aan, maar mijn meter is voor dubbel.
In de DSMR heb ik nu bij beide tarieven hetzelfde bedrag staat en dat is nu opgelost. 
Tevens heb ik ook de datalogger die remote draait geupdate. (nu krijg ik alleen geen gas metingen meer binnen)....

dgreebe, 2020-12-19:
issue kan dicht. probleem met enkel tarief is niet meer van toepassing.

closed 2020-12-19 by dgreebe