dsmrreader/dsmr-reader #1233: Toestaan om datumtijd te overschrijven

dennissiemensma, 2020-12-17:
Dit voor de situaties waarbij de slimme meter niet helemaal (of absurd ver) uit sync loopt met de klok. Gebruik op eigen risico uiteraard.

mlakerveld, 2020-12-17:
Bedankt dat je dit wil toevoegen!
 
Ter info:
Vanaf eind oktober heeft Enexis een storing waarbij slimme meters niet uitgelezen kunnen worden op afstand. Geen ramp voor DSMR Reader gebruikers uiteraard tot hier aan toe.
Behalve dan dat ze ook nieuwe slimme meters niet kunnen instellen.
Mensen die dus vanaf oktober een nieuwe slimme meter hebben gekregen bij Enexis hebben pech, want de datumtijd zal dus niet goed staan totdat hun eigen storing is verholpen.
Er is geen ETA wanneer de storing voorbij is....

Vind het gek dat ik eind november een nieuwe kreeg omdat mijne niet uit te lezen was, terwijl sinds oktober er geen een uit te lezen is 🤔.

dennissiemensma, 2021-01-10:
Gaat mee met volgende release.

closed 2021-01-10 by dennissiemensma