dsmrreader/dsmr-reader #1234: Geen gasmetingen in frontend

dgreebe, 2020-12-19:
```

DSMR-READER
    App / Python / Database                                                        v4.9 / v3.9.0 / postgresql
    Backend sleep / Datalogger sleep / Retention cleanup                                     1.0s / 0.5s / -h
    Telegram latest version read / Parser settings                                                 "50" / "4"

DATA
    Telegrams total (est.)                                                                           22679426
    Consumption records electricity / gas (est.)                                               320451 / 64007

POSTGRESQL SIZE OF LARGEST TABLES (> 500 MB)
    public.dsmr_datalogger_dsmrreading                                                                4356 MB

```

Eerder had ik eennprobleem met enkel tarief, dat is opgelost dmv een setting. Gelijk de datalogger geupdate voor de zekerheid echter sinds dien krijg ik geen gas stats in de frontend te zien. Ik zie dat deze wel worden gemeten want de data komt wel in influxdb en kan ik uitlezen met grafana.

De grafieken zijn compleet weg. Er lijkt geen referentie meer te zijn met gas oid.

dennissiemensma, 2020-12-19:
Bedankt voor je melding. Wat is hier de output van:

```
sudo su - postgres
psql dsmrreader

select read_at from dsmr_consumption_gasconsumption order by read_at desc limit 5;
```
----

Bedoel je trouwens dat de gasmetingen **via DSMR-reader** goed bij Influx komen? Of gaat dat via een andere weg?

Je zou ook nog kunnen kijken of gasgrafieken handmatig uit staat bij de checkbox onderaan deze pagina: 
```
/admin/dsmr_backend/backendsettings/1/change/
```

----

De remote datalogger hoef je overigens niet te updaten als die eerder goed werkte. Houd er rekening mee dat elke nieuwe versie daarvan subtiele verschillen kan hebben in de configuratie, zoals nieuwe of andere envvars.

dennissiemensma, 2020-12-28:
@dgreebe ben je hier nog verder mee gekomen?

dennissiemensma, 2021-01-04:
Reageer gerust op een later moment alsnog.

closed 2021-01-04 by dennissiemensma