dsmrreader/dsmr-reader #1235: port issue bij installatie

bennybubble, 2020-12-21:
Ls,

Vanwege installatie slimme meter wil ik deze gaan uitlezen met een RPI3 en dsmrreader leek mij een uitermaat geschikte applicatie hiervoor.
Tijdens installatie van part 1 loop ik tegen een probleem aan: port 80 in use (klopt want er draait nog wat anders en dat is niet te veranderen), hoe kom ik hier ***tijdens*** installatie omheen?

Foutmelding:

_Setting up nginx-full (1.10.3-1+deb9u5) ...
Job for nginx.service failed because the control process exited with error code.
See "systemctl status nginx.service" and "journalctl -xe" for details.
invoke-rc.d: initscript nginx, action "start" failed.
nginx.service - A high performance web server and a reverse proxy server
   Loaded: loaded (/lib/systemd/system/nginx.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2020-12-21 12:14:43 CET; 61ms ago
     Docs: man:nginx(8)
  Process: 30964 ExecStart=/usr/sbin/nginx -g daemon on; master_process on; (code=exited, status=1/FAILURE)
  Process: 30961 ExecStartPre=/usr/sbin/nginx -t -q -g daemon on; master_process on; (code=exited, status=0/SUCCESS)
Dec 21 12:14:41 utilpi nginx[30964]: nginx: [emerg] bind() to 0.0.0.0:80 failed (98: Address already in use)
Dec 21 12:14:41 utilpi nginx[30964]: nginx: [emerg] bind() to 0.0.0.0:80 failed (98: Address already in use)
Dec 21 12:14:42 utilpi nginx[30964]: nginx: [emerg] bind() to 0.0.0.0:80 failed (98: Address already in use)
Dec 21 12:14:42 utilpi nginx[30964]: nginx: [emerg] bind() to 0.0.0.0:80 failed (98: Address already in use)
Dec 21 12:14:43 utilpi nginx[30964]: nginx: [emerg] bind() to 0.0.0.0:80 failed (98: Address already in use)
Dec 21 12:14:43 utilpi nginx[30964]: nginx: [emerg] still could not bind()
Dec 21 12:14:43 utilpi systemd[1]: nginx.service: Control process exited, code=exited status=1
Dec 21 12:14:43 utilpi systemd[1]: Failed to start A high performance web server and a reverse proxy server.
Dec 21 12:14:43 utilpi systemd[1]: nginx.service: Unit entered failed state.
Dec 21 12:14:43 utilpi systemd[1]: nginx.service: Failed with result 'exit-code'.
dpkg: error processing package nginx-full (--configure):
 subprocess installed post-installation script returned error exit status 1_

Als alles goed is geinstalleerd kun je het portnummer wel veranderen maar zover kom ik niet:
https://github.com/dsmrreader/dsmr-reader/issues/169

Alvast bedankt!

Ben

jeroenpeters1986, 2020-12-21:
hey @bennybubble ,

weet je ook wat er nog draait? Ik heb een vermoeden van Apache2. Als dat inderdaad het geval is kun je vóór de installatie even dit intikken:
`sudo service apache2 stop` en dan stopt apache even.

Na de installatie kun je inderdaad de poort voor nginx aanpassen. Daarna zul je nginx wel even moeten herstarten. Ik hoop dat dat lukt, anders is daar wel veel over te vinden op internet.

Na installatie, het aanpassen van de poort en herstarten van nginx, kun je dan Apache weer starten met `sudo service apache2 start`.

Ik hoop dat het hiermee lukt.

bennybubble, 2020-12-21:
Bedankt, was niet apache2 maar heb de betreffende service uit kunnen zetten en toen ging dat stuk van de installatie door.
-ben

closed 2020-12-21 by bennybubble

jeroenpeters1986, 2020-12-21:
Ah oke, fijn dat het toch gelukt is :+1: