dsmrreader/dsmr-reader #1236: 3.11 upgrade

dennis071nl, 2020-12-21:
Hoi mannen, 

Ik probeer mn 3.11 installatie te upgraden maar bij stap 1 loop ik al vast. Ik werk op basis van docker, dsmr en dsmrdb draaien in principe an sich wel. 

de guide (https://dsmr-reader.readthedocs.io/nl/v4/faq.html#how-to-upgrade) loop ik al vast bij stap 1...:-(

De foutmelding die ik krijg is: 

 --- You are currently running version: 
3.11

 --- Checking Python version.
Running Python 3.8.3 (default, Jun  3 2020, 19:49:40) 
[GCC 9.3.0]

 --- Pulling remote repository for new commits...
./deploy.sh: line 30: git: command not found

 --- Merging/updating checkout.
./deploy.sh: line 36: git: command not found

 >>> Running post-deployment script. <<<

 --- Checking whether VirtualEnv is activated.
     [i] ----- Activating 'dsmrreader' VirtualEnv...
./post-deploy.sh: line 10: /root/.virtualenvs/dsmrreader/bin/activate: No such file or directory
     [!] FAILED to switch to 'dsmrreader' VirtualEnv (is it installed?)
[!] Halted. Post-deployment script exited with non-zero code

Hoop dat jullie me een beetje in de goeie richting kunnen duwen!

Thanks in advance, D

dennissiemensma, 2020-12-21:
Bedankt voor je vraag. Vermoedelijk hoef je bij docker alleen de dsmreader app container te updaten naar de laatste versie. Alle technische wijzigingen zouden al in de nieuwe container zelf gedaan moeten zijn. Bij de deploy wordt dan je bestaande database container bijgewerkt zodat die compatibel is met de nieuwere versie. @xirixiz nog aanvullingen?

dennis071nl, 2020-12-21:
Dennis, ik heb nu als image in mn docker-compose " xirixiz/dsmr-reader-docker:latest ", zou dit niet afdoende moeten zijn dan?

dennissiemensma, 2020-12-21:
Ik ben zelf helaas niet zo bedreven met docker. Ik ken het oppervlakkig, helemaal voor dsmr reader. Wellicht wordt je image nog gecached of iets?

xirixiz, 2020-12-21:
Heren, wellicht kan je deze stappen volgen 😉 

https://github.com/xirixiz/dsmr-reader-docker#upgrade-the-postgres-container-to-a-newer-release

xirixiz, 2020-12-21:
Ach, ik lees het nog verkeerd ook. Je kan inderdaad gewoon naar de laatste versie van de Docker image en dan gaat het upgraden vanzelf. Voor de zekerheid zou ik wel eerst een database dump maken zodat je iig een backup hebt voordat je upgrade.

https://github.com/xirixiz/dsmr-reader-docker#backup-and-restore-mechanism-1
https://github.com/xirixiz/dsmr-reader-docker#backup-and-restore-mechanism-2

dennissiemensma, 2020-12-22:
@dennis071nl zie bovenstaande tips, is het inmiddels gelukt?

dennis071nl, 2020-12-22:
Dennis,

Ik heb alle data weggegooid en helemaal fris opnieuw begonnen... inmiddels
draai ik met 4.9!

Dank voor jullie hulp!

dennissiemensma, 2020-12-22:
Oke, mocht je nog je oude data ergens hebben, je kunt de dag- en uurstatistieken vrij makkelijk overzetten hoor!

closed 2020-12-27 by dennissiemensma