dsmrreader/dsmr-reader #1237: FATAL: password authentication failed for user "dsmrreader"

Raarrrinator, 2020-12-22:
Goedenavond!

Ik probeer eindeloos DSMR geïnstalleerd te krijgen in een Raspbian VM op mijn NUC en wat ik probeer; stap 8 faalt steeds.

[image]

Ik heb de explained handleiding stap voor stap gevolgd en strand met de melding: FATAL: password authentication failed for user "dsmrreader"

Mijn linux kennis van het niveau knippen en plakken, dus wil ik dit opgelost krijgen, heb ik echt hulp nodig :(

Elke hulp wordt gewaardeerd (en graag in hapklare brokken!).

Bedankt alvast & groet,

Ruben

dennissiemensma, 2020-12-22:
Je kunt als `pi` user nog deze proberen. Ik neem aan dat je die wel in hoofdstuk 1 hebt uitgevoerd?

```
sudo -u postgres psql -c "alter user dsmrreader with password 'dsmrreader';"
```

Verder is het belangrijk dat je bij hoofdstuk 8 (nog) de `dsmr` user bent uit hoofdstuk 7.

```
sudo su - dsmr

./manage.py migrate
...
```

Raarrrinator, 2020-12-22:
Dankjewel voor je feedback! De migratie in stap 8 is nu gelukt. Ik vermoed dat ik bij stap 4 niet uitgelogd ben als DSMR user.

Het laatste commando van stap 8 (./manage.py dsmr_superuser) gaf wel een foutmelding:

CommandError: Either DSMRREADER_ADMIN_USER or DSMRREADER_ADMIN_PASSWORD (or both) are missing/empty in (.)env

Ik ben maar doorgegaan met 9, maar is dit achteraf nog op te lossen?

Raarrrinator, 2020-12-22:
> Je kunt als `pi` user nog deze proberen. Ik neem aan dat je die wel in hoofdstuk 1 hebt uitgevoerd?
>
> ```
> sudo -u postgres psql -c "alter user dsmrreader with password 'dsmrreader';"
> ```
>
> Verder is [...]

DSMR Reader draait, maar om in te loggen heb ik het supervisor wachtwoord nodig. Omdat de laatste fase van stap 8 aangaf dat DSMRREADER_ADMIN_USER or DSMRREADER_ADMIN_PASSWORD niet in .env te vinden zijn, kan ik niet inloggen.

Hoe kan ik DSMRREADER_ADMIN_USER or DSMRREADER_ADMIN_PASSWORD aan .env toevoegen? Ik heb het bestand gezocht, maar vind het nergens.

dennissiemensma, 2020-12-22:
In hoofdstuk 7 heb je dit gedaan:
```
cp .env.template .env
```

Als het goed is is de inhoud van `.env` (`/home/dsmr/dsmrreader/.env`) zoiets:
```
################################################################################################################
### When needed, adjust these or define them as system environment variables (the latter takes priority).    ###
###                                                                                                          ###
### For a full list of environment vars to use (or override) see:                                            ###
### https://dsmr-reader.readthedocs.io/en/latest/env_settings.html                                           ###
###                                                                                                          ###
### This config lacks a DJANGO_SECRET_KEY statement by default.                                              ###
### Run ./tools/generate-secret-key.sh to generate one.                                                      ###
################################################################################################################

### Database.
DJANGO_DATABASE_ENGINE=django.db.backends.postgresql
DJANGO_DATABASE_HOST=localhost
DJANGO_DATABASE_PORT=5432
DJANGO_DATABASE_NAME=dsmrreader
DJANGO_DATABASE_USER=dsmrreader
DJANGO_DATABASE_PASSWORD=dsmrreader

### Logging level.
###DSMRREADER_LOGLEVEL=DEBUG

### Webinterface (super)user for admin interface. Please choose your credentials carefully.
### User will be created (or its password reset) when running: ./manage.py dsmr_superuser
DSMRREADER_ADMIN_USER=
DSMRREADER_ADMIN_PASSWORD=
```

Dat bestand kun je aanpassen en uitbreiden met je eigen opties. Onderaan staan al de opties die je zoekt (leeg).

----

En als je niet weet hoe je tekstbestanden kan bewerken, is `nano` het makkelijkste:
```
sudo apt-get install nano
sudo su - dsmr
nano .env
```

Als je je bewerkingen gedaan hebt kun je daar in met `CTRL+X` > `Y` > `Enter` het bestand opslaan.

Raarrrinator, 2020-12-22:
Dankjewel, dat was me eerlijk gezegd op basis van de toegelichte handleiding niet duidelijk.

Op basis van trail & error was ik er inmiddels achter dat de laatste stap van de snelle handleiding uitvoeren ook werkt:

# Create application user
sudo sudo -u dsmr /home/dsmr/.virtualenvs/dsmrreader/bin/python3 /home/dsmr/dsmr-reader/manage.py createsuperuser --username admin --email root@localhost

Toen heb ik een password kunnen invoeren dat werkt in combinatie met user: admin

Bedankt voor de support!

dennissiemensma, 2020-12-22:
Fijn om te horen!

closed 2020-12-22 by dennissiemensma

HFeenstra, 2020-12-27:
https://github.com/dsmrreader/dsmr-reader/issues/1237#issuecomment-749795601

@Raarrrinator dank voor deze stap. Zo kreeg ik het ook aan de praat. Kan het zijn dat dit toegevoegd kan worden aan de installatie? Ik kreeg de user namelijk ook niet toegevoegd, idem melding.

Nogmaals dank!