dsmrreader/dsmr-reader #1238: PVoutput: Generated = 0

snoekieboe, 2020-12-24:
Ik gebruik DSMR reader in combinatie met PVoutput om mijn teruglevering en verbruik van energie inzichtelijk te maken.
Conform de handleiding heb ik PVoutput ingericht en mijn verbruik wordt netjes via de API tussen DSRM en PVoutput uitgewisseld.  Er is echter 1 klein probleem en dat is dat mijn "generation/generated" waarde op 0 blijft staan.

Ik zie wel teruglevering/generatie in DSMR zelf maar in de outputs op PVoutput blijft deze waarde op 0 staan.
Doe ik iets fout?

thnx!

dennissiemensma, 2020-12-25:
Bedankt voor je vraag. Je kunt het beste even #959 doorlezen.

De link naar de docs daar is een oude, deze moet je nu hebben: https://dsmr-reader.readthedocs.io/nl/v4/faq.html#log-files
Lees dat hoofdstukje en de twee eronder door voor hoe je debug logging aan en uit kan zetten en waar de logfiles staan.

Je bent daarna (kan even duren tot de volgende upload) dus op zoek naar deze info, die aangeeft wat DSMR-reader doorstuurt: https://github.com/dsmrreader/dsmr-reader/issues/959#issuecomment-610035030

dennissiemensma, 2020-12-25:
Voordat je inhoudelijk aan de slag gaat, zie trouwens het laatste comment in die thread: https://github.com/dsmrreader/dsmr-reader/issues/959#issuecomment-610507305 (#9)

Wellicht loop je tegen hetzelfde aan. DSMR-reader stuurt maar een deel van de data om de 'teruglevering' in je meter netto is,. Als daar toch behoefte aan is dan kan ik dat in een toekomstige versie alsnog wel opnemen. 
Het is voor mij wat lastig om zonder zelf zonnepanelen te hebben (en zonder PVOutput te gebruiken) in te schatten welke gegevens wel en niet wenselijk zijn daar. Vooral bruto/netto heeft in het verleden voor her en der wat verwarring gezorgd bij gebruikers, dus ik laat mij graag informeren over wat nodig is en wat niet.

dennissiemensma, 2020-12-30:
@snoekieboe ben je hier inmiddels wat wijzer door geworden? Qua PVOuput zullen jullie als gebruikers mij wat extra aanwijzingen moeten geven welke informatie waar nodig is, als de huidige opzet niet afdoende is.

dennissiemensma, 2021-01-04:
Laat maar weten als ik alsnog wat kan betekenen voor jullie.

closed 2021-01-04 by dennissiemensma