dsmrreader/dsmr-reader #1239: Django

sjaakveenendaal, 2020-12-27:
Ik krijg deze melding wat gaat er fout?

 --- You are currently running version:
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/dsmr/dsmr-reader/dsmrreader/__init__.py", line 1, in <module>
    from django.utils.version import get_version
ModuleNotFoundError: No module named 'django'

     [!] FAILED to call Django (did you activate the 'dsmrreader' VirtualEnv before running?)

dennissiemensma, 2020-12-27:
Bedankt voor je melding.

Heb je wel eerst dit gedaan?
```
sudo su - dsmr
```

sjaakveenendaal, 2020-12-27:
Ja dat heb ik gedaan

dennissiemensma, 2020-12-27:
Heb je dit (onderaan) in je `/home/dsmr/.bashrc` staan?

```
source ~/.virtualenvs/dsmrreader/bin/activate
cd ~/dsmr-reader
```

En zo ja, wat is de output van:
```
sudo su - dsmr
which python3
```

sjaakveenendaal, 2020-12-27:
vraagt om een wachtwoord en die heb ik niet aangemaakt

dennissiemensma, 2020-12-27:
Je zult altijd van de `root` user of een sudo-user naar `dsmr` moeten. Je hebt het immers ook ooit geinstalleerd met die rechten.

Als je een Pi gebruikt dan heet de sudo user vaak `pi`. Daar kun je als het goed is gewoon `sudo su - dsmr` uitvoeren. Als je niet zeker weet wie je bent, kun je altijd, op elke moment `whoami` uitvoeren.

sjaakveenendaal, 2020-12-27:
/.home/dsmr/.virtualenvs/dsmrreader/bin/python3

dennissiemensma, 2020-12-27:
Is het een bestaande installatie? Werkte het eerder wel?

Het is trouwens wel raar dat het pad begint met `/.home` ipv `/home`, of is dat wellicht een typo?

Je kunt dit nog proberen:
```
sudo su - dsmr
pip3 install -r /home/dsmr/dsmr-reader/dsmrreader/provisioning/requirements/base.txt
```

Geeft dat nog output met dat die (missende) dingen geinstalleerd heeft?

sjaakveenendaal, 2020-12-27:
inderdaad nu zonder .

Als ik pip3 install -r /home/dsmr/dsmr-reader/dsmrreader/provisioning/requirements/base.txt invoer

krijg ik een hele rij met tekst te zien beginnen allemaal met Requirment already satisfied

dennissiemensma, 2020-12-27:
En werkt dit wel?
```
sudo su - dsmr
./manage.py dsmr_debuginfo
```

Zo ja, dan dit?
```
sudo su - dsmr
./deploy.sh
```

sjaakveenendaal, 2020-12-27:
ModuleNotFoundError: No module named 'dsmrreader.settings

dennissiemensma, 2020-12-27:
Je zult echt met uitgebreidere informatie moeten komen om je beter te helpen. 

- Bij welk van de twee commands krijg je die output? 
- En is het een bestaande installatie? 
- Werkte het eerder wel? 
- Is er wel een `/home/dsmr/dsmr-reader/dsmrreader/settings.py` bestand?
- Kun je ook de input kopieren van het command zelf? Dan kan ik zien of je wel in de juiste map zit, bijvoorbeeld `(dsmrreader) dsmr@rpi4:~/dsmr-reader $`.

sjaakveenendaal, 2020-12-28:
Nieuwe installatie 

/home/dsmr/dsmr-reader/dsmrreader/settings.py 
settings.py is er niet

dennissiemensma, 2020-12-28:
Dat verklaart een hoop. Het is altijd goed om wat context te geven, want dan had ik je direct kunnen helpen.

Je hebt dan waarschijnlijk een stap overgeslagen bij de installatie. Je kunt het nogmaals proberen vanaf hoofdstuk 7: https://dsmr-reader.readthedocs.io/en/latest/installation.html#application-configuration-setup

sjaakveenendaal, 2020-12-28:
installatie opnieuw uitgevoerd nu de volledige en niet de snelle installatie gedaan werkt nu wel alleen kan ik niet inloggen.. admin en ww admin geprobeerd maar kan niet inloggen is er een mogelijkheid om wachtwoord te veranderen?

Voer de correcte gebruikersnaam en wachtwoord voor een stafaccount in. Let op dat beide velden hoofdlettergevoelig zijn.

sjaakveenendaal, 2020-12-28:
gevonden hou te wijzigen nu werkt alles weer bedankt voor je hulp Dennis

closed 2020-12-28 by dennissiemensma