dsmrreader/dsmr-reader #1240: Melding over ontbreken recente "readings" lijkt niet juist

mthmulders, 2020-12-29:
Ik draai DSMR Reader met een aparte datalogger. Als de datalogger onderbroken wordt krijg ik - vaak al na enkele minuten - een melding via Telegram:

> It has been over 12 hour(s) since the last reading received. Please check your datalogger.

Op zich heel behulpzaam, maar het is dan zeker niet 12 uur geleden dat de laatste "reading" binnenkwam. Als ik de code in **dsmr_notification/services.py** goed lees wordt die melding gestuurd wanneer de laatste "reading" langer geleden is dan **DSMRREADER_STATUS_READING_OFFSET_MINUTES** (default: 30 minuten).

Als mijn indruk klopt wil ik wel proberen een pull request te maken. Ik ben niet heel bekend met Python maar ik denk dat ik er wel uit kom.

Uitvoer van dsmr_debuginfo:
```
DSMR-READER
    App / Python / Database                                                        v4.9 / v3.6.9 / postgresql
    Backend sleep / Datalogger sleep / Retention cleanup                                  1.0s / 0.5s / 8064h
    Telegram latest version read / Parser settings                                               "None" / "3"

DATA
    Telegrams total (est.)                                                                            2502070
    Consumption records electricity / gas (est.)                                                433959 / 6711

POSTGRESQL SIZE OF LARGEST TABLES (> 500 MB)
    public.dsmr_datalogger_dsmrreading                                                                 635 MB
```

dennissiemensma, 2020-12-29:
Bedankt voor je melding en de aanvullende informatie! Ik zie dat er een verkeerde variabele gebruikt werd. De cooldown is meer bedoeld tegen te veel herhaling in korte tijd, de andere is inderdaad de 'offset' wanneer die melding naar voren komt.

Bij de volgende release is de tekst aangepast en pakt die de juiste variabele. Verder wordt de default verhoogd naar een uur, al zou die bij 30 minuten ook niet direct moeten af gaan.

dennissiemensma, 2020-12-30:
Mocht je trouwens voor die tijd de offset wat hoger willen voor je eigen installatie, dan kun je onderaan `dsmrreader/settings.py` een regel toevoegen met je eigen waarde voor `DSMRREADER_STATUS_READING_OFFSET_MINUTES`:

```
"""
    This used to be the local settings file until DSMR-reader 4.x.
    Please use .env instead. See: https://dsmr-reader.readthedocs.io/en/latest/faq/v4_upgrade.html
"""
from dsmrreader.config.production import *

DSMRREADER_STATUS_READING_OFFSET_MINUTES=180
```

Daarna doorvoeren met:
```
sudo su - dsmr
./reload.sh
```

closed 2020-12-30 by dennissiemensma