dsmrreader/dsmr-reader #1241: virtualenv: error: unrecognized arguments: --no-site-packages

eterpstra, 2020-12-30:
Hi Dennis,

Ik probeer DSMR te installeren in een LXC in Proxmox. Alles gaat goed tot de volgende stap:

`virtualenv ~/.virtualenvs/dsmrreader --no-site-packages --python python3`

Ik krijg dan gelijk de foutmelding:

```
dsmr@dsmr:~$ virtualenv ~/.virtualenvs/dsmrreader --no-site-packages --python python3
usage: virtualenv [--version] [--with-traceback] [-v | -q] [--app-data APP_DATA] [--clear-app-data] [--discovery {builtin}] [-p py] [--creator {builtin,cpython3-posix,venv}] [--seeder {app-data,pip}] [--no-seed]
                  [--activators comma_sep_list] [--clear] [--system-site-packages] [--symlinks | --copies] [--download | --no-download] [--extra-search-dir d [d ...]] [--msgpack version] [--wheel version] [--appdirs version] [--pip version]
                  [--pep517 version] [--webencodings version] [--packaging version] [--six version] [--pytoml version] [--pyparsing version] [--urllib3 version] [--pkg_resources version] [--setuptools version] [--idna version]
                  [--progress version] [--CacheControl version] [--ipaddr version] [--distro version] [--requests version] [--lockfile version] [--chardet version] [--contextlib2 version] [--colorama version] [--distlib version]
                  [--html5lib version] [--certifi version] [--retrying version] [--no-msgpack] [--no-wheel] [--no-appdirs] [--no-pip] [--no-pep517] [--no-webencodings] [--no-packaging] [--no-six] [--no-pytoml] [--no-pyparsing]
                  [--no-urllib3] [--no-pkg_resources] [--no-setuptools] [--no-idna] [--no-progress] [--no-CacheControl] [--no-ipaddr] [--no-distro] [--no-requests] [--no-lockfile] [--no-chardet] [--no-contextlib2] [--no-colorama]
                  [--no-distlib] [--no-html5lib] [--no-certifi] [--no-retrying] [--symlink-app-data] [--prompt prompt] [-h]
                  dest
virtualenv: error: unrecognized arguments: --no-site-packages
```

Wat doe ik precies fout? Het gaat om Ubuntu 20.x.x versie.

dennissiemensma, 2020-12-30:
Bedankt voor je vraag. Heb je het ook al geprobeerd zonder --site-packages?

eterpstra, 2020-12-30:
Nee nog niet, want ging er vanuit dat het nodig was.  Dit als resultaat..

```
dsmr@dsmr:~$ virtualenv ~/.virtualenvs/dsmrreader  --python python3
created virtual environment CPython3.8.5.final.0-64 in 270ms
  creator CPython3Posix(dest=/home/dsmr/.virtualenvs/dsmrreader, clear=False, global=False)
  seeder FromAppData(download=False, msgpack=latest, wheel=latest, appdirs=latest, pip=latest, pep517=latest, webencodings=latest, packaging=latest, six=latest, pytoml=latest, pyparsing=latest, urllib3=latest, pkg_resources=latest, setuptools=latest, idna=latest, progress=latest, CacheControl=latest, ipaddr=latest, distro=latest, requests=latest, lockfile=latest, chardet=latest, contextlib2=latest, colorama=latest, distlib=latest, html5lib=latest, certifi=latest, retrying=latest, via=copy, app_data_dir=/home/dsmr/.local/share/virtualenv/seed-app-data/v1.0.1.debian)
  activators BashActivator,CShellActivator,FishActivator,PowerShellActivator,PythonActivator,XonshActivator
dsmr@dsmr:~$ virtualenv ~/.virtualenvs/dsmrreader --python python3
created virtual environment CPython3.8.5.final.0-64 in 164ms
  creator CPython3Posix(dest=/home/dsmr/.virtualenvs/dsmrreader, clear=False, global=False)
  seeder FromAppData(download=False, msgpack=latest, wheel=latest, appdirs=latest, pip=latest, pep517=latest, webencodings=latest, packaging=latest, six=latest, pytoml=latest, pyparsing=latest, urllib3=latest, pkg_resources=latest, setuptools=latest, idna=latest, progress=latest, CacheControl=latest, ipaddr=latest, distro=latest, requests=latest, lockfile=latest, chardet=latest, contextlib2=latest, colorama=latest, distlib=latest, html5lib=latest, certifi=latest, retrying=latest, via=copy, app_data_dir=/home/dsmr/.local/share/virtualenv/seed-app-data/v1.0.1.debian)
  activators BashActivator,CShellActivator,FishActivator,PowerShellActivator,PythonActivator,XonshActivator
```

dennissiemensma, 2020-12-30:
Top! Ik zou zeggen probeer het eerst op deze manier. Als DSMR-reader uiteindelijk goed draait, verwacht ik er geen issues mee. Laat het ook vooral weten als het niet werkt.

----

Het argument geeft aan of (de virtualenv van) DSMR-reader wel of niet reeds geinstalleerd packages van het OS moet gebruiken. In principe is dat niet nodig, want DSMR-reader specificeert zelf al de eigen packages die nodig zijn. Voorheen kon dat via dit argument, maar ik begreep al eerder dit jaar al via https://github.com/dsmrreader/dsmr-reader/issues/1013#issuecomment-640096338 dat deze manier deprecated was. Daar loop je nu tegen aan vermoed ik.
Destijds was de tip om de tegenhanger, `--system-site-packages`, te gebruiken, maar daar liep her en der een gebruiker ook soms tegen een fout aan, doordat de OS-versies van de packages te oud waren of niet bestonden. 

----

In de volgende major release (5.x) is de kans trouwens groot dat ik het project wissel naar `poetry`. Dan hoeven we ook niet meer zelf die virtualenvs enzo aan te maken. Het is dan een kwestie van `poetry install` en die maakt dan de virtualenv aan en installeert de packages.
Hopelijk lost dat eea op, maar het zal wel even duren voordat die release er komt.

eterpstra, 2020-12-30:
Top! Het is gelukt. Heb je trouwens ook nog de path waar de database wordt opgelsagen? Deze moet ik namelijk terugzetten van een andere machine. 

Beiden draaien nu de nieuwste versie.

dennissiemensma, 2020-12-30:
In de meeste gevallen wordt de datamap van postgresql in `/var/lib/postgresql/<version>/` opgeslagen. En de config (als je die al aangepast hebt) in het gelijknamige pad in `/etc/`.

Exporteren via een dump is het meest makkelijke en veiligste:

```
sudo su - postgres
pg_dump -d dsmrreader | gzip --fast > dsmrreader-export.sql.gz

# Dump staat dan meestal in:   /var/lib/postgresql/dsmrreader-export.sql.gz
```

Importeren als je de dump op dezelfde locatie zet op de andere machine:
```
sudo us -postgres

# dropdb dsmrreader > Alleen als je via de handleiding al een nieuwe/lege had aangemaakt.
dropdb dsmrreader
createdb -O dsmrreader dsmrreader

zcat dsmrreader-export.sql.gz | psql dsmrreader
```

closed 2020-12-30 by dennissiemensma

eterpstra, 2021-01-01:
Nog even een vraag.. Ik wil de backup terugzetten via een ssh verbinding, is het dan ook gewoon mogelijk met mv command?

dennissiemensma, 2021-01-01:
Je kunt het proberen, wie weet werkt dat ook. Als het niet lukt kun je het altijd nogmaals doen via export/import op de command line.

eterpstra, 2021-01-01:
Oke. ik zit nu met het probleem dat van beide servers al een DSMR installatie is aangemaakt en dus ook een database. In de howto lees ik dat je dit alleen kan doen als er nog geen database is.

Hoe kan ik dit nu het beste aanpakken?

dennissiemensma, 2021-01-02:
Zie https://github.com/dsmrreader/dsmr-reader/issues/1241#issuecomment-752541289. Je kunt de initiele database gewoon verwijderen en een nieuwe, lege aanmaken en dan de import doen.