dsmrreader/dsmr-reader #1242: edijkstra@DSMR-Unbuntu-Server:~$ sudo sudo -u dsmr cp /home/dsmr/dsmr-reader/dsmrreader/provisioning/django/postgresql.py /home/dsmr/dsmr-reader/dsmrreader/settings.py cp: cannot stat '/home/dsmr/dsmr-reader/dsmrreader/provisioning/django/postgresql.py': No such file or directory

eterpstra, 2020-12-31:
Hi Dennis,

Helaas heb ik toch problemen bij een andere. Ik heb een Ubuntu Desktop server gemaakt bij een vriend.

Alles is geupdate maar hij stopt bij het volgende stukje:

Server@DSMR-Unbuntu-Server:~$ `sudo sudo -u dsmr cp /home/dsmr/dsmr-reader/dsmrreader/provisioning/django/postgresql.py /home/dsmr/dsmr-reader/dsmrreader/settings.py`

En gaat stuk met:

`cp: cannot stat '/home/dsmr/dsmr-reader/dsmrreader/provisioning/django/postgresql.py': No such file or directory`

Ditmaal gaat het niet om een container maar om een volledige VM van Ubuntu Desktop. 

Alle vorige stappen gingen goed, maar ik kom om 1 of andere manier niet verder.

dennissiemensma, 2020-12-31:
Bedankt voor je melding. Kan het zijn dat je een oude versie (v3) van de documentatie gebruikt? Je kunt het zien aan de URL, daar staat `v3`, `v4` of `latest` (= v4) in.

In v3 was het:
```
cp <GEKNIPT>/dsmrreader/provisioning/django/postgresql.py <GEKNIPT>/dsmrreader/settings.py
```

In v4:
```
cp <GEKNIPT>/dsmrreader/provisioning/django/settings.py.template <GEKNIPT>/dsmrreader/settings.py
```

https://dsmr-reader.readthedocs.io/en/latest/installation.html

eterpstra, 2020-12-31:
Dat was het inderdaad.. Ik heb uit het menu gewoon de snelle installatie gekozen. Wist niet dat er een andere versie was.

Thanks!

dennissiemensma, 2020-12-31:
Oke top!

closed 2020-12-31 by dennissiemensma