dsmrreader/dsmr-reader #1243: Server Error / LookupError: No electricity readings found for: date

arjenfvellinga, 2021-01-01:
Ik heb sinds vandaag DSMR Reader via docker draaien. De telegram berichten komen binnen, live grafiek en dashboard toonden data. Daarna heb ik de tarieven van mijn energieleverancier toegevoegd:

[image]

Initieel had ik tariff 1 en 2 omgedraaid, dus die heb ik omgewisseld. Daarom heb ik `./manage.py dsmr_stats_recalculate_prices` uitgevoerd zodat de wijziging met terugwerkende kracht wordt toegepast. Echter geeft de front end nu de volgende error:

```
Server Error
Sorry, something unexpected happened.
Exception:

LookupError: No electricity readings found for: 2021-01-02

Traceback:

  File "/usr/local/lib/python3.9/site-packages/django/core/handlers/base.py", line 179, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)

  File "/usr/local/lib/python3.9/site-packages/django/views/generic/base.py", line 70, in view
    return self.dispatch(request, *args, **kwargs)

  File "/dsmr/dsmr_frontend/mixins.py", line 11, in dispatch
    return super().dispatch(request, *args, **kwargs)

  File "/usr/local/lib/python3.9/site-packages/django/views/generic/base.py", line 98, in dispatch
    return handler(request, *args, **kwargs)

  File "/usr/local/lib/python3.9/site-packages/django/views/generic/base.py", line 159, in get
    context = self.get_context_data(**kwargs)

  File "/dsmr/dsmr_frontend/views/dashboard.py", line 31, in get_context_data
    context_data['consumption'] = dsmr_consumption.services.day_consumption(

  File "/dsmr/dsmr_consumption/services.py", line 246, in day_consumption
    raise LookupError("No electricity readings found for: {}".format(day))

```

`/admin` is wel bereikbaar.

Mijn idee is dat de herberekening niet juist gegaan is, enig idee hoe ik dit kan herstellen?

arjenfvellinga, 2021-01-01:
Closing this now, the script took longer than I was expecting. Front end works again.

closed 2021-01-01 by arjenfvellinga

dennissiemensma, 2021-01-02:
Bedankt voor je melding en mooi dat het al opgelost is. Gaf `dsmr_stats_recalculate_prices` trouwens geen output met hoever die was (welke dagen die mee bezig was)?

arjenfvellinga, 2021-01-02:
ik had het via de command line uitgevoerd, en die returned direct zonder output. En omdat ik nog niet 1 dag aan data had ging ik er vanuit dat het ook best snel kon zijn.