dsmrreader/dsmr-reader #1245: Python upgrade documentatie

GieltjE, 2021-01-03:
Het upgraden van b.v. pyton3.7 naar python3.8 is nog niet omschreven in de documentatie, ik vermoed dat dit met python 3.x vaker voor gaat komen (mijn systeem dropte zonder veel vragen 3.7 vorige week).

Voor zover ik gezien/gemerkt heb is het niet meer dan:
rm  /home/dsmr/.virtualenvs/dsmrreader/bin/python
sudo -u dsmr virtualenv /home/dsmr/.virtualenvs/dsmrreader --system-site-packages --python python3
sudo -u dsmr /home/dsmr/.virtualenvs/dsmrreader/bin/pip3 install -r /home/dsmr/dsmr-reader/dsmrreader/provisioning/requirements/base.txt
sudo supervisorctl restart all

Heb ik nog iets gemist? Zie wel dat er b.v. in /home/dsmr/.virtualenvs/dsmrreader/bin/ wat 3.7 referenties over blijven bijvoorbeeld.

dennissiemensma, 2021-01-03:
Bedankt voor je vraag. Je kunt het beste de *hele* virtualenv helemaal weggooien en opnieuw aanmaken met de nieuwe python versie. Zie de oude dsmrreader 3.x upgrade hoofdstuk 5 voor een voorbeeld. https://dsmr-reader.readthedocs.io/en/v4/faq/v3_upgrade.html#python-version-upgrade-part-2-2

GieltjE, 2021-01-03:
Super, komt deze docu er ook nog voor v4 bij te staan?

dennissiemensma, 2021-01-04:
Ik voeg het meestal toe bij een major upgrade van dsmr-reader, wanneer Python zelf ook EOL gaat, omdat bij een major update ook handwerk nodig is. Of als er heel veel vragen over komen. Het lijkt erop dat zelfs Python 3.6 nog tot eind dit jaar ondersteund wordt: https://devguide.python.org/#status-of-python-branches

Ik wil verder ook zoveel mogelijk vasthouden aan de Python-versies van elke distro, om te voorkomen dat men zelf moet gaan compilen om eerder te updaten. En degenen die dat wel eerder willen, zoals in jouw geval, zijn vaak technisch genoeg om het (vlot) op te lossen en waar nodig assisteer ik.
Bij een OS-upgrade zit er namelijk vaak ook al een Python upgrade in en bij de eerstvolgende major DSMR-reader update, [bump ik dan de minimale Python-versie](https://dsmr-reader.readthedocs.io/en/latest/intro.html#software-requirements), met de instructies die je nu ook gezien hebt.

closed 2021-01-04 by dennissiemensma

mkruiver, 2021-09-20:
Dus er komt zo te lezen geen advies om te upgraden naar Python 3.8 of 3.9, maar een user zou het wel kunnen doen, analoog aan de 3.7 upgrade?

dennissiemensma, 2021-09-20:
@mkruiver het advies is altijd om de laatst ondersteunde versie te gebruiken en bij een install op een relatief nieuw OS is dat meestal al een van de nieuwste (bijvoorbeeld Python 3.9).

Je kunt het moment ook aangrijpen bij een herinstallatie van DSMR-reader en, als het OS zelf al niet Python 3.9 aanbiedt, die handmatig installeren. Dat laatste gaat namelijk doorgaans vrij eenvoudig en alleen het (her)aanmaken van de virtualenv voor DSMR-reader nodig.

DSMR-reader wordt momenteel in alle 4 beschikbare Python-versies getest en ondersteund:
- https://github.com/dsmrreader/dsmr-reader/actions/runs/1249096812

Policy van mij is:
- Als er een Python-versie EOL gaat, zoals 3.6 eind dit jaar, zal ik die bij de eerstvolgende major release (dus DSMR-reader 5.x) niet meer ondersteunen.
- Als er een nieuwe Python-versie komt, zal ik **proberen** deze te ondersteunen in de huidige DSMR-reader die op dat moment beschikbaar is. Mits het forward compatible is.

Tot nu toe is dat laatste altijd forward-compatible geweest, maar ik hebt daar zelf beperkt invloed. Stel dat Python 3.10 over een paar weken niet werkt, zonder ingrijpende wijzigingen, dan blijft DSMR-reader voorlopig max Python 3.9 ondersteunen.

dennissiemensma, 2021-09-20:
Tevens wordt het (als het goed is) vanaf DSMR-reader 5.x ook wat makkelijker qua virtualenv, omdat die dan in de root van het project komt te staan als `/home/dsmr/dsmr-reader/.venv/` (makkelijker te vinden/weg te gooien is) en tevens beheerd dwordt oor een tooltje dat om PIP heen zit (Poetry). Poetry maakt dan de virtualenv aan, doet de updates wat soepeler/sneller dan nu.