dsmrreader/dsmr-reader #1246: Meternotiatie mindergas.nl word niet geaccepteerd

casrenooij, 2021-01-04:
Beste wensen allemaal !

Ik draai DSMR op een Debian 10 virtuele server op ESX. 
Alles lijkt te werken, behalve de upload naar mindergas.nl. Het lijkt erop dat het een decimaal/duizendtallen notatie probleem is

op de statistics pagina zie ik 1,738.058 staan.

Test ik het zoals is ergens las met de . en , aangepast lukt het ook niet 
curl -v -H "Content-Type:application/json" -H "AUTH-TOKEN:XXXXXXXXXXXXXXXXX" -d "{ \"date\":\"<2021-01-04>\", \"reading\": 1.737,645 }" https://www.mindergas.nl/api/gas_meter_readings
------>>   {"errors":["765: unexpected token at '{ \"date\":\"\u003c2021-01-04\u003e\", \"reading\": 1.737,645 }'"]}

Test ik met : 
curl -v -H "Content-Type:application/json" -H "AUTH-TOKEN:XXXXXXXXXXXXXXXXX" -d "{ \"date\":\"<2021-01-04>\", \"reading\": 1737 }" https://www.mindergas.nl/api/gas_meter_readings 
dan komt de meterstand keurig in de lijst op mindergas.nl te staan. 

Weet iemand hoe dit op te lossen is ?

cas@dsmr:~$ sudo localectl
[sudo] wachtwoord voor cas:
   System Locale: LANG=nl_NL.utf8
                  LANGUAGE=nl_NL
       VC Keymap: n/a
      X11 Layout: us
       X11 Model: pc105

```
DSMR-READER
    App / Python / Database                                                        v4.9 / v3.7.3 / postgresql
    Backend sleep / Datalogger sleep / Retention cleanup                                   1.0s / 5.0s / 672h
    Telegram latest version read / Parser settings                                                 "50" / "4"

DATA
    Telegrams total (est.)                                                                              13855
    Consumption records electricity / gas (est.)                                                   1609 / 309

UNRESOLVED ISSUES
    Too many outgoing MQTT messages queued for transit                                                    now

```

dennissiemensma, 2021-01-04:
Bedankt voor je melding. Ik mis even de inhoudelijke foutmelding. Zou je de exacte foutmelding **uit de DSMR-reader (backend) log** willen delen?

De foutmelding die je met `curl` krijgt komt omdat je daar inderdaad een verkeerde notatie gebruikt. Het is goed om te weten dat de getallen in de webinterface van DSMR-reader **gelokaliseerd** zijn en daarmee afhankelijk zijn van de locale van je browser. Dat is dus **niet** perse hetzelfde formaat als waarin DSMR-reader iets uploadt!

De mindergas-integratie is namelijk lange tijd ongewijzigd en ik zie in mijn eigen account daar ook gewoon dagelijkse meterstanden t/m gisteren.

casrenooij, 2021-01-05:
Hi Dennis, 

uit : /var/log/supervisor/dsmr_backend.log : 
[2021-01-04 13:02:31,721] ERROR    CLIENTS: Init error: MQTT: Failed to connect to broker
[2021-01-04 13:02:32,036] ERROR    MinderGas: Upload failed (HTTP 422): {"error":"Datum is niet beschikbaar"}
[2021-01-04 13:02:32,037] ERROR    Unexpected status code received
Traceback (most recent call last):
  File "/home/dsmr/dsmr-reader/dsmr_mindergas/services.py", line 32, in run
    export()
  File "/home/dsmr/dsmr-reader/dsmr_mindergas/services.py", line 90, in export
    raise AssertionError(_('Unexpected status code received'))
AssertionError: Unexpected status code received
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/tr
oubleshooting.html#logging
[2021-01-04 13:02:35,756] ERROR    MQTT: Failed to connect to broker (192.168.1.71:1883): [Errno 104] Connection reset by peer

Kan het zijn dat mindergas.nl de meterstand meer één keer per dag accepteert ?

Cas.

dennissiemensma, 2021-01-05:
Klopt. Je kunt het beste je handmatige invoer van gisteren uit mindergas weghalen. Of wachten tot morgenochtend.

dennissiemensma, 2021-01-05:
Sidenote: Je hebt MQTT aangezet, maar je broker lijkt niet bereikbaar. Zie de foutmeldingen in dezelfde log.

closed 2021-01-05 by dennissiemensma