dsmrreader/dsmr-reader #1247: Configuratie menu blijft terug verwijzen naar dezelfde pagina.

quadrophenianl, 2021-01-06:
Goede avond Dennis,

Hier ben ik weer eens na enige tijd. De afgelopen maanden heb ik een keer of 10 geprobeerd om versie 3 te updaten naar versie 4, maar dit wilde niet lukken en bracht veel fouten met zich mee. Gelukkig iedere keer een complete backup achter de hand gehad. Aangezien versie 3 gewoon deed wat hij moest doen, heb ik die gewoon laten draaien.

Maar goed, nieuw jaar - nieuwe uitdagingen. Wederom enkele pogingen gedaan zonder succes. Toen dacht ik, ik de-installeer het gewoon en laat de database staan en installeer vervolgens de nieuwste versie. Vreemd genoeg verliep dit naadloos en de interface werkte gewoon onder versie 4. Zelfs alle gegevens waren in de web interface verwerkt, tot zover dus geweldig.

Nu is het enige wat momenteel nog spaak loopt het configuratie menu. Dit opent wel, maar als ik klik op een willekeurig item klik, blijft deze terug verwijzen naar het configuratie menu. Soms, heel soms wordt er wel een menu item geopend.

Ik ben in ieder geval super blij dat hij nu loopt op de laatste versie. Maar is er ook nog een mogelijkheid om deze laatste probleempjes op te lossen?

Ik hoor het graag!

Groeten Mark

dennissiemensma, 2021-01-06:
Bedankt voor je melding. Je kunt dit nog even proberen:

```
sudo rm -rf /var/www/dsmrreader/static

sudo mkdir -p /var/www/dsmrreader/static
sudo chown -R dsmr:dsmr /var/www/dsmrreader/

sudo -u dsmr /home/dsmr/.virtualenvs/dsmrreader/bin/python3 /home/dsmr/dsmr-reader/manage.py collectstatic --noinput
```

Dit kopieert alle statische files voor de webinterface (opnieuw). Probeer vooral daarna even een [geforceerde refresh in je browser te doen](https://www.getfilecloud.com/blog/2015/03/tech-tip-how-to-do-hard-refresh-in-browsers/) of je cache daarin weg te gooien.

Los dat iets op?

dennissiemensma, 2021-01-06:
En daarna nog een deploy voor de zekerheid dat alles bijgewerkt is:

```
sudo su - dsmr
./deploy.sh
```

quadrophenianl, 2021-01-07:
Top, Dennis! Het werkt volgens mij weer vlekkeloos, waarvoor dank!

Een kleine donatie lijkt me na een jaar of 2 gebruik wel op zijn plek, maar ik kan in de nieuwe versie deze knop niet meer vinden. Kun je me deze even doorsturen?

Nogmaals dank en een fijne dag!

dennissiemensma, 2021-01-07:
Fijn om te horen! 

Ik ben gestopt met persoonlijke donaties. Voel je vrij om iets te doneren aan een van de andere mensen of software waardoor DSMR-reader mogelijk is gemaakt: https://dsmr-reader.readthedocs.io/en/latest/credits.html

closed 2021-01-07 by dennissiemensma