dsmrreader/dsmr-reader #1248: Phase voltage/current are set to 0 when data is not send through API

rubenverhoef, 2021-01-07:
I'm using the API to send my meter data to the backend using an [ESP](https://github.com/rubenverhoef/ESP-DSMR-Reader).
I've just implemented the `phase_voltage_lx` and `phase_power_current_lx`.  
This works, but has the following "problem":  
I'm only sending data when it is required by the API, the value is available and the value is changed. Right now when the voltage or current is not changed the DSMR backend is making an graph point with value `0`.

[image]

Can the backed be adjusted so when there is no update on the voltage/current it will not plot an new value or just the same as the previous?

Thanks!

dennissiemensma, 2021-01-07:
Bedankt voor je melding. 

Het kan zijn dat ik je verkeerd begrijp, echter als je `phase_voltage_lx` en /of `phase_power_current_lx` meegeeft, **zul je die elke keer moeten meegeven met de huidige waarde**. 
Dit API endpoint is echt een alias voor een al geparst telegram, waar dus elke keer alle (voorgaande) velden in aanwezig zijn. Het is gebouwd op hoe de slimme meters zelf ook hun data teruggeven in de telegrammen.

Mocht je trouwens op je ESP toegang hebben tot het ruwe telegram, dan is het misschien nog makkelijker om het **v1**-endpoint te gebruiken. Die accepteert het ruwe telegram en dan heb je er bijna geen werk aan. 
Zie de remote datalogger voor een voorbeeld: https://github.com/dsmrreader/dsmr-reader/blob/v4/dsmr_datalogger/scripts/dsmr_datalogger_api_client.py#L78-L83

Het **v2**-endpoint is meer een fallback voor als iemand het ruwe telegram niet heeft, maar wel de losse velden. Of bijvoorbeeld voor het met terugwerkende kracht importeren van metingen uit een heel ander systeem.

rubenverhoef, 2021-01-08:
Om de verstuurde data te reduceren verzend ik dus niet dezelfde waarde als de vorige keer. Dit werkt (naar mijn weten) met de rest van de waardes wel, maar niet met de `phase_voltage_lx` / `phase_power_current_lx`.

Ik kan gewoon alle aanwezige data altijd sturen, ongeacht dat deze niet veranderd is, maar ik vond het juist mooi om geen data te versturen als deze niet noodzakelijk is.

De V1 API gebruik ik niet met dezelfde reden, ik zie meer in het parsen en datareductie op de ESP dan alles elke seconde over het netwerk sturen.

Mijn vraag is dus of je kan toevoegen dat alleen data wordt weggeschreven in de backend als deze ook daadwerkelijk wordt "geïnjecteerd" door de API. Als deze desbetreffende data dus niet aanwezig is in de `POST` /  `PATCH` dat er geen `0` wordt weggeschreven. Eventueel kan de laats bekende, of geen waarde, worden weggeschreven.

Mocht je zeggen dat dat echt niet de bedoeling is dan zal ik zorgen dat alle data gewoon verstuurd wordt. Maar ik vond dat juist een mooie toevoeging van deze API methode.

dennissiemensma, 2021-01-08:
Bedankt voor je toelichting! Ik begrijp je gedachte en manier van denken. Het zou zeker wat toevoegen in een andere situatie zoals MQTT, waarbij een datazender een of meerdere/vele ontvangers kan hebben. In dat protocol zit dan ook een feature om bijvoorbeeld de broker telkens de laatste waarde te laten cachen, zodat nieuwe ontvangers niet hoeven te wachten op nieuwe/gewijzigde data. En ook opties om wel of niet data opnieuw te verzenden als er geen response is.

Voor een REST API is het denk ik wat rechtlijniger, tenzij die specifiek gemaakt is voor bovenstaand scenario. Hoewel het semantisch wel iets kan toevoegen, denk ik niet dat het het werk en de complexiteit waard is t.o.v. de besparing. Het gaat namelijk echt over slechts een aantal Bytes per bericht. Daarnaast draaien de meeste omgevingen in een thuisnetwerk, waarbij data en bandbreedte al minder schaars is en denk ik dat de huidige opzet voldoet.

Dus als antwoord op je vraag raad ik je aan om gewoon alle data te sturen, via welke vorm je dan ook wil.

_Als je trouwens nog makkelijker data wil besparen, kun je ervoor kiezen om slechts elke 10, 20 of zelfs 60 seconden een telegram te sturen. Dat heeft in DSMR-reader alleen invloed op de live grafieken, maar niet op je uur- of dagverbruik._

rubenverhoef, 2021-01-09:
Oke bedankt voor de uitleg, en misschien ben ik een beetje teveel op de bespaar tour.

Ik zal zorgen dat me data gewoon elke keer verstuurd wordt 👍

closed 2021-01-09 by rubenverhoef