dsmrreader/dsmr-reader #1249: Electricity grouping type

rogierlommers, 2021-01-08:
Misschien stomme vraag, maar wat moet ik nu precies invullen bij `Electricity grouping type`? Een clean install geeft me dit:

[image]

Ofwel, per minuut. Echter achter value `By reader` staat default. Dit is nu dus niet het geval? Hoe bepaal ik nu waarop ik deze value moet zetten? En welke impact heeft dit?

dennissiemensma, 2021-01-08:
Bedankt voor je vraag. Het kan zijn dat die logica niet helemaal lekker is of afhankelijk van je meterversie.

De keuze hangt er vanaf hoeveel meetpunten in je de live grafieken wil. Voor het meten van het gebruik maakt het verder niet uit omdat daarvoor altijd het begin en einde van een uur of dag genomen wordt. Groeperen zorgt dus voor wat minder data.

Voor DSMR v4 meters zijn er 6 metingen per minuut. Voor DSMR v5 meters 60, dus vandaar de optie om die 6 of 60 te groeperen om het wat leesbaarder te maken.

closed 2021-01-09 by rogierlommers

dennissiemensma, 2021-01-10:
In de volgende release zal ik het verduidelijken in de admin.